MarlinFirmware/Marlin #2421: Development vs dev

eboston, 2015-07-11:
If I understand right, we should be doing pull requests against the Development branch.  What is the dev branch for?

Wackerbarth, 2015-07-11:
As has been stated before, the dev branch is a rebasing of "Development" and will become the Release Candidate.  The files in it are identical to the files in Development.

Wackerbarth, 2015-07-11:
@eboston -- If my response was adequate, please close this issue.

eboston, 2015-07-12:
Only thing I would add is that a note about it could help in the readme file.

closed 2015-07-12 by eboston

Wackerbarth, 2015-07-12:
Yea, The whole issue of "branches" was somewhat naive. Because the github merge workflow is orientated toward multiple workers on independent sections of the code, we need to have a transition layer to serialize multiple changes to the same (monolithic) piece of code. Additionally, I suspect that there will be a compelling demand to "support" other-than-the-very-latest releases.

The repository structure that was proposed is not really adequate to handle all of those issues.
I suggest that it needs to be updated.  However, until I can get "sign-off agreement", that will not happen.

thinkyhead, 2015-07-13:
It will be interesting to see how the dev branch develops, how it will affect the commit history of the project, the blame log, etc. The myriad differences between Release 1.0.2 and the current Development branch are not always so easy to encapsulate. I would like to hear more about the workflow proposed to build the dev branch from the Development branch. Is it to be based on the HEAD of Release? How are commits from Development brought into dev?

> a compelling demand to "support" other-than-the-very-latest releases

Part of the object with Marlin is to make previous releases obsolete and encourage users to adopt the latest release, so they don't feel compelled to go back to some older version. Basically, if someone is about to re-flash their board, we should make it most attractive for them to download the latest release, where we've already fixed bugs they may encounter in the old release. There is little or no enthusiasm among developers to support what is basically older, buggier code, and I think most users are savvy enough to understand that older versions should be considered unsupported in terms of fixing bugs. The only exception might be small boards that can't hold the newest code, but in fact the newest code should in many cases be far smaller than the older code.

Wackerbarth, 2015-07-13:
@thinkyhead -- Although there is something to be said for a "clean" release (where no line of code is changed more than once), I don't think it is realistic to expect this project to become that formalized.

Therefore, I have been limiting the rewrite of history to the elimination of "spaghetti" merges. I do this by rebasing each PR merge on the "dev" branch that resulted from applying this procedure at the immediately previous merge. This results in a linear chain of changes where each change contains the modifications caused by the PR, and only those changes.  Each line of code should be attributed to the author in the same manner that it is in the current "Development" branch.