MarlinFirmware/Marlin #2422: Where to define pins/functions?

eboston, 2015-07-11:
Now that things have settled down a bit, I am getting back to the display I have been working on.  I am wanting to add some new functionality to use the RGB back light of the display.  Because other displays may have RGB back lights, I did not want to make it a function of my specific display.  In addition to this, it will require pins to be defined to deal with the back light.

What I was thinking would be to define a set of pins for a board to access the back light.  For example, I am working with the RAMPS board so in the pin file for it, I would add 3 pin defines such as RGBBL_RED, RGBBL_GREEN, and RGBBL_BLUE.  Then say in Conditionals.h or pins.h, set a define, RGBBL_ENABLED, if all three of the pin definitions are defined.  Then in locations like gcode_M150, use the RGBBL_ENABLED define to perform the action needed.

Am I on the right track with this?  I'd like to get the locations for the defines figured out before I start the work.

Wackerbarth, 2015-07-11:
Generally, I think that you are on the right track. However, the "where to put it" issue is something that is presently ill-defined. Specifically, these pins are not a function of the RAMPS board itself, but rather a part of the wiring connection your particular display to a particular interface board. So, for now, I would suggest that you place things in a "feature" file and include it in your configuration. Then, when we develop a better specification for our HAL, you can move then into the appropriate location.

Also, RGBBL_BLUE, etc., is somewhat cryptic. And turning on a feature because some pin is, or is not defined, is not, IMHO, a good coding style. We need to be more assertive about the intent. I would prefer that you describe the feature as "HAS_RGB_BACKLIGHT" as an option and then, in the Sanity checks, you can verify that the appropriate pins have been provided to implement the feature.

eboston, 2015-07-11:
I'm still not completely sure how I should be going about this. First, 
the RGBBL_BLUE were just examples and shorter than typing the whole 
thing out.  The feature, HAS_RGB_BACKLIGHT, could be implemented 
different ways depending on the display.  So I could see a file that 
contained the implementation for a particular display, but the file to 
be loaded would be dependent on the display.  So I guess there could be 
one file with any code needed to support this feature and each function 
#ifdef according to the display that is defined.  Something like...

#if defined(DISPLAY1)
void RGBBL_Function1()
{
}
#end if

#if defined(DISPLAY2)
void RGBBL_Function1()
{
}

DISPLAY1 and DISPLAY2 are just examples/placeholders and would be 
replaced with the actual names.  The pins each display would use would 
also be defined in a similar way in the .h file.

Is this what you are thinking of?  And where would you suggest the 
inclusion of the .h file be located at?

Wackerbarth, 2015-07-11:
"First, the RGBBL_BLUE were just examples and shorter than typing the whole
thing out."  -- First, let me say that, over the years, I have grown to prefer verbosity to cryptic names.
Particularly because "source code space" (character positions) is no longer an issue (We are not physically punching each character into a paper tape), and modern editors do code completion for you (type just a few characters and you get the fully expanded name), I see little reason to have cryptic names.

However, RGBBL_BLUE, as the name of a parameter is less of an issue than the name of the option.
IMHO, that name needs to convert as much information as possible to the user who is deciding whether he needs it, or not.

As to the other files, your approach seems fine, for now. It will be easy to cut-and-paste from that structure into any that we might develop.

As to the issue of "where?". The present structure seems to be very flat. I'd just stick an #include in the Configuration.h (guarded by the HAS_RGB_BACKLIGHT which you would specify just above it) and then wait until a better organization gets defined.

eboston, 2015-07-11:
The "shorter" name was for the email, not the actual code. I was typing it
on my phone and shorter is easier.  :)

On Jul 11, 2015 3:54 PM, "Richard Wackerbarth" <notifications@github.com>
wrote:
>
> "First, the RGBBL_BLUE were just examples and shorter than typing the
whole
> thing out." -- First, let me say that, over the years, I have grown to
prefer verbosity to cryptic names.
> Particularly because "source code space" (character positions) is no
longer an issue (We are not physically punching each character into a paper
tape), and modern editors do code completion for you (type just a few
characters and you get the fully expanded name), I see little reason to
have cryptic names.

Wackerbarth, 2015-07-11:
Great, we don't seem to have an issue. I had no indication that you were operating in that mode. Mobile devices are great for some kinds of communication. But writing code is not one of them.

eboston, 2015-07-12:
Looking at the configuration.h file, there is only one include file there.  It would seem that Marlin.h might be the better place for the "feature" include file.

Wackerbarth, 2015-07-12:
Personally, I hope that we will get rid of the Configuration_adv.h and just fold it into a single Configuration.h . But this is based on the idea that we will have "default" configuration files and the Configuration.h will just override the non-default items.

There is a good argument that all of this should be pushed up to Marlin.h . However, as things are presently (mis)configured, I think that keeping your addition to the (somewhat customized) Configuration.h is a better choice in the short term.

Remember that I consider this to be only an interim way to include the code.  I hope that a better structure will evolve.

eboston, 2015-07-12:
Since the feature include file is going to have to know which board is defined, the actual include would have to be done after the board definition.  Personally, I don't like adding include files this way, but understand that it has to be in certain situations.  Is this going to be one of them or should we find a better place for the include?

Wackerbarth, 2015-07-12:
I recognize, and agree with, your concern. However, in this case, rather than "doing it correctly", I advocate handling your case with minimal invasiveness. It's much easier to advocate the inclusion of something in an isolated context.

eboston, 2015-07-12:
How would you feel about it being included at the end of Configuration.h 
after the thermistortables?

Wackerbarth, 2015-07-12:
That seems like a good place.  In my mind, we should end up grouping all of the "local display" functionality together. Since this is a part of that, wherever it may end up, that is where it belongs.
But moving things from one area to another is just "bookkeeping". It will be easy to move things around once we develop a reasonable structure.

eboston, 2015-07-12:
If you have time, could you look in my eboston/Marlin RGB_Backlight branch and see if this is matches to what you are looking for.  Added files RGBBacklight.cpp/.h.  The end of Configuration.h has the new include.  Sanity checks at the bottom of SanityCheck.h.  Changes in Marlin_main.cpp is to the M150 code (line 3984).

thinkyhead, 2015-07-14:
What kind of backlight are you using? We have some code in place for the BlinkM color LED which uses I2C for setting the RGB intensity values. Is BlinkM a potential option for your display?

eboston, 2015-07-14:
BlinkM is a color ring around the rotary encoder.  The LCD I am using has 3 LEDs (Red, Green, Blue) that controls the color of the display's background.  They are 3 separate lines to control, not I2C link BlinkM.  It's possible that a lot of the code in place for BlinkM will work for my display,  I haven't look through where BlinkM is used to know for sure. I might  just need to modify the function that writes the value out.

thinkyhead, 2015-07-16:
@eboston Sounds like you would need 3 PWM lines, which are pretty scarce, especially as some of those timers are stolen for other interrupts in Marlin.

eboston, 2015-07-16:
Ideally, 3 PWM pins would be good.  But on the RAMPS board, they are not 
available.  3 digital pins are so I can do on/off for each color.  A bit 
more limiting, but still functional.  I am writing the code as if I do 
have PWM pins so if they are available on another controller, it can be 
used as such.

thinkyhead, 2015-07-16:
I see. With digital pins (1 per LED) you can get up to 7 different colors at full intensity, which should be pretty good. Red, Yellow, Green, Indigo, Blue, Violet, White. If you pulsed them all together PWM-style you could perhaps adjust their brightness too.

eboston, 2015-07-16:
Yeah.  But I'm not going to go to all that work.  :)  Some of the colors 
on the unit I have are weaker than the other so several of the combined 
colors look the same.  I mainly want the Red, Green, and Blue.  What I 
was thinking of would be to have the display showing Blue when the 
printer is idle.  Change it to Green when it starts printing.  And show 
Red if an error occurs.  Just something simple and basic to start with.

thinkyhead, 2015-07-16:
@eboston Darn that display. A Neopixel or BlinkM style light would be so much easier to adapt. I think you're on the right track using the digital pins, but I suppose you also have the freedom to use any free PWM pins for one or two of them.

eboston, 2015-07-16:
My goal is to use the standard reprapdiscount full graphic display adapter
for the display so that limits the pins I can use. If I design a new
adapter, I could get to some free PWM pins.  But that is another project.
:)

closed 2016-04-23 by jbrazio