MarlinFirmware/Marlin #2424: Hangout -- July 18 UTC 2100

Wackerbarth, 2015-07-11:
Name  | Summer Time Zone
------------ | -------------
@thinkyhead    |  UTC-7
@Wackerbarth  |   UTC-5
@Jnesselr    |    UTC-5
@clefranc     |   UTC-4
@alexborro    |   UTC-3
@fmalpartida   |  UTC+1
@AnHardt     |    UTC+1
@Wurstnase    |   UTC+1
@CONSULitAS   |   UTC+1
@paclema     |    UTC+1
@boelle    |      UTC+1
@emartinez167   |  UTC+8
@ntoff    |       UTC+10

[The "Official" UTC Clock](http://www.timeanddate.com/time/aboututc.html)
[Link to the Hangout](https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia)

Please provide input about what we need to discuss.
[Link to Google Document listing discussion topics (Week 29)](https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing)

The primary current issue is getting bed leveling working to a point that we can do a Release Candidate.
However, we also need to review all of the pending items and possibly include some of them in this release.

Additionally, we need to make some plans about how we intend to move forward. There are a number of things that various developers would like to have included, but they are holding off to minimize impact on the release of the current improvements. How do we want to handle this aspect?

thinkyhead, 2015-07-13:
I don't mind doing this an hour or three earlier, since this is pretty late for most of us.

Wackerbarth, 2015-07-13:
I, too, would be happy with any reasonable time change.

emartinez167, 2015-07-13:
Apologies in advance but I am on holidays in the Great Barrier reef and
won't be able to attend this time.

fmalpartida, 2015-07-14:
O will try to make it. An hour before would help as it would be 11 PM here.

Wackerbarth, 2015-07-16:
@thinkyhead @Jnesselr @clefranc @alexborro @fmalpartida @AnHardt @Wurstnase @CONSULitAS @paclema  @ntoff

Hearing no objections,  PLEASE NOTE THE REVISED TIME

Jnesselr, 2015-07-16:
Objection! I apologize, but something came up and I won't be able to make
it this meeting.

Wackerbarth, 2015-07-16:
@Jnesselr -- So, why are you objecting to the time change? I assume that you cannot make either one of them.

Jnesselr, 2015-07-16:
That is correct. I was just joking. No real objections here.

thinkyhead, 2015-07-16:
@Jnesselr Oh, you! Objection overruled. :stuck_out_tongue_winking_eye:

CONSULitAS, 2015-07-18:
@wackerbarth
New time is great!
But I apologize, I am very much to far from any reasonable Internet connection. (Great place, btw)

fmalpartida, 2015-07-18:
I may need to miss out the HO tonight, kids out of school and are demanding attention and their afternoon evening out. 
I will try to call in or tap in at the beginning or end. 
One thing that I would like to add to the agenda is the integration into the integ branch from dev. I think that at this point, defects can be expedited there to get ready for release. 
This should allow for new features to be developed in the development branch.

AnHardt, 2015-07-18:
@fmalpartida :+1:

closed 2015-07-18 by Wackerbarth