MarlinFirmware/Marlin #2430: How to manage a moving platform

zeryabmoussaoui, 2015-07-13:
Is it possible to configure Marlin to manage a moving-plaform ?

If not , how about this kind of features ?

AnHardt, 2015-07-13:
> Is it possible to configure Marlin to manage a moving-platform ?

Yes it is. For a normal printer, working in a cartesian coordinate system, moving the bed is the same as moving the nozzle - but the opposite direction.

zeryabmoussaoui, 2015-07-13:
@AnHardt 
Thanks for your answer .
i planned to make the bed 2DoF and the noozle 1DoF, so alll i have to do is to consider it as a 3 DoF noozle ?

AnHardt, 2015-07-13:
Pardon.
What does DoF mean?

Wurstnase, 2015-07-13:
Degree of freedom :)

zeryabmoussaoui, 2015-07-13:
Thank you very much ,  I was expecting some specific treatments in this case ,  do you know any printer with the plateform moving based on Marlin ?

Wurstnase, 2015-07-13:
Velleman for example.

zeryabmoussaoui, 2015-07-13:
Thank you very much ! 
This issue can be closed.

closed 2015-07-13 by Wurstnase