MarlinFirmware/Marlin #2434: Stack buffer overflow in CardReader::lsDive

avri-schneider, 2015-07-13:
If `prepend` is empty string (`strlen(prepend) == 0`), `path` can hold only `FILENAME_LENGTH + 1` bytes (`FILENAME_LENGTH` + terminating `null` character)

```cpp
    int len = strlen(prepend) + FILENAME_LENGTH + 1;
    char path[len];
```

Then later it is concatenated with an additional `"/"`, space for which was not allocated on the stack (`lfilename` + `"/"` + terminating `null` character).
```cpp
if (prepend[0] == '\0') strcat(path, "/"); // a root slash if prepend is empty
strcat(path, prepend);
strcat(path, lfilename);
strcat(path, "/");
```

closed 2015-07-16 by AnHardt

avri-schneider, 2015-07-16:
Not sure this is enough.

What is allocated is:
`FILENAME_LENGTH` + `trailing slash` + `terminating null`

What is copied over is:
`prepend slash` + `FILENAME_LENGTH` + `trailing slash` + `terminating null`

Need to add another `+ 1`.

AnHardt, 2015-07-16:
`FILENAME_LENGTH` is defined as 13 =  8 + "." + 3 + '/0'. So the `terminating null` is already in.

thinkyhead, 2015-07-16:
I will change the comment on that line to be more accurate.

avri-schneider, 2015-07-16:
@AnHardt Thanks. I stand corrected.

thinkyhead, 2015-07-16:
#2455 allocates only the space needed, saving stack space for filenames shorter than 12 characters in length.

avri-schneider, 2015-07-16:
`int len = strlen(prepend) + (prepend_is_empty ? 1 : 0) + strlen(lfilename) + 1;`
if `prepend_is_empty` there's no need for `strlen(prepend)`...

avri-schneider, 2015-07-16:
Is this better?
`int len = (prepend_is_empty ? 1 : strlen(prepend)) + strlen(lfilename) + 1;`

avri-schneider, 2015-07-16:
Also, `strcat(path, prepend);   // 1 character minimum` should be called only if `!prepend_is_empty` (otherwise it is a redundant function call, and the comment is incorrect, as nothing will be concatenated)

I would change:
```c
else
        path[0] = '\0';

      strcat(path, prepend);   // 1 character minimum
```
to:

```c
else
{
        path[0] = '\0';
        strcat(path, prepend);   // 1 character minimum
}
```

thinkyhead, 2015-07-16:
Probably going to go with this…

      // Allocate enough stack space for the full path to a folder, trailing slash, and nul
      boolean prepend_is_empty = (prepend[0] == '\0');
      int len = (prepend_is_empty ? 1 : strlen(prepend)) + strlen(lfilename) + 1 + 1;
      char path[len];

      // Append the FOLDERNAME12/ to the passed string.
      // It contains the full path to the "parent" argument.
      // We now have the full path to the item in this folder.
      strcpy(path, prepend_is_empty ? "/" : prepend); // root slash if prepend is empty
      strcat(path, lfilename); // FILENAME_LENGTH-1 characters maximum
      strcat(path, "/");       // 1 character

thinkyhead, 2015-07-16:
It looks like the code to test for an empty prepend could be removed, saving 28 bytes, if all the calls to `lsDive("", ...)` were changed to `lsDive("/", ...)`. That would cost 4 bytes, so total code reduction would be an additional 24 bytes.