MarlinFirmware/Marlin #2435: If G29 is used after G28 X Y then Z coordinate is NOT displayed

lrpirlet, 2015-07-13:
It looks like G29 ABL is able to determine the Z coordinates. In fact, X and Y must be homed while Z is computed in G29 ABL. So far so good...

However, the display will NOT assign a value to the Z axis when it has NOT been explicitly homed.

This is not a big deal, the workaround is to always home all 3 axis before to issue G29 ABL... 

I would be nice to have the Z coordinate displayed at the end of G29 ABL.

AnHardt, 2015-07-14:
I you asked me, i'd sugest to simply home Z with G28.
G28 is quite complex. Fiddling in all that into G29 will complicate things even more.

For a very limited scope (for example no deltas, no G92, ) you can try:

    @@ -1233,7 +1233,7 @@ static void setup_for_endstop_move() {
          feedrate = homing_feedrate[Z_AXIS];
    
           // Move down until the probe (or endstop?) is triggered
          float zPosition = -10;
          float zPosition = -(Z_MAX_LENGTH);
          line_to_z(zPosition);
          st_synchronize();
    
    >@@ -2862,6 +2862,7 @@ inline void gcode_G28() {
              current_position[Z_AXIS] = z_tmp - real_z - zprobe_zoffset; // The difference is added to current position and sent to planner.
            #endif
            sync_plan_position();
            axis_known_position[Z_AXIS] = true;
          }
        #endif // !DELTA

lrpirlet, 2015-07-14:
@AnHardt Thanks, I was NOT thinking of "correcting" G29 except maybe by forcing the homing of all 3 axis in G29 depending on a define in the configuration.h file.

 I was more thinking of documenting this behaviour in the 103 version so people stop reporting that as  bug... 
The no display when position "uncertain" was introduced with 103... We should document that not using G28 for Z homing before using G29 puts the Z position in "uncertain" mode... 

*Please note this is MY opinion...*

thawkins, 2015-07-15:
@lrpirlet  perhaps G29 Should display an error if Z is unhomed or undefined. It would all the stupid gcode starts on the net that say

G28 X0 Y0
G29

AnHardt, 2015-07-15:
Prepared two varietys:
https://github.com/AnHardt/Marlin_O/pull/9 only giving a hint
https://github.com/AnHardt/Marlin_O/pull/10 stopping the G29

Any preferences?

I'd prefer the stopping variant.

lrpirlet, 2015-07-15:
@AnHardt Thanks, it looks like #2446 you have introduced perfectly answer my request... 

To be honest, I would not mind stopping G29 if the press of a button (either on the display or in the Host software) can restart the print... The choice would then be active... But I am not sure we should kill a procedure that used to work... also, to stop the G29 for no Z coordinate display would be a real annoyance for those printer owners without display...

AnHardt, 2015-07-15:
https://github.com/AnHardt/Marlin_O/pull/10 not really stops the printer. Only the G29 is not performed. Then the printer is ready for the next command.

avri-schneider, 2015-07-16:
Why not automatically home required axis when needed by calling G28 from G29?

thinkyhead, 2015-07-16:
> Why not automatically home required axis by calling G28 from G29?

Well, the `gcode_G28` function isn't set up for that. We'd have to send a `G28` command, and then there's no facility to wait for that command to complete before proceeding. Plus, there is a religious objection, which is that G-Codes shouldn't invoke other G-Codes. In any case, the main thing we're trying to solve is the homing of Z before beginning a Z probe. If a machine uses a Z_MAX endstop, it would make the most sense to home there, whereas if a machine uses a Z probe as its Z endstop, then it makes the most sense to use the first probe as the Z homing. If a machine has a Z_MIN endstop, plus a separate bed probe, then it makes the most sense to use that endstop to home.

closed 2016-04-23 by jbrazio