MarlinFirmware/Marlin #2436: Allow averaging for G29

foulowl, 2015-07-13:
Hi folks,

I realized that part of my bed auto leveling issue might have to do with repeatability. ( https://github.com/MarlinFirmware/Marlin/issues/2413 ) If I run G29 multiple times, I get very similar results, but not _exact_ results each time. My guess is that there is some fundamental inaccuracy with my servo.

Thus, I have a feature request to suggest that might help here. I would like to propose G29 R<val> where val=number of times to perform the bed leveling action. Ie, G29 R5 would perform bed autoleveling 5 times. It would then average the data gathered for each point, which I believe should allow for greater accuracy. 

Thanks for all your hard work devs!!

Roxy-3DPrintBoard, 2015-07-13:
I suggest you do several M48 commands at different places on the bed.   Try these commands and report the numbers you get:

M48 X 100 Y 100
M48 X 50 Y 50
M48 X 150 Y 150
M48 X 100 Y 100 L 7
M48 X 50 Y 50 L 7 
M48 X 150 Y 150 L 7

This will help you (and us) understand how repeatable your Z-Probe is.

foulowl, 2015-07-13:
Thank you! Let me try that.

foulowl, 2015-07-14:
Doing E1 for all tests, since it is likely an issue with the servo positioning if any.
M48 X60 Y60 E1: 50 microns stdev
M48 X100 Y100 E1: 50 microns stdev
M48 X150 Y150 E1: 30 microns stdev
M48 X60 Y60 L7 E1: 30 microns stdev
M48 X100 Y100 L7 E1: 50 microns stdev
M48 X150 Y150 L7 E1: 30 microns stdev

So perhaps the issue isn't with my servo, but maybe it's the issue with qr_solve mentioned here: https://github.com/MarlinFirmware/Marlin/issues/2413#issuecomment-120664122

foulowl, 2015-07-14:
This might still be a useful feature, but I don't want to clog your issue list unnecessarily.

Thanks!

closed 2015-07-14 by foulowl