MarlinFirmware/Marlin #2437: Revert M204 to use S and R arguments

thinkyhead, 2015-07-13:
Before March 11, 2015 the arguments to `M204` were `S` for printing moves and `T` for retract moves. The commit https://github.com/MarlinFirmware/Marlin/commit/9157cbd8f340579958901c2e77dbcfba687d54d7 by @alexborro changed the arguments to `P` for printing moves, `R` for retract moves, and `T` for travel moves. This kind of change has been cited by @repetier as potentially problematic for host and slicer developers, because it redefines the meaning of an existing argument. Moreover, the use of `T` for anything other than the Tool Number is generally discouraged, because it runs slightly afoul of GCode standards. The original `S` argument was meant to follow general GCode standards because `S` is typically used in cases where a single value is needed, so when codes are extended with new arguments (such as `R` for travel moves) the old single argument tends to be retained as the `S` code.

So, I propose reverting `M204` back to the original — with `S` for print moves and `R` for travel moves, and that we do not extend `M204` for retract moves, because `M207 Fnnn` already allows you to set the retract feedrate.

repetier, 2015-07-14:
Just to be clear. Since I noticed this by chance I have rewritten the host for version 1.5 to handle the new M204 correctly. Only older hosts will show wrong eeprom data and can not set all values.

So switching back is fine if you do it 100% so that the new host thinks it is the old version, otherwise you would worsen the problem even more.

ntoff, 2015-07-14:
Wasn't the problem of redefining stuff already discussed and you all kind of agreed to not do this without consulting the developers of host programs?

Wurstnase, 2015-07-14:
@thinkyhead To be compatible we could also support 'S'. This will set 'T' and 'P' simultanious.
@ntoff Yes, but not any 'issues' are found before discussing that stuff.

thinkyhead, 2015-07-16:
@ntoff This is an old change, by someone not deeply involved in those more recent discussions.

thinkyhead, 2015-07-16:
@repetier If you were to grab the Marlin version from `M115` that could tip you off about which `M204` to expect also.

I hope to document any additions or changes to G-Codes between Marlin versions on the marlinfirmware.org wiki, so host and slicer developers will have a definitive reference for what is implemented in each version, starting with "1.0.0" from around October, 2014.

thinkyhead, 2015-07-16:
> To be compatible we could also support 'S'.

True. I support that idea.

repetier, 2015-07-16:
I do not really trust versions any more. I've seen vendors changing names and version if they like. That sometimes even broke firmware detection and then I get questions like why does feature xy not work. Reason is often changed firmware name (here repetier was removed so repetier eeprom support was not active). And I also guess no one increased version when it was introduced. If version shows everything you need to increase with each new/modified command. 

I do distinguish the version here by the result. The new one has a new data that was not present, so I used that to detect. That way a 100% undo will also work regardless of the version.

thinkyhead, 2015-07-16:
> I do not really trust versions any more.

I can understand that. I wish we had some way to enforce it so that `M115` was meaningful. One way would be to make a version (or other indicator) that's buried in the source outside of `Configuration.h`. But ultimately, I agree that the best solution is to keep backward compatibility as far as possible.

Wackerbarth, 2015-07-16:
"make a version (or other indicator) that's buried in the source outside of Configuration.h" --
Except that it is included by Configuration.h, you already have that.

thinkyhead, 2015-07-16:
> no way to actually control anything that is in the compiled firmware

True, but if it were obscure enough… even something silly like including a special keyword like "squirrel-bait" in the `M115` report (or elsewhere), or adding some new hard-coded command that always outputs the base version. I know it's reaching, but otherwise hosts can only _glean_ from the responses they get what to expect. It would be nice to be able to give them something explicit.

thinkyhead, 2015-07-16:
> To be compatible we could also support 'S'

@Wurstnase I'm happy to report that `M204` currently still supports `S` for legacy reasons.

Wurstnase, 2015-07-16:
Right, I have forgotten it. But only the set-function and not the get-one which causes the problem with RepetierHost pre1.5.

thinkyhead, 2015-07-17:
@Wurstnase I suppose the output of `M503` could be changed, so that if the two feedrates are equivalent, it will output `S` instead…
```
  SERIAL_ECHO("  M204", acceleration);
  if (acceleration == travel_acceleration) {
    SERIAL_ECHOPAIR(" S", acceleration);
  }
  else {
    SERIAL_ECHOPAIR(" P", acceleration);
    SERIAL_ECHOPAIR(" R", retract_acceleration);
    SERIAL_ECHOPAIR(" T", travel_acceleration);
  }
```

Wurstnase, 2015-07-17:
Yes, but what does happen when you set P and T to the same and want it set back to different values? I see crying people to Repetier "the setting is hiding, your tool is broken".

repetier, 2015-07-17:
Do not make it complicated. Currently it is included and from M503 output it is pretty easy to distinguish the 2 versions. So for me it is ok to keep current behaviour as a special case. And since it happend before the compatibility discussion occured, lets keep it as backward compatibility issue and just let us concentrate to not increase the amount of these incompatibilities in future. Users just have to update the host now if they update firmware and want to use eeprom correctly. We have already users using it that way so it is too late to undo anyway.

Wurstnase, 2015-07-17:
:+1: 
Does any other Host has a EEPROM-Menu? With 'S' all old softwares can set this attribute. 

Only the answer could be reseted. Actually when I send a M204 Sxxx the firmware answers with 'set travel and print-acc'. This could be the old one. Anything else should be as it is.

thinkyhead, 2015-10-12:
Check out the latest RC (RC2) and see that it's doing the right thing. Re-open if this is still worth discussing.

closed 2015-10-12 by thinkyhead