MarlinFirmware/Marlin #2439: Dual Extruder works only with "MOTHERBOARD  BOARD_ULTIMAKER" setting!?

Ozista, 2015-07-14:
Hi,

MarlinFirmware has truly got better with every new versions as this last "Marlin-Stable" thanks to all in the community!

I have had a great run with my new CoreXY so I am now advancing to dual extruder.

My Arduino Mega 2560 board was working fine with "#define MOTHERBOARD 33 //BOARD_ULTIMAKER" setting up till now but when I changed "#define EXTRUDERS 1" TO "#define EXTRUDERS 2", following errors being generated after "Verifying the code in "Arduino 1.0.6"

"temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_RPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_RPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1249:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1249:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_RPORT" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_WPORT" on integer constant
temperature.cpp:1271:1: error: pasting "DIO" and "-" does not give a valid preprocessing token
temperature.cpp:1271:1: error: invalid suffix "_PIN" on integer constant
temperature.cpp: In function 'void __vector_22()':
temperature.cpp:1249: error: 'DIO' was not declared in this scope
temperature.cpp:1249: error: 'DIO' was not declared in this scope
temperature.cpp:1271: error: 'DIO' was not declared in this scope

Conclusion:
I can only run my CoreXY when CONFIGURATION_H configured with:
"MOTHERBOARD  33
#define EXTRUDERS 1"

or

"MOTHERBOARD  BOARD_ULTIMAKER
#define EXTRUDERS 2" 

Further to my findings, my "#define REPRAP_DISCOUNT_SMART_CONTROLLER" is seem to be effected by the  'BOARD_ULTIMAKER" setting.

Thanks again for your time.
Ozista

AnHardt, 2015-07-14:
When you want to use a RAMPS with 2 Extruders you have to use BOARD_RAMPS_13_EEB      34// RAMPS 1.3 / 1.4 (Power outputs: Extruder0, Extruder1, Bed)!

Ozista, 2015-07-14:
Thank You for the idea AnHardt!!

Please Exuse me for my ignorance i taught it was an error.
I have done some research and yes the

"#ifndef MOTHERBOARD
 #define MOTHERBOARD BOARD_RAMPS_13_EEB // RAMPS 1.3 / 1.4 (Power outputs:
Extruder0, Extruder1, Bed)
#endif"

will work for -me without the "34" in front of the "//"

Thanks again
Ozista

Istvan Toth
Designs & IT
Ph.:      +61 2 9957 3085
Mobile: +61 4 0724 6296
Email: antopi.hu@gmail.com
Skype & Twitter: *ozista*

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then delete this message and any attachments.
P Please consider the environment before printing this e-mail.

closed 2015-07-15 by AnHardt

thinkyhead, 2015-07-16:
> When you want to use a RAMPS with 2 Extruders you have to use BOARD_RAMPS_13_EEB 34

@AnHardt Is this a hard limitation – enough that we should sanity-check it? Perhaps the pins files should include a master value like `MAX_EXTRUDERS` that we more generally test against….

AnHardt, 2015-07-16:
@thinkyhead 
I can't make general statement. In this particular case EFB and EEB switch the pin between Fan and second Extruder (and that is all the difference). If you'd have an external MOSFET-driver you could use quite any pin but you'd have to define the pin by yourselves (like on the other - not RAMPS boards).

thinkyhead, 2015-07-16:
@AnHardt Yes, it makes sense to leave it open for users to experiment. I think it's probably enough that we're just checking for the existence of the needed pin definitions. If there's no heater 1 pin, for example.

Ozista, 2015-07-16:
Thank you guys I hope I contributed at lease a smidget but I got my answer and learned a great deal.

Ozista

thinkyhead, 2015-07-16:
@Ozista We're learning a lot too! Marlin is just an endless box of new discoveries. :wink:

Ozista, 2015-07-16:
Thank you Scott.

Cheers Istvan