MarlinFirmware/Marlin #2447: revert

setar, 2015-07-15:
commit  
SHA:	c90e80d1a379bfad99a618fef1dd2540d3927713
Author:	mkile <mkiles81@gmail.com>
Date:	Tue Jul 14 2015 11:49:35 GMT+0300 (MSK)
Subject:	Slightly updated translation

It contains many errors.
please rollback and if possible to assign the vote to future language file.

Wackerbarth, 2015-07-15:
@mkile , @setar -- Although I neither authored the change, nor committed it, I hope that you (and any other speakers of Russian) can reach a consensus as to the appropriate translations.

@setar -- Please propose a Pull Request with the translations that you prefer.

setar, 2015-07-15:
ок
We made the Pull request after discussion
sorry for the fuss  :)

closed 2015-07-16 by thinkyhead