MarlinFirmware/Marlin #2450: keeps homing repeatedly

coalman321, 2015-07-16:
i have a printr bot and replaced the board. i am currently setting it up to run with a ramps 1.4. as i was setting up the end-stops and testing them it began to go to "home" and try to keep going even though it was hitting the button repeatedly. i had checked to see if the end stop was hooked up correctly. it was and it was properly inverted. for some reason it seems to ignore the stop.
it also will not move the motors when told i don't know if this is related
i have never personally worked with marlin and reconfiguring it before any help would be appreciated.

thinkyhead, 2015-07-16:
Please post your configuration, and the results of your testing with `M119`.

coalman321, 2015-07-16:
how?

thinkyhead, 2015-07-16:
You can paste your configuration here (prefaced and followed by three backticks), or post it as a [Gist](https://gist.github.com/), then post the link here, or use [Pastebin](http://pastebin.com/) and post the link here.

coalman321, 2015-07-16:
this should work
https://gist.github.com/18d29b55fb80756566e0.git

thinkyhead, 2015-07-16:
Try uncommenting this line:

    //#define DISABLE_MAX_ENDSTOPS

coalman321, 2015-07-16:
ok but i dont' have max endstops..

thinkyhead, 2015-07-16:
Yes, that's the point. Marlin doesn't know that until you tell it, by enabling that option.

coalman321, 2015-07-16:
oh ok i get it now
i must have forgot it when i changed to a new fresh version.
i had a version i had changed several months ago
any idea about the motor not working when i tell it to move?

thinkyhead, 2015-07-16:
I looked at the Printrbot Marlin fork and its default `Configuration.h` still has that line commented. That might be okay for Marlin 1.0.0, but going forward it's definitely not going to work.

coalman321, 2015-07-16:
im running 1.0.2 so thats probably why

thinkyhead, 2015-07-16:
@coalman321 Just in case there are other important changes, you should do a comparison of your `Configuration.h` and `Configuration_adv.h` to those included in Release 1.0.2. There might be one or two other important tweaks made since 1.0.0, or whichever version the Printrbot is basing its distribution firmware on.

coalman321, 2015-07-16:
i just tried it
it is still doing the same thing it was before..
hitting the switch and ignoring it
the motors are also no t responding still

AnHardt, 2015-07-16:
Use M119 ans see if you can trigger the endstops manually.

coalman321, 2015-07-16:
when i hit home?
or do i actually press it myself
because i have tried that

coalman321, 2015-07-16:
Note: i tried to run the motors with the printrbot configs in the motion settings and it locks the motors

coalman321, 2015-07-16:
here is my current configuration.h
https://gist.github.com/7504835fb5a92cbf7ad4.git

AnHardt, 2015-07-16:
@coalman321 
Switch off the printer.
Move the extruder to a place where no enstops should be triggered.
Turn on the printer.
Connect a host program.
Send M119
Trigger an endswitch. (with a finger ore scew driver)
Send M119 during triggered.
Release the endswitch.
Send M119.
Trigger the next endswitch.
Send M119.
...
until you tested all endstops.

Compare if the outputs of M119 is what you'd expect.
M119 should only report endstops you actually have on the printer. If it reports more - deactivate them in the Configuration.h.

coalman321, 2015-07-16:
ok

coalman321, 2015-07-16:
nevermind
i found the log in my host and have been using the command
when i run it it tells me that all of the mins are open even if they are contacted and all of the maxes are triggered

AnHardt, 2015-07-16:
https://gist.github.com/7504835fb5a92cbf7ad4.git
Uh - no Configuration is really bad. :-)

coalman321, 2015-07-16:
ok i changed to default configs and got to where i was last
i realized that i may not have had a pullup resistor in the stop so i added one
all of the endstops now read open
but the min stops are being ignored even when triggered

also any idea what this means?
printer stopped due to errors. Fix the error and use M999 to restart.(Temperature is reset. Set it after restarting.)

thinkyhead, 2015-07-16:
That last configuration still doesn't define `DISABLE_MAX_ENDSTOPS`. Uncomment that so that the machine isn't reading the non-existent endstops.

coalman321, 2015-07-16:
oops my bad
i just added the pullup to the x axis and am testing now

coalman321, 2015-07-16:
any idea about the error though?

AnHardt, 2015-07-16:
Your log is very short. But i guess you got a MINTEMP error. Set up your #define TEMP_SENSOR_0 (#2452)

coalman321, 2015-07-16:
it is not currently hooked up for the fact i am working on it. can i make it a dummy for now?

AnHardt, 2015-07-16:
Yes you can. But don't forget to to change again to a real sensor when you try to heat anything.

coalman321, 2015-07-16:
ok well i set it to zero and i have got the x and y axis to move according to the host and am fixing the z axis now

coalman321, 2015-07-16:
all axis are now responding properly
thank you for all of your help
i will keep the thread open for a while longer in case i run into other problems

coalman321, 2015-07-16:
where is the size function located because the printer stops well be fore it hits its mechanical limits?

coalman321, 2015-07-16:
never mind i found it

coalman321, 2015-07-17:
quick question, is there a way to set the voltage for a hotend slot t say 12 volts and hover around there?

closed 2015-07-17 by coalman321

thinkyhead, 2015-07-18:
@coalman321 The voltage will be fixed, dependent on what your board and your PSU can handle, but the current is controlled by PWM, which you probably know always sends 12V to the output, but in high-frequency pulses via MOSFETs. You can set the voltage to stay on, with no PWM (and in fact this will happen automatically when you need to send the full current), but most of the time you want PWM.

AnHardt, 2015-07-18:
@coalman321 
If you integrate a PWM over time, 100% means full voltage, 0% means 0 volt.