MarlinFirmware/Marlin #2452: Printer stops

xKamix, 2015-07-16:
While I was printing, the printer suddenly stopped and I had to restart it. When I restarted the printer, I started to get this error. I pressed Reset in Pronterface hopinh that would fix it, but it didn't. Ive checked the hot end and it is intact. I'm not sure what is wrong, Ive downloaded the newer version of pronterface, but not marlin. I have the 3D Printer from Makeblock.
Here is the log when I connect the printer;
Connecting...
start
Printer is now online.
echo: External Reset
Marlin 1.0.0
echo: Last Updated: Jul 15 2015 13:22:22 | Author: (CC, default config)
Compiled: Jul 15 2015
echo: Free Memory: 4340  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
Error:0
[ERROR] Error:0
 
: Extruder switched off. MINTEMP triggered !
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
[ERROR] Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
 
echo:SD init fail

If anyone could help me, that would be greatly appreciated. The extruder doesn't heat up and the motors also dont move.

thinkyhead, 2015-07-16:
Is it possible your hot-end thermistor became detached, failed, or has a short? Check the connections and test the thermistor with a multimeter to make sure it's okay.

Unfortunately, you're running an older version of Marlin, so we can't look into code issues that might be giving a spurious error. You may have better luck with 1.0.2 or the latest Development branch, and then we would be able to more easily troubleshoot. You may also get a more comprehensive error message.

AnHardt, 2015-07-16:
```
Error:0
[ERROR] Error:0

: Extruder switched off. MINTEMP triggered !
```
means the values Marlin can red from your thermometer in the first extruder are implausible low.
That can be caused by:
Defect thermistor, unconnected thermistor, wrong entry in TEMP_SENSOR_0 or a to high number in HEATER_0_MINTEMP

xKamix, 2015-07-17:
@AnHardt Should I try changing the values in Marlin code? Also, it worked before, so it would probably be the thermometer? If it is, how would there be a way to fix it? Purchasing a new one? Somehow connecting it?

thinkyhead, 2015-07-18:
Do you get meaningful temperature readings from the thermistors? You can use your host software "Monitor printer" feature, or issue the `M105` command.

AnHardt, 2015-07-18:
@xKamix 
If it worked before and you have not changed anything in the software, don't change it.
On almost all printers the nozzle is moved - so the thermistors cabling. Most likely the failure is there. Disconnect the thermistor on the board side and measure the resistance. A infinite resistance probably means a broken cable. To be sure measure the thermistor directly. Values around 75-100 KOhm are normal depending on the type of the thermistor and the temperature.

closed 2015-09-25 by AnHardt