MarlinFirmware/Marlin #2454: Firmware.

bothal3, 2015-07-16:
Hallo There.
I have just started with my Arduino mega, Ramps 1.4 and Drivers to get X,Y and Z movement.
Is there perhaps a list or table that state the different xx.h files and there purpose?
How can I remove the ones I do not need at the moment?

Does a version exist that does not include extruders, endstops and temperature sensors?
Thank you.
Louis.

thinkyhead, 2015-07-16:
We don't have all the files documented anywhere (yet), and this is currently a bit of a moving target, because we have been making many changes in the last few months. But, you don't need to remove files. Marlin will only include the code it needs, as configured in `Configuration.h` and `Configuration_adv.h`. The `.h` files are only included from those `.cpp` files that need them.

If you're trying to get down to the minimum code required to do various movements with your specific hardware, you can start by removing the `pins_BOARDNAME.h` files that you don't need, get rid of `example_configurations`, `scripts`, `configurator`, `libraries`, the items in `fonts` that you don't need, and the various `dogm_font_data_FONTNAME.h` files. You can strip out all the `language_[code].h` files except `language_en.h`. Then, try compiling with only the options you need (by editing `Configuration.h` and `Configuration_adv.h`), and as you go, try removing other files that seem extraneous, compiling after each one is moved out of the Marlin folder. Eventually you will have just the minimum files you need, and you can go into the code and start stripping out the conditional code that you're not using, again doing compile tests as you make your edits.

bothal3, 2015-07-17:
Thank you Scott.

I will try that. The reason why I must try that is not the size but it seems as if the program is waiting for temperatures to be reached (that I do not have) before I can start moving the axis.
I also suspect that the pin configutration might not be correct for the Arduino mega and Ramps 1.4 combination. I do not find these names in the config.h and avd files.

Best regards.

Louis.

Date: Thu, 16 Jul 2015 13:16:38 -0700

bothal3, 2015-07-17:
If of any use for a start. [List of files in Marlin folder]
The file names are fairly self explanatory. Maybe a list, of the different parts covered , could be included next to each file name (where applicable)?

Regards.

Louis.
  
    
    
  
  
 		 	   		  
boards.h	3105	13/07/2015	07:38:40
cardreader.cpp	13700	13/07/2015	07:38:40
cardreader.h	3079	13/07/2015	07:38:40
Configuration.h	35863	13/07/2015	07:38:40
Configuration_adv.h	21682	13/07/2015	07:38:40
ConfigurationStore.cpp	14963	13/07/2015	07:38:40
ConfigurationStore.h	472	13/07/2015	07:38:40
digipot_mcp4451.cpp	1639	13/07/2015	07:38:40
dogm_font_data_marlin.h	15776	13/07/2015	07:38:40
dogm_lcd_implementation.h	19573	13/07/2015	07:38:40
DOGMbitmaps.h	13203	13/07/2015	07:38:40
fastio.h	84806	13/07/2015	07:38:40
language.h	6832	13/07/2015	07:38:40
language_an.h	8076	13/07/2015	07:38:40
language_ca.h	7558	13/07/2015	07:38:40
language_de.h	7597	13/07/2015	07:38:40
language_en.h	7408	13/07/2015	07:38:40
language_es.h	8071	13/07/2015	07:38:40
language_eu.h	7675	13/07/2015	07:38:40
language_fi.h	7520	13/07/2015	07:38:40
language_fr.h	7596	13/07/2015	07:38:40
language_it.h	7552	13/07/2015	07:38:40
language_nl.h	7521	13/07/2015	07:38:40
language_pl.h	7646	13/07/2015	07:38:40
language_pt.h	7960	13/07/2015	07:38:40
language_ru.h	8560	13/07/2015	07:38:40
LiquidCrystalRus.cpp	10878	13/07/2015	07:38:40
LiquidCrystalRus.h	3193	13/07/2015	07:38:40
Marlin.h	8652	13/07/2015	07:38:40
Marlin_main.cpp	156422	13/07/2015	07:38:40
MarlinSerial.cpp	7369	13/07/2015	07:38:40
MarlinSerial.h	5692	13/07/2015	07:38:40
motion_control.cpp	6961	13/07/2015	07:38:40
motion_control.h	1372	13/07/2015	07:38:40
pins.h	81162	13/07/2015	07:38:40
planner.cpp	42319	13/07/2015	07:38:40
planner.h	6814	13/07/2015	07:38:40
qr_solve.cpp	41218	13/07/2015	07:38:40
qr_solve.h	1082	13/07/2015	07:38:40
Sd2Card.cpp	23398	13/07/2015	07:38:40
Sd2Card.h	9287	13/07/2015	07:38:40
Sd2PinMap.h	13058	13/07/2015	07:38:40
SdBaseFile.cpp	56198	13/07/2015	07:38:40
SdBaseFile.h	18315	13/07/2015	07:38:40
SdFatConfig.h	4554	13/07/2015	07:38:40
SdFatStructs.h	25313	13/07/2015	07:38:40
SdFatUtil.cpp	2585	13/07/2015	07:38:40
SdFatUtil.h	1462	13/07/2015	07:38:40
SdFile.cpp	3421	13/07/2015	07:38:40
SdFile.h	1463	13/07/2015	07:38:40
SdInfo.h	9146	13/07/2015	07:38:40
SdVolume.cpp	12223	13/07/2015	07:38:40
SdVolume.h	9210	13/07/2015	07:38:40
Servo.cpp	12193	13/07/2015	07:38:40
Servo.h	6002	13/07/2015	07:38:40
speed_lookuptable.h	12177	13/07/2015	07:38:40
stepper.cpp	41247	13/07/2015	07:38:40
stepper.h	4754	13/07/2015	07:38:40
temperature.cpp	51510	13/07/2015	07:38:40
temperature.h	5513	13/07/2015	07:38:40
thermistortables.h	39297	13/07/2015	07:38:40
ultralcd.cpp	53128	13/07/2015	07:38:40
ultralcd.h	4392	13/07/2015	07:38:40
ultralcd_implementation_hitachi_HD44780.h	32703	13/07/2015	07:38:40
ultralcd_st7920_u8glib_rrd.h	3953	13/07/2015	07:38:40
vector_3.cpp	4667	13/07/2015	07:38:40
vector_3.h	1696	13/07/2015	07:38:40
watchdog.cpp	1919	13/07/2015	07:38:40
watchdog.h	521	13/07/2015	07:38:40

AnHardt, 2015-07-17:
Unzip your Marlin in a fresh folder.
Open boards.h - look up the name for a RAMPS board. (usually `BOARD_RAMPS_13_EFB`)
Open `Configuration.h` - find the entry   `#define MOTHERBOARD` - replace the name behind   MOTHERBOARD with  `BOARD_RAMPS_13_EFB`
In  `Configuration.h` find the line where TEMP_SENSOR_0 is defined - replace the number behind TEMP_SENSOR_0 with the number of your thermistor. (see table above that entry) (if you do not want to connect one now, set to 0)
Save the Configuration.h
Compile.

Even less complicated things would be if you'd decide to start with the `Development` branch, instead of the  `Release` or `1.0.2`. There the changes above are the defaults.

bothal3, 2015-07-17:
Thank you very much. This sounds short and sweet.

Which "development" file should I download?

Zhang-Jia/Marlin-Development
robscar/Marlin-Development
Napalm1432/Marlin-for-Scara-arm
or
jakeBarwell/nativeMessegeBoard?

Or am I complete lost?

Regards.

Louis.

Date: Fri, 17 Jul 2015 02:52:13 -0700

AnHardt, 2015-07-17:
As you are here, how about https://github.com/MarlinFirmware/Marlin/archive/Development.zip
on https://github.com/MarlinFirmware/Marlin/tree/Development?

closed 2015-11-03 by AnHardt