MarlinFirmware/Marlin #2463: Doesn't go all the way up?

coalman321, 2015-07-17:


AnHardt, 2015-07-17:
More input please. 
My crystal ball says: "Home the axis" (G28)

coalman321, 2015-07-18:
believe it has something to do with the speed 
because the belts move normally however the z axis is a threaded rod.
it matters not if i home it. the software endstop kicks in at "150 mm" and where the z axis stops is not 150 mm its only about 50-75 mm.

thinkyhead, 2015-07-18:
@coalman321 When you move the Z axis by 20mm (from console or LCD) does it actually move 20mm?

AnHardt, 2015-07-18:
So try lower speeds, or if your z axis is running soothe.

 * http://calculator.josefprusa.cz

My (M5 dual stepper-single driver 12V) Prusa i3 is not faster than 3.5mm/second at Z.

coalman321, 2015-07-18:
no 150 m corresponds to about 50 mm
also where is the setting?

thinkyhead, 2015-07-18:
DEFAULT_STEPS_PER_MM (?) for Z. Calculate based on the number of steps per rotation of the motor (usually 200x16=3200) and the screw pitch (usually 0.8mm). With those values, it's 4000 steps per mm.

coalman321, 2015-07-18:
ok. will try when i get my new drivers in a week.
thermal paste is bad for them right?

pdbogen, 2015-07-19:
@coalman321 specifically, the setting is here: https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Configuration.h#L482

For starts, it sounds like you need to increase this value by a factor of three, so that the firmware will perform three times as many steps when you tell it to move.

Nprod, 2015-07-19:
@coalman321 Just use one of those little heatsinks with adhesive backing on one side. Also be aware that the driver chips are designed to have their thermal vias at the bottom, and thats where a heatsink works best: http://www.elecfreaks.com/wp-content/uploads/2014/06/24.jpg

coalman321, 2015-07-30:
ok got the drivers and the setting fixed. quick question, is there a way that i can tell the printer to act as as a laser cnc? i have a small 12 volt 2.5 watt laser to test with and i wanted it to understand heat up versus on when needed. i have it hooked up to the hotend any help would be appreciated.

pdbogen, 2015-07-30:
You should be able to use M42 to enable/disable the laser; just make sure the pin it's actually connected to is not assigned to anything (or, at least, not assigned to anything in the "sensitive pins" list: https://github.com/MarlinFirmware/Marlin/blob/f504949620d7724f58c45bf6860d0b1a33a5def2/Marlin/pins.h#L2967)

Wurstnase, 2015-07-30:
> You should be able to use M42 to enable/disable the laser;

But this is not in snyc with G1.

coalman321, 2015-07-30:
so i would need to move it to the auxilary area?

coalman321, 2015-07-30:
like aux 1?

pdbogen, 2015-07-30:
You wouldn't necessarily need to switch pins to use M42; you would just need to make sure the pin is *not* assigned as the heater pin you're using, so that you can control it with M42.

But what @Wurstnase points out is important; M42 is an immediate command, not a queued commands. If you queue up movements and then issue M42 to turn the laser on/off, it will take effect *immediately*, without respect to the status of the queued movements. You would need to issue an M400 or G4 P0 to wait until queued movements are complete before issuing an M42 to change the laser state (and it means the printer won't be able to plan/accelerate across laser state changes.)

coalman321, 2015-07-30:
laymans terms please im new to this whole make your own printer thing

AnHardt, 2015-07-30:
How about using the extruders step pulses? They are in sync and the number of pulses per mm is adjustable.

coalman321, 2015-07-30:
so hook the laser to 1a and 1b and tell the pulse rate to set to different depths? i like that one.

pdbogen, 2015-07-30:
> You wouldn't necessarily need to switch pins to use M42; you would just need to make sure the pin is not assigned as the heater pin you're using, so that you can control it with M42.

The M42 command has a list of pins it will refuse to change. Included by default in this list is basically every pin normally associated with a component of the printer, including the hot end heater pins. You said the laser is connected to the hot end, so I assume you connected it as though it were a heater. Thus, M42 will refuse to change the pin, unless you modify the appropriate pins_BOARD.h header so that it no longer believes the pin to be a heater pin. For example, if you connected the laser to the "second hot end" heater pin, you would want to set HEATER_1_PIN (0 is the first heater, 1 is the second, 2 is the third, etc.) to -1.

> M42 is an immediate command, not a queued commands. If you queue up movements and then issue M42 to turn the laser on/off, it will take effect immediately, without respect to the status of the queued movements. 
The printer normally accepts more movement commands than it can immediately execute. This is good, because it means the printer will usually have another movement to execute immediately after the current one is finished, and so it can plan for how quickly to accelerate by looking ahead into the queue.
The M42 command is not part of the queue, however, so it will take effect immediately. Suppose you wanted the laser to engrave a box, so you sent commands like this:
```
M42 Pn S255 ; set pin n to maximum
G1 X0 Y100 F300 ; move to (0,100)
G1 X100 Y100 F300 ; move to (100,100)
G1 X100 Y0 F300 ; move to (100,0)
G1 X0 Y0 F300 ; move back to (0,0)
M42 Pn S0 ; engraving finished, disable laser
```
Since the G1 commands get queued up, they will be *accepted* instantly but executed in order, one at a time. The M42 commands, however, are not queued up, so they are both *accepted* instantly and *executed* immediately, meaning that the last M42 command will take effect *before* the *first* G1 command is finished; so the laser will blink on, the printer will begin moving, and the laser will turn back off.

The way to fix this is to insert an M400 command, which is "wait for current moves to finish, like so:
```
M42 Pn S255 ; set pin n to maximum
G1 X0 Y100 F300 ; move to (0,100)
G1 X100 Y100 F300 ; move to (100,100)
G1 X100 Y0 F300 ; move to (100,0)
G1 X0 Y0 F300 ; move back to (0,0)
M400 ; Wait for moves to finish before disabling laser
M42 Pn S0 ; engraving finished, disable laser
```

...This has no issues with our simple G-code, but for a more complex print (where you might want to turn off the laser for "travel" moves), it will hamper performance somewhat, because the printer cannot plan moves/accelerations "across" that M400 boundary.

coalman321, 2015-07-31:
ok i wired it to behave when the extruder is active.. but when i tried the stepper it would not turn on do i need to change the steps?

pdbogen, 2015-07-31:
That's not a good idea. The only pins on an ordinary RepRap that can source 30W of current are the heater pins.

(It's possible the actual stepper pins from the actual stepper driver to the stepper motor could, but you're not going to be able to control them the way you want.)

pdbogen, 2015-07-31:
(Hm. Are you using a dedicated laser driver board with a TTL control pin, or are you hooking your diode directly up to the printer? If you're connecting the diode directly, you'll probably burn it out; none of the normal outputs on a printer are current-controlled, and any amount of overcurrent will destroy a laser diode. Ask me how I know this...)

coalman321, 2015-07-31:
i have a driver to control the current but it has no ttl control

pdbogen, 2015-07-31:
Well, as long as it's current-controlled, controlling power directly vs a dedicated TTL pin doesn't really make a difference; but it does mean that you'll need to ensure you're using something that can source sufficient current. Direct AVR pins _cannot,_ and the enable/step lines from a stepper driver are going to be low-current as well. You should really stick with heater lines, because those can definitely source enough current to drive your laser diode.

My recommendation remains the same: Reconfigure the firmware so that it doesn't use the heater pin as HEATER_n_PIN, which will let you control it with M42. You'll have to be careful to put an M400 before any M42 in your GCode, so that the printer finishes movements before changing laser state. This won't be optimal, but Marlin isn't designed for laser engraving, so you won't really be able to _get_ optimal. (You could open a separate Issue to ask for better support for laser diodes, but that won't be fulfilled for a long while, I suspect.)

coalman321, 2015-07-31:
what if i had a ttl driver? would i be able to do it then?

pdbogen, 2015-08-02:
The only thing a TTL driver gets you is the freedom to not worry about current draw. There's still no straightforward way to control a laser diode with Marlin as it is; anything you do right now would be a hack. (Ideally, you want to be able to specify a PWM value for the laser diode alongside as part of a move command; but Marlin doesn't accept such a value. It doesn't sound like you can do this yourself right now, so you'd want to open a new Issue as a feature request; but it'll likely be a while before it gets done.)

AnHardt, 2015-08-02:
Again.
When you like to laser- ___cut___ (not picture engrave) you'd like to have a constant energy per mm.
The number of stepper pulses for the extruder is then directly proportional to the way and adjustable by the amount of extruded material.
Routing the extruder step pulses to a MOSFET driven output (like the extruder heater) can be done via the pin-file.

coalman321, 2015-08-02:
another question more related to the original topic, when i tell my extruder to run, instead of going smoothly it pulsates. is that something i can fix?

thinkyhead, 2015-08-04:
@coalman321 Interesting. Could the current be too high to that motor?

coalman321, 2015-08-04:
i dont think so. im using the printr bot direct drive extruder and and the preferred steps per mm for the exturder itself. im even using the same psu which only allows for 6 amps so i dont think so but i could be wrong.

coalman321, 2015-08-05:
how do i get to the pins.h in arduino?

AnHardt, 2015-08-05:
For the newer Arduino IDEs the drop-down tab selector is scrollable. But an external editor is the first choice.

thinkyhead, 2015-08-12:
> …even using the same PSU…

@coalman321 The current to the motors is tuned by turning a tiny potentiometer on the stepper driver module, or on the electronics board. With some boards the current can be tuned through code. If the current to an individual motor is tuned too low or too high all kinds of interesting things can happen.

coalman321, 2015-09-05:
ok well i ajusted the pot on the driver and the "jackhammer" stopped

coalman321, 2015-09-22:
got it printing great now... time to go further, can marlin support a cnc style machine at the same time (without changing or updating firmware beforehand)?

coalman321, 2015-09-22:
ok scratch that for now.... back to the laser, i have switched the pins 9 (the heater pin) and 36 (the step pin) it compiled just fine until i change the extruder amount... 
it gave me this error first
https://gist.github.com/736725f066938bbc2eeb.git

then it gave me this when i changed it to one...

https://gist.github.com/c5b57963970a9fa9f552.git

am i doing something wrong here?

clefranc, 2015-09-22:
@coalman321 Your links doesn't work!

coalman321, 2015-09-23:
whoops,
here is error 1

thermistortables.h:1094: error: 'temptable_999' was not declared in this scope
 #define _TT_NAME(_N) temptable_ ## _N
                      ^
temperature.cpp:131:49: note: in definition of macro 'ARRAY_BY_EXTRUDERS'
   # define ARRAY_BY_EXTRUDERS(v1, v2, v3) { v1, v2 }
                                                 ^
thermistortables.h:1095:21: note: in expansion of macro '_TT_NAME'
 #define TT_NAME(_N) _TT_NAME(_N)
                     ^
thermistortables.h:1121:29: note: in expansion of macro 'TT_NAME'
 # define HEATER_1_TEMPTABLE TT_NAME(THERMISTORHEATER_1)
                             ^
temperature.cpp:150:101: note: in expansion of macro 'HEATER_1_TEMPTABLE'
   static void *heater_ttbl_map[EXTRUDERS] = ARRAY_BY_EXTRUDERS( (void *)HEATER_0_TEMPTABLE, (void *)HEATER_1_TEMPTABLE, (void *)HEATER_2_TEMPTABLE );
                                                                                                     ^
thermistortables.h:1094: error: 'temptable_999' was not declared in this scope
 #define _TT_NAME(_N) temptable_ ## _N
                      ^
temperature.cpp:131:49: note: in definition of macro 'ARRAY_BY_EXTRUDERS'
   # define ARRAY_BY_EXTRUDERS(v1, v2, v3) { v1, v2 }
                                                 ^
thermistortables.h:1095:21: note: in expansion of macro '_TT_NAME'
 #define TT_NAME(_N) _TT_NAME(_N)
                     ^
thermistortables.h:1121:29: note: in expansion of macro 'TT_NAME'
 # define HEATER_1_TEMPTABLE TT_NAME(THERMISTORHEATER_1)
                             ^
thermistortables.h:1122:41: note: in expansion of macro 'HEATER_1_TEMPTABLE'
 # define HEATER_1_TEMPTABLE_LEN (sizeof(HEATER_1_TEMPTABLE)/sizeof(*HEATER_1_TEMPTABLE))
                                         ^
temperature.cpp:151:94: note: in expansion of macro 'HEATER_1_TEMPTABLE_LEN'
   static uint8_t heater_ttbllen_map[EXTRUDERS] = ARRAY_BY_EXTRUDERS( HEATER_0_TEMPTABLE_LEN, HEATER_1_TEMPTABLE_LEN, HEATER_2_TEMPTABLE_LEN );
                                                                                              ^
thermistortables.h:1094: error: 'temptable_999' was not declared in this scope
 #define _TT_NAME(_N) temptable_ ## _N
                      ^
temperature.cpp:131:49: note: in definition of macro 'ARRAY_BY_EXTRUDERS'
   # define ARRAY_BY_EXTRUDERS(v1, v2, v3) { v1, v2 }
                                                 ^
thermistortables.h:1095:21: note: in expansion of macro '_TT_NAME'
 #define TT_NAME(_N) _TT_NAME(_N)
                     ^
thermistortables.h:1121:29: note: in expansion of macro 'TT_NAME'
 # define HEATER_1_TEMPTABLE TT_NAME(THERMISTORHEATER_1)
                             ^
thermistortables.h:1122:69: note: in expansion of macro 'HEATER_1_TEMPTABLE'
 # define HEATER_1_TEMPTABLE_LEN (sizeof(HEATER_1_TEMPTABLE)/sizeof(*HEATER_1_TEMPTABLE))
                                                                     ^
temperature.cpp:151:94: note: in expansion of macro 'HEATER_1_TEMPTABLE_LEN'
   static uint8_t heater_ttbllen_map[EXTRUDERS] = ARRAY_BY_EXTRUDERS( HEATER_0_TEMPTABLE_LEN, HEATER_1_TEMPTABLE_LEN, HEATER_2_TEMPTABLE_LEN );
                                                                                              ^
'temptable_999' was not declared in this scope

and error 2

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:665: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:665: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 141 [ ivtmp.168 ])) [4 MEM[base: _101, offset: 0B]+0 S4 A8])) temperature.cpp:449 99 {*movsf}
     (expr_list:REG_INC (reg:HI 141 [ ivtmp.168 ])
        (nil)))
temperature.cpp:665: confused by earlier errors, bailing out
unable to find a register to spill in class 'POINTER_REGS'

coalman321, 2015-10-01:
anybody?

coalman321, 2015-10-02:
well no answer 
closing

closed 2015-10-02 by coalman321