MarlinFirmware/Marlin #2466: Multi-extruder support broken

pdbogen, 2015-07-18:
Howdy!

Firmware multi-extruder support appears to be broken. For testing purposes, I configured a Melzi board to think it has two extruders that all share the same pins. (This requires a little hackery when setting temperature, but otherwise appears to work fine.)

A simple G-code produces obviously incorrect movements. Basic steps to reproduce:
1. Configure your printer with:
```
#define EXTRUDERS 2
#define TEMP_SENSOR_1 TEMP_SENSOR_0
#define E1_STEP_PIN E0_STEP_PIN
#define E1_DIR_PIN E0_DIR_PIN
#define HEATER_1_PIN HEATER_0_PIN
#define E1_ENABLE_PIN E0_ENABLE_PIN
#define TEMP_1_PIN TEMP_0_PIN
```

IOW, tell the printer that extruder 1 is connected to the same pins as extruder 0. This appears to work fine, and lets you "simulate" multiextruder support.

Now, print (something like the attached one-layer print), and on my system the behavior becomes unpredictable. Usually the resulting behavior is a mixture of consuming planned movement blocks but only moving the extruder, not X, Y, or Z; but sometimes it executes moves normally (at this point, usually adjacent to an endstop or off the bed.). Sometimes the first several moves will be executed normally and then it will start eating blocks.

I've tried to dig into this but I haven't come up with anything concrete. I have the following clues:
* Moves appear to be _planned_ normally, so the problem _seems_ to be somewhere in execution
* Stripping extruder movements from the GCode "prints" fine, regardless of the selected extruder. This makes me thing the problem is in the area of line 650 in stepper.cpp

I'm happy to answer any questions or do any troubleshooting I can to help figure out the issue.

pdbogen, 2015-07-25:
Just a quick note to confirm that I'm still able to reproduce this with today's Development branch. I'm going to dig in and try to figure out what the problem is.

pdbogen, 2015-07-26:
I think I've narrowed it down somewhat. I disabled everything in gcode_T in Marlin_main.cpp related to switching heads _except_ the change to active_extruder, and it printed fine.

I'm not executing T* commands with moves, so prepare_move shouldn't be at issue. The most complicated thing left is sync_plan_position, so I'm going to try disabling just that.

pdbogen, 2015-07-26:
Progress of a sort! I've found that 89a304fd9831bfc66bd23ef2e708e3887f2bdbf1 works. This means I can at least bisect.

pdbogen, 2015-07-26:
Indeed, disabling DISABLE_INACTIVE_EXTRUDER works around the problematic code. It still isn't clear to me _why_ this is causing the problem I observed, however, and so it's not clear how to fix the underlying functionality.

thinkyhead, 2015-07-27:
> It still isn't clear to me why this is causing the problem I observed

Because when the "other extruder" is inactive the firmware will disable it. Since the "first extruder" is also the "second extruder" the "first extruder" gets disabled when the "second extruder" is idle, and the "second extruder" gets disabled when the "first extruder" is idle.

To get this kind of multi-extruder to work properly, you will need to implement it as a feature with a configuration option and not just by hacking pins. The configuration option can then be used to alter the behavior of `DISABLE_INACTIVE_EXTRUDER` and any other code that would treat the two extruders as separate and distinct.

pdbogen, 2015-07-27:
Your assessment is correct, but I believe there is something else wrong.

The problem I'm seeing could conceivably be related to the printer spending too much time enabling/disabling extruders; however, when I set up Marlin "in practice" to use multiple extruders, they don't share pins, and so the theoretical overlapping enable/disable stuff isn't an issue; but the behavior is the same, i.e., lots of missing movements.

When the pins do overlap, I suspect that even though the enable/disable stuff is overlapping, the printer isn't spending enough time for it to be an issue. enable_e0 and disable_e0 are only one instruction each for non-TMC non-L6470 steppers; and this only happens as part of planning, which is relatively rare.

(That said, reading the code, I would expect that printing with an E1 that shares pins with E0 would never result in extruder movements, since E1 is enabled _and then_ E0 is disabled. But whatever, because the problem I was seeing _wasn't_ missing extruder movements, it was missing Cartesian movements; i.e., the print head would sit in one place while the extruder was running and the printer was claiming to be executing planned moves.)

I don't know why the code in question would be causing this problem; I observed that it was responsible empirically, via git bisect; 29e7639933e29feaf58bc657e80c47e88716b02b doesn't work with the scenario I've described above, and the previous commit (5a1832403043c0a6ccd6988e1804bb988e9e7086) does, hacked pins and all; and then disabling meant that my X/Y/Z motors were moving everything around properly even on the newest Development commit.

pdbogen, 2015-07-27:
(oh, n.b., the "overlapping pins" was just so that the printer would run with reasonable behavior thinking it had two extruders without the electronics/mechanical components necessary to support it; I never expected it to _actually_ print, but it should still move in XYZ correctly; it doesn't, and that's the same problem as in my 'real life' setup where the second extruder is on another board, controlled via serial, with sane pin redirections and configuration options; but it was easier to bisect with a config hack than with the full slave-board changes.)

thinkyhead, 2015-07-27:
> the print head would sit in one place while the extruder was running

Ew, that is different. And it still occurs even when pins don't overlap, eh?

Have you done some controlled tests, such as generating a simplified GCode file with movements in a circle, so you can observe the issue for a while? You might be able to isolate how and whether E movements are involved, and where the movements are being "dropped."

The way you describe it, it sounds like a lower level issue, such as a pins conflict, so that signals to the stepper motors are being lost, rather than a planner issue where movements are not being sent to the steppers. These are really the only two levels at which I can imagine these symptoms are being produced. The higher-level code that sends movements to the planner has no logical instance where movements will simply not be sent on to the planner.

So I think we need to focus on the planner-to-stepper and stepper-to-signals levels to isolate the cause.

pdbogen, 2015-07-27:
I've trimmed down GCode to a relatively simple case (initially, in the belief that it was a problem with the gcode, thus as part of trying to isolate the hypothesized problematic line). My current test case is a pretty simple print, and just the first layer of it, at that; however, the problem shows up within the first move, really, after heating up the extruder; it either moves to the active area and begins printing, or it doesn't execute that move, sits in place with the extruder running, then sporadically executes some moves (usually running into the endstop at that point, because Marlin has updated its internal position state but not its external.)

I've noticed that the problem only occurs when there are moves involving an extruder; removing the E* values from the moves manually and the print "works" properly.

I've added debugging code to the stepper ISR to show me the move-to-be-executed when it comes off the queue, and a properly-behaving firmware's moves are not substantially different from an improperly-behaving firmware; i.e., the problem is definitely not in the planner, at least as far as putting the movements onto the queue.

(Actually, it strikes me that the _planner_ enabling/disabling extruders is problematic. I'd definitely support changes to make the planner more "functional"- in the "functional programming" sense- and reduce side-effects like extruder enable/disable.)

Believe me, I spent hours reading over source code (focused on the stepper ISR mostly, but also the planning code) and trying different debugging/hypotheses before I eventually did a bisect. Luckily the change I landed on was a simple one. I'll spend _some_ time trying to isolate the problem with that code; but for now I have other hurdles to overcome (MarlinSerial is in my sights right now.)

thinkyhead, 2015-07-29:
@pdbogen Here's an experiment. Set the following settings to `false`:
```cpp
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
#define DISABLE_E false
```
If they are already `false` in your setup, then I will look elsewhere…

pdbogen, 2015-07-29:
https://github.com/pdbogen/Marlin/blob/newmarlin/Marlin/Configuration.h#L360 is the actual config I run with, for your reference. Those defines are already defined as false.

thinkyhead, 2015-07-31:
@pdbogen Ok, good. With those settings the steppers will stay active. So there should be no signals to the stepper ENABLE pins during printing.

We'll have to start looking elsewhere. If the effects are 100% reproducible, we might be able to isolate the specific case where moves are being dropped (logically).

As another (weird) thing to test, what kind of result do you see if you run the GCode with E moves left in, but simply disconnect the E stepper motor?

pdbogen, 2015-07-31:
They're 100% reproducible. I haven't done anything objective to check, but visually the malbehavior appears to be the same each time (i.e., which moves get skipped vs which get executed.) I'll come up with an as-short-as-possible GCode that demonstrates the issue. (And possibly convince my roommate to let me try it out on his printer with the same two-extruders-as-one hack.)

thinkyhead, 2015-08-03:
I'm doing a detailed diff between those two points in history, pr673 and pr1133. I'll let you know if anything obvious jumps out at me.

thinkyhead, 2015-08-03:
Have you also done any tests with only minimal options? For example, turn off thermal protection, watchdog, and any other optional features that you have enabled in your configuration. That will also help isolate if any features are interacting badly with movement or timing.

pdbogen, 2015-08-03:
Oh, sorry for not being clear. I did a git bisect and found exactly the change that broke this behavior:

```
29e7639933e29feaf58bc657e80c47e88716b02b is the first bad commit
commit 29e7639933e29feaf58bc657e80c47e88716b02b
Author: Dim3nsioneer <Dim3nsioneer@gmx.ch>
Date:   Mon Jun 2 08:13:09 2014 +0200

    Add switch unused feeder(s) off

    Having the non-active feeder motors powered on all the time is not
    necessary. A feature to deactivate the unused feeder motors has been
    implemented. The feature is enabled on default but can be switched off
    in the configuration.
```

Every change after this exhibits the "bad" behavior; every change before this works (even with extruder pin hacks.)

thinkyhead, 2015-08-03:
So, the associated feature is `DISABLE_INACTIVE_EXTRUDER`. Are you using that, and does turning it off help?

thinkyhead, 2015-08-04:
Well then, the problem has to be here someplace:
```cpp
  // Enable extruder(s)
  if (block->steps[E_AXIS]) {
    if (DISABLE_INACTIVE_EXTRUDER) { //enable only selected extruder

      for (int i=0; i<EXTRUDERS; i++)
        if (g_uc_extruder_last_move[i] > 0) g_uc_extruder_last_move[i]--;
      
      switch(extruder) {
        case 0:
          enable_e0();
          g_uc_extruder_last_move[0] = BLOCK_BUFFER_SIZE * 2;
          #if EXTRUDERS > 1
            if (g_uc_extruder_last_move[1] == 0) disable_e1();
            #if EXTRUDERS > 2
              if (g_uc_extruder_last_move[2] == 0) disable_e2();
              #if EXTRUDERS > 3
                if (g_uc_extruder_last_move[3] == 0) disable_e3();
              #endif
            #endif
          #endif
        break;
        #if EXTRUDERS > 1
          case 1:
            enable_e1();
            g_uc_extruder_last_move[1] = BLOCK_BUFFER_SIZE * 2;
            if (g_uc_extruder_last_move[0] == 0) disable_e0();
            #if EXTRUDERS > 2
              if (g_uc_extruder_last_move[2] == 0) disable_e2();
              #if EXTRUDERS > 3
                if (g_uc_extruder_last_move[3] == 0) disable_e3();
              #endif
            #endif
          break;
          #if EXTRUDERS > 2
            case 2:
              enable_e2();
              g_uc_extruder_last_move[2] = BLOCK_BUFFER_SIZE * 2;
              if (g_uc_extruder_last_move[0] == 0) disable_e0();
              if (g_uc_extruder_last_move[1] == 0) disable_e1();
              #if EXTRUDERS > 3
                if (g_uc_extruder_last_move[3] == 0) disable_e3();
              #endif
            break;
            #if EXTRUDERS > 3
              case 3:
                enable_e3();
                g_uc_extruder_last_move[3] = BLOCK_BUFFER_SIZE * 2;
                if (g_uc_extruder_last_move[0] == 0) disable_e0();
                if (g_uc_extruder_last_move[1] == 0) disable_e1();
                if (g_uc_extruder_last_move[2] == 0) disable_e2();
              break;
            #endif // EXTRUDERS > 3
          #endif // EXTRUDERS > 2
        #endif // EXTRUDERS > 1
      }
    }
    else { // enable all
      enable_e0();
      enable_e1();
      enable_e2();
      enable_e3();
    }
  }
```

pdbogen, 2015-08-04:
I _was_ using DISABLE_INACTIVE_EXTRUDER, and disabling it _does_ fix the problem on recent versions of Development.

I agree that's the location causing the problem. It may indeed be a timing issue; I can try reordering the enables/disables; it'll be interesting to see if that helps.

AnHardt, 2015-08-04:
Autsch. Why is in
```
    if (DISABLE_INACTIVE_EXTRUDER) { //enable only selected extruder
```
the `if`` not an `#if`?

thinkyhead, 2015-08-04:
@AnHardt It's just been that way for a long while, no particular reason. We should change it. There's also no reason it has to be a `true` / `false` setting either.

As for its effect on this issue, the compiler should remove the code either way, whether it's `#if` or `if(...)`.

thinkyhead, 2015-08-04:
> it'll be interesting to see if that helps

Yes! Honestly, that's all I's got.

Dim3nsioneer, 2015-08-04:
Throw that code out. End the discussion. Please.

thinkyhead, 2015-08-08:
@Dim3nsioneer Sorry, are we boring you? We still haven't solved this issue, and "that code" is ancillary to the discussion.

Dim3nsioneer, 2015-08-08:
@thinkyhead : We might have gotten each other wrong. I had the impression you were hesitating to delete a piece of code I contributed and which seems to cause serious problems. I just wanted to state, that it would be absolutely ok for me if the DISABLE_INACTIVE_EXTRUDER feature would be deleted for the time being (or declare it as non-working, whatever...) in order to have a stable, multi-extruder capable version of Marlin (i.e. it's better to have a working code with one feature less than the other way round).

thinkyhead, 2015-08-12:
@Dim3nsioneer Thank you for clarifying. It helps to bring some context to new discussions.

Can users of multi-extruder setups just leave `DISABLE_INACTIVE_EXTRUDER` commented out, so all extruder motors will stay active? The problem is that there really isn't any formal multi-extruder feature, so already the solution has been kind of a hack. Turning off `DISABLE_INACTIVE_EXTRUDER` would just be another necessary part of that hack approach. In other scenarios, as far as I am aware, `DISABLE_INACTIVE_EXTRUDER` is fine.

closed 2015-11-24 by AnHardt