MarlinFirmware/Marlin #2472: Bye-bye Makefile?

thinkyhead, 2015-07-19:
With the latest Arduino tools the `Makefile` is no longer required, but some users may still value it for automated builds in exotic environments. Should the `Makefile` go away at this point, or should we continue to maintain it as a reference and as a utility?

pdbogen, 2015-07-19:
When you say "latest Arduino tools", are you referring to the Arduino GUI or is there a CLI interface? I actually replace the length Makefile shipped with Marlin with a much shorter one <https://github.com/pdbogen/Marlin/blob/newmarlin/Marlin/Makefile> based on arduino-mk, since I prefer not to use the arduino IDE.

thinkyhead, 2015-07-19:
@pdbogen I'm referring to the Arduino IDE / GUI.

Your shorter `Makefile` is very nice. So, it depends on having http://dev.ed.am/arduino-mk installed…? I'll have to check that project out and see what it's doing to simplify things.

We don't really know how many users are actually using the `make` method to do builds at this point, or how many of those users will want to keep using it for the next release of Marlin, so this is a kind of poll to figure out how useful it still is. After all, most of us just update our settings, flash each printer once, and move on. Do you find it extra convenient to have a `Makefile` for your workflow?

Roxy-3DPrintBoard, 2015-07-19:
Let's hold off removing it for just a little while.   The Makefile approach is useful for dealing with the name space issues when trying to separate out the G28 and G29 files.   I suspect it isn't needed often.  But right now I'm finding it helpful.

Wackerbarth, 2015-07-19:
@pdbogen - When you say that you "prefer not to use the arduino IDE", is it the build rules and/or compilers or is it the User Interface?  I ask because, the two aspects are somewhat separate. In particular, just as you can do with Xcode on Mac OS X, you can call the build portion from the command line without using the User Interface.

There are a number of build systems around and many of them COULD be used, However, I feel that we need to avoid the duplication and effort to keep things synchronized.

Wackerbarth, 2015-07-19:
@Roxy-3DPrintBoard -- To address the namespace issue, we need to move Marlin into a library structure as has been done with grbl, Cosa, etc.  I see little point in attempting to change things with the current file structure.

pdbogen, 2015-07-19:
@thinkyhead https://github.com/sudar/Arduino-Makefile is the correct link.

@Wackerbarth Mostly the UI itself; though also, the lack of incremental rebuilds is problematic. It looks like, at some point, Arduino added an official command-line tool <https://github.com/arduino/Arduino/blob/ide-1.5.x/build/shared/manpage.adoc>; I don't know if it supports incremental building. It might, in any case, be easy to write a Makefile wrapper to invoke that tool.

thinkyhead, 2015-07-19:
Our Travis CI `.travis.yml` uses `~/bin/arduino` as a matter of fact.

sorki, 2015-07-20:
-1

Makefile is the only sane way to build marlin for me. I've did some improvements to it in latest pull request and plan to do some more.

Rewriting Makefile to use arduino-mk sounds like a good idea though current makefile works fine now.

Wackerbarth, 2015-07-20:
@sorki  -- Well, IMHO, using arduino-mk is a bad idea because it is unnecessary, redundant, and just another piece for which we would need to maintain compatibility.

Because of our user base, I feel that we MUST support the Arduino IDE. So, I question WHY you feel that you need more than a trivial makefile that performs the build using the `arduino` command line invocation of the IDE.

BTW, your really should include the build hook call to `generate_version_header_for_marlin` as a part of your makefile

sorki, 2015-07-20:
I don't use **any** IDE. There's no need for arduino IDE to build and flash Marlin or any other wiring based software.

Wackerbarth, 2015-07-20:
@sorki - By that argument, there is also no need for a makefile. We ware building complete operating systems before the original `make` was written.
I'm not asking you to consider the IDE infrastructure as anything more than a complex command line compiler.
`arduino`, `make`, `ino`, `ant`, `arduino-mk`, `cmake`, etc. are alternatives.

But each accomplishes the same, or at least similar, result using a description file that is particular to that tool.

As Marlin Developers, if is wasteful of our effort to attempt to maintain multiple alternatives.
And, as your PR's indicate, things do get out of sync.

sorki, 2015-07-20:
This is about convenience for CLI users, not about buildsystems and alternatives. Current makefile works fine and I'm willing to contribute patches to keep it usable and in sync with current distributions. So please keep the current makefile as an alternative to using arduino IDE.

thawkins, 2015-07-20:
I missed this conversation

1. Most distros struggle to provide anything more than 1.0.6 at this point, making Marlin dependent on a version that has to be hand installed is problematic. 

2. Make is well understood simple technology. 

3. The Arduino IDE is garbage for anything more than half a dozen files, even on the latest version its impossible to open all of the files in the project because the "more tabs" drop down link does not scroll. On  an average  laptop screen only about 60% of the files are reachable. It does not support project wide search, it has very very little coding support.  It may be widely accessible, but its far too far dumbed down to be usefull. 

4. Most other firmwares dont use the arduino IDE, sailfish uses "scons",  reprapfirmware can use arduino, but uses eclipse with avr plugin, repetier uses codeblocks.... 

I use cLIon,  with the makefile. 

If you are going to do anything, then i woukd suggest arduino-cmake ,  it can generate any kind of make system, including makefiles and arduino project files from a single source. And most ide's can deal with it easily.

Wackerbarth, 2015-07-20:
> I missed this conversation
> Most distros struggle to provide anything more than 1.0.6 at this point, making Marlin dependent on a version that has to be hand installed is problematic.
But it is equally problematic to use ANY other system such as `arduino-cmake` suggested below.
> Make is well understood simple technology.
I'm not such that I would agree. Most users have no idea how/why to use features of a makefile. And then there is the issue of “which variant of `make`?” — But, generally, I will agree that it is understandable to most of the developers who would be altering the makefile.
> The Arduino IDE is garbage for anything more than half a dozen files
I am not suggesting the the IDE UI be used for complex projects. Rather, it is a vehicle for a set of build tools which can be invoked from the command line. (And that invocation can be pre-set in the IDE of your choice.)  And the Arduino IDE is quite adequate for the “casual” user who wants to do no more than tweak his Configuration, recompile, and flash the firmware to his printer.

pdbogen, 2015-07-20:
It's probably a great property if the documented standard approach for building the project is the same used by CI, i.e., the official arduino CLI tool. (Even if a Makefile wrapper for it is shipped, too; which should be simple, if it really just amounts to a set of indirections to the arduino tool.)

oysteinkrog, 2015-07-20:
I want a makefile, being dependent on the Arduino IDE would be really horrible.
I don't really understand why anyone would want to remove it, as long as it is being maintained?

Wackerbarth, 2015-07-20:
"... as long as it is being maintained?"  Therein lies the issue.  The current makefile does not produce the same result. It needs "maintaining", on a continuing basis.

thawkins, 2015-07-21:
http://platformio.org/#!/

This looks cool

Supports arduino, make and cLion, plus eclipse cdt and netbeans

thawkins, 2015-07-21:
I am maintaining it, as are others, since we use it.

daid, 2015-07-21:
@Wackerbarth different Arduino versions also produce different results. And the Arduino environment produces different results depending on which OS you are running. And some cases the Makefile produces BETTER results as it does not link in some unused code.
"It's different" is a horrible argument.

But something less maintenance intensive wouldn't hurt. (Or maybe a small update to atleast use the ${wildcard} construct as that mimics the Arduino environment a bit better)

Wackerbarth, 2015-07-21:
@daid -- No, "it's different" IS the point. We need to have ONE way to build the STANDARD and expect everyone who seeks support to be working against that standard. If your makefile builds from a different set of sources, then that should not be our issue.  I expect you, and the rest of us who are sufficiently sophisticated, to resolve issues and present analysis and recommendations rather than questioning "Why doesn't ...XXX ... work?"
Distributing a makefile that isn't up-to-date just encourages the generation of combinations that we don't want to support.

As for the "better results", we can also achieve those using the `arduino` build tool. But, to accomplish that, we need to refactor the source into an appropriate library structure as has been done on other projects. That effort is "on hold" only because of the pressure to convert the current Development into something releasable.

Roxy-3DPrintBoard, 2015-07-21:
> Wackerbarth commented :+1:
> As for the "better results", we can also achieve those using the arduino build tool. But, to accomplish that, we need to refactor the source into an appropriate library structure as has been done on other projects. That effort is "on hold" only because of the pressure to convert the current Development into something releasable.

Wackerbarth, can you provide some details on how the code would be sliced and partitioned into different libraries?   How many different libraries are you thinking?

Wackerbarth, 2015-07-21:
@Roxy-3DPrintBoard -- I haven't given the exact breakdown too much thought. However, WRT the wy that Arduino mucks with sketches, the idea is to take essentially everything out of the sketch domain and move it into libraries. Under those conditions, I don't think that there is anything that you would do with a makefile that we cannot do using (the command line version of) the IDE as the generator.

But, as to breaking things up, I would start by splitting out the display, the motion control, and the temperature control.

Roxy-3DPrintBoard, 2015-07-21:
Is your thinking each of those things (Display, Motion Control, Temperature Control, what ever else make sense) would be separate projects, and those libraries would get modified outside of the main Marlin project?

It would be nice if any change in any sub-project automatically got rebuilt into the main project without assistance from the person trying to build the firmware.

Wackerbarth, 2015-07-22:
No, I don’t see them as separate projects. Rather separate components of one overall project.

However, using git submodules, they could be split into separate (virtual if not physical) repositories if there is any advantage in doing so.

But then you have to be careful about keeping modifications in sync across them.

> On Jul 21, 2015, at 5:22 PM, Roxy-3DPrintBoard <notifications@github.com> wrote:
> Is your thinking each of those things (Display, Motion Control, Temperature Control, what ever else make sense) would be separate projects, and those libraries would get modified outside of the main Marlin project?

daid, 2015-07-22:
@Wackerbarth even if you remove the makefile, you'll still end up with different build results as I explained. So while having different build results is an potential for issues (I see that) removing the Makefile does not solve that.

Also, github submodules are horrible. Don't even think along that route, it's horrible maintaince and user wise, as it requires tons of user-interaction.

Wackerbarth, 2015-07-22:
> On Jul 22, 2015, at 1:28 AM, daid <notifications@github.com> wrote:
> 
> @Wackerbarth <https://github.com/Wackerbarth> even if you remove the makefile, you'll still end up with different build results as I explained.
> 
And, as I explained, THOSE differences are something that we will need to address (when they are an issue).
However, the Arduino IDE (build tool portion), with its recipe patterns, is effectively a makefile generator and compiler package.

The issue with having a separate makefile is that it REQUIRES that someone keep it in sync with changes to the source file structure. Otherwise, it is not even attempting to compile the same “program”.  As Marlin Developers, distributing a separate makefile doesn’t create value to us. It just adds an additional maintenance burden.
> So while having different build results is an potential for issues (I see that) removing the Makefile does not solve that.
> 
It doesn’t SOLVE the entire issue. But removing it was reduce the number of potential sources for those issues.
> Also, github submodules are horrible. Don't even think along that route, it's horrible maintaince and user wise, as it requires tons of user-interaction.
> 
They do address the issue of including separately maintained subsystems. Although using them does require user interaction, I find it no worse than attempting to include source archives for those subsystems.

Note that I was not advocating have separately maintained subsystems at all. I explicitly said “COULD … IF ..."

thawkins, 2015-07-22:
I don't want to see the Makefile removed, I'm willing to maintain it if
nobody else wants to, the arduino build system is a non starter for me as
my platform only supports versions up to 1.0.6 unless you start messing
around with manual installs.

Roxy-3DPrintBoard, 2015-07-22:
Nobody is asking this question:    Would we be better of just plain switching to the Makefile and quit using the IDE ?   The only good thing the IDE does for me is upload the firmware image to the AVR.

daid, 2015-07-22:
avrdude can upload to the AVR just fine (and that's actually what the IDE is already calling)

thawkins, 2015-07-22:
Note also the teensyduino drivers required to get Marlin into sanguinololus
and printrboards, does not work with anything greater than 1.0.6 as far as
we can tell.

thawkins, 2015-07-22:
Gets my vote...

sorki, 2015-07-22:
@thawkins

Why? I've never had to install any drivers to flash sanguinololus - just avrdude.

Wackerbarth, 2015-07-22:
> On Jul 22, 2015, at 8:47 AM, Tim Hawkins <notifications@github.com> wrote:

>  as my platform only supports versions up to 1.0.6 unless you start messing
> around with manual installs.

Well, then you should consider using an upgraded infrastructure within the IDE.

Note that I do not use the “Arduino” platform, but define a “marlin" platform instead.

If you need particular “tools” to do uploads, the 1.6.x IDE can handle them. You just need to install the tools in your IDE configuration once. Then the IDE installs will work.

thawkins, 2015-07-22:
sorry, wrong one, i was thinking of the geeetech board, which looks just
like a Sanguinololu, but its just a red printrboard clone . anyhow the same
comment stands re the printrboard.

Note I binned two printrboards because they where an absolute pain in the
ass to work with.

I should know this I have two sangiunololus sitting in my to be done box,
and I can see now they have a {probably clone) FTDI chip on them.

thawkins, 2015-07-22:
Both Fedora 22 and the latest version of Ubuntu install 1.0.6 from their
repos. Thats what I mean by platform.

for any  Linux user that does am "apt-get install arduino" or "yum install
arduino" will get 1.0.6.

Wurstnase, 2015-07-22:
Too comfortable to download the latest IDE from arduino.cc? There is no installer for that. Just download the zip-file, unpack, maybe an chmod +x for the arduino start-file. Not really much todo.

Maintaining a makefile is much more complicated? :)

thinkyhead, 2015-07-22:
How do other printer firmware projects arrange their files and manage their build options? How important is it to be able to produce binary-identical builds for potential troubleshooting purposes? Have some other Arduino-based firmware projects figured out a way to use subfolders within AVR projects, perhaps by separating out code into libraries within the project?

We're always looking at ways to be able to use sub-folders, at least. If it means having other components as libraries, each with a separate repository, that could work, but it might be overkill for a project of such a modest size as Marlin.

So it sounds like the Makefile may still have some usefulness, and @thawkins is willing to keep it up-to-date. #2477 and #2479 both make changes to the Makefile, some of which I'm not sure about, so please (anyone in this thread) have a look at those and see what you think.

Roxy-3DPrintBoard, 2015-07-22:
> Have some other Arduino-based firmware projects figured out a way to use subfolders within AVR projects, perhaps by separating out code into libraries within the project?

> We're always looking at ways to be able to use sub-folders, at least.

If we could just isolate different sections of the code into their own sub-folders, that would give us a good start on cleaning things up.   I have not tried to do anything with sub-folders within Arduino, but my guess is that would go a long way to cleaning the code up.

Wackerbarth, 2015-07-22:
@thinkyhead -- As I previously pointed out, both Cosa and grbl have structure to the organization of their files. We really need to be moving in that direction. But I have held off because it creates another significant reorganization and will require that users upgrade to a newer version of the IDE.

Wackerbarth, 2015-07-23:
@sorki -- Your Makefile does not work with the Arduino 1.5 or 1.6 distributions.  Since you do REQUIRE the IDE to be present (to supply the core source files), you need to either have it work with ALL of the distributions or move it into one of the distribution dependent support areas.

IMHO, this is an additional verification that we should remove the Makefile entirely.

thawkins, 2015-07-23:
The Makefile is one single file, which those of us that use it are quite
willing and capable to hack on it, the Makefile as it stands I believe
works with all versions if you set the options/paths right. It appears that
the only people arguing for its removal are those folks who don't use it.

If the Makefile stops working then those of us that use it will fix it. As
I did when I could not get my project to compile with the RepRap Discount
Full graphic Display.

I don't see how it is any risk or burden to the people who don't use it.
This whole issue only arose because we DID maintain it. My original PR only
had two commits in it, one for the U8Glib and buzzer.cpp and one for a
small typo. Things got confused when a second PR appeared with the same
fixes done differently, plus a whole load of other stuff.

If you don't use the Makefile then by all means screw up your eyes and
pretend you cant see it. But don't start wrecking other peoples workflows
because you don't like the idea of something.

To be truth full the whole process of submitting half a dozen lines of
corrections to a file that people claim is not used or not useful has left
me with a bad taste in my mouth, and i'm starting to wonder if its worth
making any further contributions. I was using this as a means of testing
the mechanism of submitting code changes before becoming more engaged with
the project, i am a heavy user of marlin, and I use sailfish too several
other of my bots, i decided to get more engaged with both projects as my
free time opens up. But now i'm wondering if it is worth it.

I have 4 bots and a CNC, two run Marlin, two run sailfish, the CNC runs
grbl.  I have been coding C++/C for 25 years. Administering Linux for 12
years.

Wackerbarth, 2015-07-23:
> On Jul 22, 2015, at 8:35 PM, Tim Hawkins <notifications@github.com> wrote:
> 
> The Makefile is one single file, which those of us that use it are quite
> willing and capable to hack on it, the Makefile as it stands I believe
> works with all versions if you set the options/paths right. It appears that
> the only people arguing for its removal are those folks who don't use it.

Of course I don’t use it. It DOES NOT WORK (even after I set the path as described).
Your maintenance is incomplete.

> If the Makefile stops working then those of us that use it will fix it. As
> I did when I could not get my project to compile with the RepRap Discount
> Full graphic Display.

> I don't see how it is any risk or burden to the people who don't use it.

The burden is that users will build something that is “different” from what we expect and then raise questions as to why it doesn’t behave as they expect.

> This whole issue only arose because we DID maintain it. My original PR only
> had two commits in it, one for the U8Glib

Is that the version that we are distributing?

> and buzzer.cpp and one for a
> small typo. Things got confused when a second PR appeared with the same
> fixes done differently, plus a whole load of other stuff.
> 
> If you don't use the Makefile then by all means screw up your eyes and
> pretend you cant see it.
It isn’t ME. It is the host of other, less experienced, users that we attempt to support.

> But don't start wrecking other peoples workflows because you don't like the idea of something.
I’m sure that you can maintain your own Makefile, for your own use, without it being a part of the official distribution.
> To be truth full the whole process of submitting half a dozen lines of
> corrections to a file that people claim is not used or not useful has left
> me with a bad taste in my mouth, and i'm starting to [...]

Don’t get me started on “experience”. I, too, have multiple printers and a CNC.
And “Linux” is “Johnny come lately”. I started building lab instruments with Fairchild F8 processors and wrote operating systems before Unix was started at Bell Labs.

> On Thu, Jul 23, 2015 at 8:47 AM Richard Wackerbarth <
> notifications@github.com> wrote:

thawkins, 2015-07-23:
Before the changes it did not work if any lcd was selected (buzzer.cpp
missing), now it runs when 1.0.5 or 1.0.6 is used plus it now supports
graphic and non displays. So where is the regression? It may not work for
you, but it works fine for me and many other people. So why do you want to
kill it?

Wackerbarth, 2015-07-23:
> On Jul 22, 2015, at 9:46 PM, Tim Hawkins <notifications@github.com> wrote:
> 
> Before the changes it did not work

Exactly WHY I want to kill it.

> if any lcd was selected (buzzer.cpp missing),

> now it runs when 1.0.5 or 1.0.6

Obsolete! We need something that supports more modern versions of the Arduino core.

> is used plus it now supports
> graphic and non displays. So where is the regression? It may not work for
> you, but it works fine for me and many other people. So why do you want to
> kill it?

I agree that it is an improvement to the PARTIAL support under some conditions.
But we need something that works with the latest Arduino code, supports automatic versioning, etc.

If you are not going to do ALL of those things, then NONE is a better option for us.

daid, 2015-07-23:
So, kill G29 as well, as that's not working correctly as well right now? "It does not work, so kill it!" horrible way to work.

Also, "it does not work even after I put in the proper paths" is bad to horrible as a bug report. For the stuff I maintain that would get a "need more information + closed" as reply.

thawkins, 2015-07-23:
But it works for me, and it works for a population of other users too, I
still don't understand the logic that says that it should be deleted,
because it does not work for you. Are you the only user that matters? Fine,
we can add a note to the Makefile that says its only usable up to 1.0.6,
and later we can change that as we sort out the problems, but because it
does not do 100% of your requirement is not a reason for deleting it.

*I will commit to testing and working out how we can make it work with
later versions. it need not effect you, but deleting it will significantly
impact my workflow. *

*Note also that the Makefile is not mentioned in any docs, so its a power
user tool, its not the sort of thing that new Marlin users are going to use
if it is not mentioned anywhere, so i don't think its going to create any
significant support requirements. *

*This is the first discussion we have had about about the Makefile in
years, and its only because it improved, so to-date it has created
near zero support load. I think I asked one question about it a year ago
and somebody responded immediately with a solution that was not perfect but
good enough. *

*I think you are trying to fix a non-problem by having it deleted. You are
seeing scary monsters under your bed. *

thawkins, 2015-07-23:
I thought I would show what a "not working" makefile looks like when it runs. 

[thawkins@timdev04 Marlin]$ make
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_analog.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_digital.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_pulse.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_shift.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/WInterrupts.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ili9325d_320x240.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1601_c128032.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_bitmap.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ks0108_128x64.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1608_240x128.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_circle.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_lc7981_160x80.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1608_240x64.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_clip.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_lc7981_240x128.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1610_dogxl160.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_api_16gr.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_lc7981_240x64.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1611_dogm240.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_api.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_lc7981_320x64.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1611_dogxl240.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_attiny85_hw_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ld7032_60x32.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1701_dogs102.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_common.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_null.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_uc1701_mini12864.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_pcd8544_84x48.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_ellipse.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_pcf8812_96x65.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_font.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_hw_usart_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_sbn1661_122x32.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_font_data.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1306_128x32.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_parallel.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1306_128x64.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_line.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_port_d_wr.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1309_128x64.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_ll_api.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_ssd_i2c.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1322_nhd31oled_bw.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_page.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_st7920_custom.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1322_nhd31oled_gr.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_pb14v1.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_st7920_hw_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1325_nhd27oled_bw.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_pb16h1.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_st7920_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1325_nhd27oled_bw_new.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_pb16h2.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_std_sw_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1325_nhd27oled_gr.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_pb16v1.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_sw_spi.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1325_nhd27oled_gr_new.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_pb16v2.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_com_arduino_t6963.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_dev_ssd1327_96x96_gr.c
  CC    /usr/share/arduino/libraries/U8glib/utility/u8g_pb32h1.c
[... 70 lines cut ...]
  CXX   ultralcd.cpp
  CXX   digipot_mcp4451.cpp
  CXX   vector_3.cpp
  CXX   qr_solve.cpp
  CXX   buzzer.cpp
  CXX   /usr/share/arduino/libraries/LiquidCrystal/LiquidCrystal.cpp
  CXX   /usr/share/arduino/libraries/U8glib/U8glib.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/main.cpp
  CXX   /home/thawkins/tmp/MarlinBuild/Marlin.elf
  COPY  /home/thawkins/tmp/MarlinBuild/Marlin.hex

AVR Memory Usage
----------------
Device: atmega2560

Program:  109090 bytes (41.6% Full)
(.text + .data + .bootloader)

Data:       4779 bytes (58.3% Full)
(.data + .bss + .noinit)

   text	   data	    bss	    dec	    hex	filename
 108458	    632	   4147	 113237	  1ba55	/home/thawkins/tmp/MarlinBuild/Marlin.elf

[thawkins@timdev04 Marlin]$

thawkins, 2015-07-23:
Ok, a bit more hacking on the Makefile and it now works with 1.6.5 

I suspect it will be fine with other variants but for now im using that as the cutoff point. 

I have added path changes driven off the ARDUINO_VERSION that corrects for.:

1. in 1.6.x there are hardware dependent library folders for avr and sam,  forced the make file to add this if the version is 165 or greater
2. LiquidCrystal.c is in a slightly different place added both paths
3. 1.6.0 has a scheduler dummy for avr, it requires an additional "hooks.c" file from the core to be added to dummy out calls made in main. 

I have a RAMPS Board with 1) non graphic lcd, 2) graphic lcd. I will makesure i can compile and upload to the hardware for 

1. 1.0.6 - nolcd
2. 1.0.6 - text lcd
3. 1.0.6 - graphic lcd 
4. 1.6.5 - no lcd
5. 1.6.5 - text lcd
6. 1.6.5 - graphic lcd

sample output for 1.6.5 and REPRAP_FULL_GRAPHIC_CONTROLLER
[thawkins@timdev04 Marlin]$ make
  CC    /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/wiring.c
  CC    /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/wiring_analog.c
  CC    /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/wiring_digital.c
  CC    /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/wiring_pulse.c
  CC    /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/wiring_shift.c
  CC    /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/WInterrupts.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ili9325d_320x240.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1601_c128032.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_bitmap.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ks0108_128x64.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1608_240x128.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_circle.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_lc7981_160x80.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1608_240x64.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_clip.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_lc7981_240x128.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1610_dogxl160.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_api_16gr.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_lc7981_240x64.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1611_dogm240.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_api.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_lc7981_320x64.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1611_dogxl240.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_attiny85_hw_spi.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ld7032_60x32.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1701_dogs102.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_common.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_null.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_uc1701_mini12864.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_pcd8544_84x48.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_ellipse.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_hw_spi.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_pcf8812_96x65.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_font.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_hw_usart_spi.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_sbn1661_122x32.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_font_data.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ssd1306_128x32.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_parallel.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ssd1306_128x64.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_line.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_port_d_wr.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ssd1309_128x64.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_ll_api.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_com_arduino_ssd_i2c.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_dev_ssd1322_nhd31oled_bw.c
  CC    /home/thawkins/bin/arduino-1.6.5/libraries/U8glib/utility/u8g_page.c
[... 89 lines cut ...]
  CXX   ultralcd.cpp
  CXX   digipot_mcp4451.cpp
  CXX   vector_3.cpp
  CXX   qr_solve.cpp
  CXX   buzzer.cpp
[... cut ...]
  CXX   /home/thawkins/bin/arduino-1.6.5/hardware/arduino/avr/cores/arduino/main.cpp
  CXX   /home/thawkins/tmp/MarlinBuild/Marlin.elf
  COPY  /home/thawkins/tmp/MarlinBuild/Marlin.hex

AVR Memory Usage
----------------
Device: atmega2560

Program:  109168 bytes (41.6% Full)
(.text + .data + .bootloader)

Data:       4779 bytes (58.3% Full)
(.data + .bss + .noinit)

   text	   data	    bss	    dec	    hex	filename
 108536	    632	   4147	 113315	  1baa3	/home/thawkins/tmp/MarlinBuild/Marlin.elf

[thawkins@timdev04 Marlin]$

Wackerbarth, 2015-07-23:
> On Jul 23, 2015, at 12:28 AM, daid <notifications@github.com> wrote:
> 
> So, kill G29 as well, as that's not working correctly as well right now? "It does not work, so kill it!" horrible way to work.
> 
The difference is that G29 provides a functionality that is not otherwise available.

The Makefile provides only an alternative implementation to a function that is present and working.

The objection is to the requirement to maintain multiple versions of the same thing.
> Also, "it does not work even after I put in the proper paths" is bad to horrible as a bug report.
> 
> For the stuff I maintain that would get a "need more information + closed" as reply.
> 

I would agree if it were not for the fact that it is trivial for you to reproduce my “results” (error messages).
Since I have no incentive to “fix” the issue, I leave it to you to perform the analysis.

closed 2015-07-23 by daid

Roxy-3DPrintBoard, 2015-07-23:
> Note also that the Makefile is not mentioned in any docs, so its a power
user tool, its not the sort of thing that new Marlin users are going to use
if it is not mentioned anywhere, so i don't think its going to create any
significant support requirements.

One suggestion might be appropriate here to resolve everybody's concerns.   Would it make sense to have a comment at the top of the Makefile saying it is for advanced users and the normal way to generate the Marlin firmware is to fire up Arduino, load the sketch, compile and upload?

That way, anybody that stumbled on the Makefil would know instantly if it was of value to them.  And if it isn't of value, what the correct steps are for them to make forward progress.

Wackerbarth, 2015-07-23:
> On Jul 23, 2015, at 12:41 AM, Tim Hawkins <notifications@github.com> wrote:
> 
> But it works for me, and it works for a population of other users too

I don’t doubt that it works for certain build environments.

> , I still don't understand the logic that says that it should be deleted, because it does not work for you.

It matters because it does not work in other, more modern, environments and it is redundant.

When we add new “features” or reorganize our file structure (which is badly needed), I don’t want to be expected to make the necessary changes in two places. Having to do so is a violation of a fundamental design principal.

> we can add a note to the Makefile that says its only usable up to 1.0.6,
(Or we can move it to the 1.0.6 support area)

> and later we can change that as we sort out the problems, but because it
> does not do 100% of your requirement is not a reason for deleting it.
>
> *I will commit to testing and working out how we can [...]

Not really! You always have the option of resurecting the earlier version of a file and including it in your personal source tree.
I’m sure that you already do something like that to customize your Configuration.h.

> *Note also that the Makefile is not mentioned in any docs, so its a power
> user tool, its not the sort of thing that new Marlin users are going to use
> if it is not mentioned anywhere, so i don't think its going to create any
> significant support requirements.*

All the more reason to move it out of the main tree and into a support folder.

> *I think you are trying to fix a non-problem by having it deleted. You are
> seeing scary monsters under your bed. *

I see those “scary monsters” because I believe that we are about to move forward and get away from the obsolete file structure that is hindering development. “Cleaning house” is a part of that process.

Wackerbarth, 2015-07-23:
On Jul 23, 2015, at 2:54 AM, Tim Hawkins <notifications@github.com> wrote:
> Ok, a bit more hacking on the Makefile and it now works with 1.6.5

Excellent!

Now, if you will make those changes available, I would be interested in incorporating such a pull request.

thawkins, 2015-07-23:
I will tidy up and generate a PR in a day or two, i need to do more testing
with different versions.  I want to checkout 1.5.x

pdbogen, 2015-07-23:
(If even with the modern Arduino support we want to continue to maintain that the makefile is not the official build process, it might be useful to override the default target to be something like:
```
.PHONY: warn_unofficial i_understand
warn_unofficial:
	echo "Warning! Use of the makefile is not supported. Please use the standard Arduino build tools (such as the arduino CLI) to build this if you intend to report an issue or seek support."
	echo "To use the makefile anyway, please run 'make i_understand'
i_understand: <whatever the default build target is; build?>
```
This won't hinder current users at all, unless they happen to be running just "make" instead of "make build", and won't impact things like "make flash.")

thawkins, 2015-07-23:
I could live with that