MarlinFirmware/Marlin #2475: Regarding _APPLY_STEP(AXIS) from the file Marlin4Due

PisalPrasad123, 2015-07-20:
#define _APPLY_STEP(AXIS) AXIS ##_APPLY_STEP
	#define _INVERT_STEP_PIN(AXIS) INVERT_## AXIS ##_STEP_PIN

I am not clear with the meaning of :_APPLY_STEP(AXIS)(!_INVERT_STEP_PIN(AXIS),0);
Where the _INVERT_STEP_PIN(AXIS) is a boolean variable 

Also please clear the meaning of 
#define STEP_END(axis, AXIS) _APPLY_STEP(AXIS)(_INVERT_STEP_PIN(AXIS),0)

Wurstnase, 2015-07-20:
Don't try to compare this part with Marlin4Due. I changed something there. With the Due you have more power so this don't need double or quad stepping. In that case I moved some code to be sure to have enough delays between the stepping signals.

In that case I use STEP_START and STEP_END, which is close to STEP_ADD and STEP_IF_COUNTER.

Also I think, I hide a little bit to much the logic in APPLY_STEP. X_APPLY_STEP(v,Q) etc. are defined in the beginning of the stepper.cpp and need some input-variables. So (!_INVERT_STEP_PIN(AXIS),0) is (v,Q).

PisalPrasad123, 2015-07-20:
Thank you for your response.
I am not much clear about the 
#define STEP_END(axis, AXIS)_APPLY_STEP(AXIS)(_INVERT_STEP_PIN(AXIS),0) 
What is the significance of _INVERT_STEP_PIN and 0. in the above statement.

Why are we negating it here : why !_INVERT_STEP_PIN(AXIS)
_APPLY_STEP(AXIS)(!_INVERT_STEP_PIN(AXIS),0);

Wurstnase, 2015-07-20:
The '0' can be ignored. This second input variable is only for dual x carriages. I can't say exactly how this will work.

Invert is inverted logic. If you enable inverting for your stepper driver (some need a logical 0 to enable) you say activate my stepper. But activate is 1. In that case we activate inverting. So !1 == 0.

But if you don't need to invert you need invert the invert :)

PisalPrasad123, 2015-07-20:
Smart bit of coding with the preprocessors there. 
I wanted to know the difference in performance of "ENABLE_HIGH_SPEED_STEPPING" vs it not defined. I see that you calculate the new timer value in between. Is this of any major significance in the step rate?  Its also very different from original marlin , or bobc's code, where the high and low for the step is given together with just the counter and position in between.

Im building a SLA 3D priter, and changing this peice of code to increment/ decrement a DAC value, and need to hiest step rates possible ( results in how fast i can increment/decrement the DAC output)

Wurstnase, 2015-07-20:
High speed stepping in Marlin4Due is only needed if you need double or quadstepping. Everything up to approx. 120.000 steps/s don't need this. If you have stepping rates which are higher as 120kHz you can use it. But this also needs very fast stepper driver which can handle stepping signals less than 1µs.

If you use 8bit electronics this piece of code won't help you. This is only for fast ARM-processors.

PisalPrasad123, 2015-07-20:
The entire thing of using your code is to use Arduino due with its onboard DAC's, and yes will be requireing higher step rates, since the DAC's are part of the SAMX3 chip itself, i expect it to be pretty fast. 

Thanks for your help :)

thinkyhead, 2015-07-22:
> Smart bit of coding with the preprocessors there.

Thanks! :grinning:

closed 2015-07-22 by thinkyhead