MarlinFirmware/Marlin #2476: PROBE_SERVO_DEACTIVATION_DELAY

MoonshineSG, 2015-07-20:
After an update to the latest Development my Z-probe started shaking at every move. This used to happen before I enabled the PROBE_SERVO_DEACTIVATION_DELAY. 

Looking at the codes, I noticed that all the references to "delay(PROBE_SERVO_DEACTIVATION_DELAY);" have been removed... 

Can this be fixed ?

MoonshineSG, 2015-07-20:
Nevermind... I found it... there's a new setting called DEACTIVATE_SERVOS_AFTER_MOVE....

closed 2015-07-20 by MoonshineSG

AnHardt, 2015-07-20:
PROBE_SERVO_DEACTIVATION_DELAY was recently renamed to SERVO_DEACTIVATION_DELAY and is now located in the Servo configuration block (close to the end) in Configuration.h.
This was meant to deactivate the servos for ___all___ moves - not only that related to the PROBE.
To activate, uncomment DEACTIVATE_SERVOS_AFTER_MOVE some lines above.

thinkyhead, 2015-07-22:
#2491 should help.