MarlinFirmware/Marlin #2480: PID for Bed

MoonshineSG, 2015-07-20:
I got a new V3 aluminium bed and ran a PID tuning on it. The values look extremely high compared with anything else I've seen as PID parameters. I ran the test twice and values were both times high (but not equal)  
```
#define  DEFAULT_bedKp 864.44
#define  DEFAULT_bedKi 169.10
#define  DEFAULT_bedKd 1104.75
```
is this normal ?

pdbogen, 2015-07-20:
These are high, but might be sane. kP=864 means that at about .3 away from the target temp, kI and kD will start to have a major impact. This is within 5-6 seconds of target heating, on my printer, which is a goodly amount of time for the microcontroller to react.

The most important question is: Do these parameters work well for you? Are you seeing any particular issue using these? (I.e., too long of initial rise? too much oscillation? too much slump, i.e., settling below the target?)

brainscan, 2015-07-20:
From memory they seem similar to mine using the same bed.

MoonshineSG, 2015-07-21:
The settings actually work fine. I get stable temp in about 3-4 min without much oscillation.  Quite a steady increase with a slow down towards reaching closer to the target temperature. 

I just found that the numbers are extremely high, and not **fully** understanding how these PID work, I was slightly concerned... 

Thanks.

pdbogen, 2015-07-21:
Happy to help! If you're satisfied, I believe you should have a "close issue" button at the bottom of this issue?

MoonshineSG, 2015-07-21:
Found it. :)

closed 2015-07-21 by MoonshineSG