MarlinFirmware/Marlin #2485: Q: How to test a value in an array with the CPP?

AnHardt, 2015-07-21:
Currently this is untested. All information needed is known at compile-time. But i can't find the trick how to test this with the CPP.

```
#define NUM_SERVOS 1
#define SERVO_ENDSTOPS {-1, -1, 0}

int servo_endstops[] = SERVO_ENDSTOPS;

  #ifdef SERVO_ENDSTOPS
    for (int i = X_AXIS; i <= Z_AXIS; i++) {
      if (servo_endstops[i] >= NUM_SERVOS) {
        SERIAL_ERROR_START;
        SERIAL_ERRORPGM("Servo Nr:");
        SERIAL_ERROR(servo_endstops[i]);
        SERIAL_ERRORPGM(" for axis:");
        SERIAL_ERROR(i);
        SERIAL_ERRORLNPGM(" >= NUM_SERVOS!");
        LCD_ALERTMESSAGEPGM("Err: SERVORANGE");
        Stop();
      }
      if (servo_endstops[i] >= 0)
        servo[servo_endstops[i]].move(servo_endstop_angles[i * 2 + 1]);
    }
  #endif
```

pdbogen, 2015-07-21:
Can you clarify what it is you're trying to "test", and what you mean by "test"?

AnHardt, 2015-07-21:
if (servo_endstops[i] >= NUM_SERVOS)

pdbogen, 2015-07-21:
That doesn't help much. If by "CPP" you mean the C Pre-processor, you can't replace this if() with a preprocessor statement, because the value "i" is only available at runtime. You could unroll the loop yourself, which would be trivial to guard with preprocessor statements, a la:
```
#define NUM_SERVOS 1
// #define SERVO_ENDSTOP_X -1 // Left undefined to indicate disabled
// #define SERVO_ENDSTOP_Y -1 // Left undefined to indicate disabled
#define SERVO_ENDSTOP_Z 0

#if NUM_SERVOS > 0
  #ifdef SERVO_ENDSTOP_X
    #if SERVO_ENDSTOP_X >= NUM_SERVOS
      #error SERVO_ENDSTOP_X servo number out of bounds!
    #else
      servo[SERVO_ENDSTOP_X].move(servo_endstop_angles[SERVO_ENDSTOP_X*2+1]);
    #endif // SERVO_END_STOP_X >= NUM_SERVOS
  #endif // defined SERVO_ENDSTOP_X
#endif // NUM_SERVOS >0
```

And then repeat everything from #ifdef SERVO_ENDSTOP_X to its corresponding #endif for each axis. (But, this seems like it verges on premature optimization; is this code going to run so often that it's worth complicating it up with preprocessor stuff, when the simple for loop may well be good enough?)

AnHardt, 2015-07-21:
The advantage of the CPP variant is not time, but space. The strings for the error have not to be in the memory then.
Do you see a way without the change of the initialization of the array? (What means changing 12+ files)

pdbogen, 2015-07-21:
You could probably do:
```
#define NUM_SERVOS 1
#define SERVO_ENDSTOP_X -1 // -1 indicates disabled
#define SERVO_ENDSTOP_Y -1
#define SERVO_ENDSTOP_Z 0
#define SERVO_ENDSTOPS { SERVO_ENDSTOP_X, SERVO_ENDSTOP_Y, SERVO_ENDSTOP_Z }
```

and then use the preprocessor code from above, ignoring the SERVO_ENDSTOPS array here and just referencing the raw values. You'll need to adjust the #ifdef, though:
```
#if NUM_SERVOS > 0
  #if SERVO_ENDSTOP_X > -1
    /* stuff from above */
  #endif // SERVO_ENDSTOP_X > -1
#endif // NUM_SERVOS > 0
```

AnHardt, 2015-07-21:
@pdbogen 
Thanks for now.

Anyone with an idea without changing all the `Configuration.h`s?

Wurstnase, 2015-07-21:
with notepad++ you can easy replace stuff in many files.

thinkyhead, 2015-07-22:
@AnHardt You can see the basic error I am correcting here… You want to compare `i` with `NUM_SERVOS` for one thing. For another, you can get the number of elements automagically with `sizeof(arr)/sizeof(*arr)`…

    for (i = 0; i<sizeof(servo_endstops)/sizeof(*servo_endstops); i++) if (i >= NUM_SERVOS) error();

thinkyhead, 2015-07-22:
@AnHardt Actually, what @pdbogen said is about the only way. You can't introspect into arrays using the preprocessor.

closed 2015-07-22 by AnHardt