MarlinFirmware/Marlin #2486: Auto Bed Leveling Innaccurate. Probing Z speed possible culprit?

Natealus, 2015-07-22:
Well I know I'm getting to pretty fine detail once I start printing at 100 microns. But the quality of finer layers is just undeniably better than the 200-800 micron layers (0.2mm-0.8mm). It also highlights any calibration and movement inconsistencies which may not even be noticeable at larger layer sizes. But I've been noticing my right side and other random spots seem to get closer than the rest of the part especially over at the left. I've moved the offsets to see if that would change the balance but it didn't seem to make a significant difference.

What I do think though as I was testing #define XY_TRAVEL_SPEED 4000, which governs Traveling X, Y, and Z during none probing movement. I turned it down to 2000 then 500. Its good to have a value to affect that for different setups.

But it did not affect the downward Z probing speed. I'm starting to wonder if it's moving too fast to achieve the 0.01mm or 10 micron probing resolution that I need to bring the plane even enough to squash a 100 micron layer down to 70 micron bed to nozzle height.

I was wondering since it seems to already be a separate speed value for that downward probing movement, if we couldn't add a #define right below #define XY_TRAVEL_SPEED 4000 for Z_PROBING_SPEED. It's currently going like 1000 or more cause it probed down faster than all the other XY and up Z moves at 500 mm/min.

I believe if we let it move down significantly slower we would have far more accurate readings from the trigger to stepper position planner. And being able to set that really slow might help us to diagnose if there's other issues.

AnHardt, 2015-07-22:
@Natealus 
As the name says XY_TRAVEL_SPEED has no Z component. The Z_RAISE_*_PROBING? moves are done with DEFAULT_MAX_FEEDRATE. The first approach to an endstop is done with HOMING_FEEDRATE. The second and only precision relevant approach to the endstop is done with HOMING_FEEDRATE / HOMING_BUMP_DIVISOR. HOMING_BUMP_DIVISOR can be found in Configuration_adv.h.

Natealus, 2015-07-22:
Well we're talking about the G29 auto leveling process. My G28 homing works just fine and the homing and bumping rates I have found and adjusted. But changing XY_TRAVEL_SPEED certainly did affect the initial Z movement down to Z_RAISE_BEFORE_PROBING position for auto leveling as well as the speed it moves up after triggering the Z probe. 

It did not affect the probing movement down to each point though. As I don't think it should, but it would help to have a separate variable to modify that measurement probing speed.

Natealus, 2015-07-22:
Sorry, it might help to mention I'm using a delta so my effector travels down to the stationary bed. It seems the square printer units are the most popular so I thought I should mention I'm in that other category lol.

AnHardt, 2015-07-22:
Oh sh... . I remember.
Sorry, my description was for cartesians. I'm not very familiar with the delta code. Just beginning to hunt some obscure behavior with cartesian homing - leveling. 
However, the above summary of speeds (for cartesian printers) may help someone else.

Natealus, 2015-07-22:
Yeah I only have max endstops and no mins except the Z probe which it attached to the effector on a servo. The probing down sequence seems to be on its own speed cause its frequency/sound of the stepper speed is different from my homing and bump speed though. Might be specific to the delta bed leveling code.

I'm getting antsy to test the LCD work you've done man, but I just wanted to keep things simple while I test the bed leveling. Certainly much improved over the last couple months. :)

thinkyhead, 2015-07-24:
@Natealus The probe-down feedrate on DELTA appears to be `homing_feedrate[Z_AXIS] / 4`. And it only probes down once, with no slow probe.

Natealus, 2015-07-24:
Ah I see I evened all mine out. By default the Z tower homes much slower than the other two cause...that's typical in other square machines. But deltas are usually all even. I could try taking that homing_feedrate[Z_AXIS] value down to adjust the down rate. 

And yeah it does no slow probe. If that "bump" could occur it might enhance the precision of the probings.

thinkyhead, 2015-07-24:
> By default the Z tower homes much slower than the other two 

@Natealus The homing rate on deltas is supposed to be set the same for X, Y, and Z. See the delta examples in `example_configurations`…

    #define HOMING_FEEDRATE {200*60, 200*60, 200*60, 0}  // set the homing speeds (mm/min)

> … that "bump" … might enhance the precision

Yes it certainly could. Maybe we should include that. We already have the homing bump multiplier for normal homing. For probing we could either use the bump divisor (moving at the homing bump speed) or we could add a new configuration option, something like `Z_PROBE_BUMP_FEEDRATE` so that it can be set separately.

Natealus, 2015-07-24:
Yes in general thats not how one would want to do it but it did affect Z-Probing speed. Slowed it Way down.

// delta homing speeds must be the same on xyz
#define HOMING_FEEDRATE {100*60, 100*60, 30*60, 0}  // set the homing speeds (mm/min)

You were right @thinkyhead those were all the same (I lowered it to not be so rough on my frame) But the probe does not move at Z homing bump speed. It slows to snail pace when bumping the Z tower. It is however affected by the HOMING_FEEDRATE as the example above. I have found the number to not perfectly match but be within 30ish microns. Better than it was, the bottom layer seems to be forming more evenly.

I do like the suggestion of a new configuration option like Z_PROBE_BUMP_FEEDRATE so we don't affect other homing operations with modifying for the Z-probe.

Wackerbarth, 2015-07-24:
I see no reason why the probing bump rate should be any different from the tower (max) bump rate on a data printer. The thing that we do know is that we know approximately where we will find the bed on the first probe, and after a G28 we are far from that point.  So, it would make sense to move "near" the bed at travel speed and then proceed to actually find the bed at bump speed. Or perhaps, if we intend to double bump the first point, reduce that leg to homing rate.

Natealus, 2015-07-24:
Well that HOMING_FEEDRATE is the only one that affect the downward probing sequence. All of the other moves from moving down to the bed to moving up from a probe and over to the next spot all run off XY_TRAVEL_SPEED in the bed leveling section.

Natealus, 2015-07-25:
I've also merged the recent changes to G29 and it looked like Z_RAISE_AFTER_PROBING was involved in the changes but I still get that old funky probing array

<--1200.08 -950.17 100.07 150.13 0.17 -149.89 -299.78 -449.67 -599.55 -549.92 99.90 0.00 
<-700.03 -449.98 -300.02 250.13 200.18 150.11 100.14 50.11 0.05 349.95 499.68 0.00 
<-1049.44 900.00 300.12 350.13 400.20 450.11 500.07 550.01 600.00 649.86 -49.92 0.00 
<-199.63 1149.62 1099.96 1050.22 1000.28 950.23 900.11 850.04 800.03 749.97 700.02 0.00 
<-1149.34 1199.57 1249.79 1299.92 1349.98 1399.92 1449.94 1499.88 1549.86 1599.82 1649.78 1699.68 
<-2298.76 2249.13 2199.51 2149.84 2099.97 2050.03 2000.03 1949.88 1899.87 1849.86 1799.79 1749.62 
<-2248.71 2299.03 2349.36 2399.66 2449.77 2499.84 2549.69 2599.73 2649.63 2699.61 2749.50 2799.33 
<-2398.56 3348.59 3299.01 3249.44 3199.71 3149.82 3099.88 3049.88 2999.71 2949.61 2899.59 2849.34 
<-3347.82 3398.31 3448.79 3499.27 3549.51 3599.61 3649.75 3699.63 3749.58 3799.48 3849.32 0.00 
<-3497.61 3648.14 4298.28 4248.96 4199.38 4149.67 4099.74 4049.98 3999.79 3949.62 3899.52 0.00 
<-3847.49 5097.29 4948.41 4348.41 4399.16 4449.49 4499.61 4549.54 4599.48 4649.48 4149.66 0.00 
<-0.00 0.00 0.00 0.00 4848.43 4799.33 4749.62 4699.68 0.00 0.00 0.00 0.00 
<-ok
<-echo:endstops hit:  Z:5126.19 Z_PROBE:5126.19
<-X:-51.00 Y:135.00 Z:4912.11 E:0.00 Count X: 5031.01 Y:4936.49 Z:5126.20

Natealus, 2015-07-28:
@thinkyhead I have reduced my HOMING_FEEDRATE to
HOMING_FEEDRATE {100*60, 100*60, 20*60, 0} 
and it does seem to give more consistent probing. I also did an old school tower to center delta radius calibration and got that much better to true. The print tests so far are looking pretty good.

If you were to make the probing speed option a bump...how do bumps work? Does it throw out the first number from the faster speed and takes the slower bump's number? Or does it average the two out?

I'm trying to think of an easy way to input the probing arrays that get output from auto bed leveling and do several iterations of G29 to put together a min, max, average, and range to determine how accurate different probing speeds are.

Wackerbarth, 2015-07-28:
@Natealus - Nate, If you have a delta, you need to set all three homing rates to the same value. (The smaller one).

Natealus, 2015-07-29:
That's rule of thumb yes but I changed it to make Z-Probing slower and more accurate. That's why I opened this issue so we could make a separate setting to control that. My homing bump rates come out the same though cause I adjusted those to bump evenly.

Natealus, 2015-07-29:
So in the interest of making sure that wasn't the issue I changed back to defaults and all even homing feedrates to produce these numbers. It appears there is something happening in the middle of the probing array. And it is inconsistent too because the probing arrays with tolerances of 1.5mm are usual and probably correct. 

But sometimes in the middle it seems to not move over to the true next position and seems to double tap the same spot...including the z-probe servo moving up and down. I can tell when that happens cause the rhythm is different than the normal successful probing sequences. There was also one that probed itself 200mm up from the bed and gave all the same 23ish number. I've gotten successful prints and total disasters because of the 5mm error it seems to get into. 

Thinking its variables I'm using or something else that got changed I have tried probing offsets and M666 and removing the M666. Reverting a recent change with Z_AFTER_PROBING seemed to help but I dont think that was the root of the issue cause it still happens. I can't quite lock down where the bug is or the conditions cause I just set it to go several times with these numbers.

http://pastebin.com/y1Mg0wrX

thinkyhead, 2015-07-31:
> (not sure what that weird styling is...supposed to be asterisks)
The backtick character is your friend. Use a single backtick as a quote character to make `inline code` or use 3 backticks around a block to get this:
```
some text
  between backticks
    which look like this ```…```
```

thinkyhead, 2015-07-31:
@Natealus My [debug_G29](https://github.com/thinkyhead/Marlin/tree/debug_G29) branch is up-to-date with the latest Development code. Give it a try and post the log output here. The log should be helpful to isolate where any logical problems might be happening.

Natealus, 2015-07-31:
Awesome I would be happy to do so. Do you want me to let it go to print? Currently the most reliable way I've been doing is a macro to g28 and g29 E and if the probing sequence is successful, I start the print then my starting g-code raises 50mm and then centers the effector after the temp checks then prints.

thinkyhead, 2015-07-31:
@Natealus There's no logging added for anything apart from `G28` and `G29`, so printing won't give any extra information about that part. But it might show how well (or badly) the leveling matrix is being applied.

Natealus, 2015-07-31:
@thinkyhead So...in all this testing even from May and such...it has very consistently had a section that drags the nozzle too close near the side of the circle by the Y tower. Understandably it's a Rostock Max V2 frame being held together by particle board but I thought bed leveling would compensate that? 

I've tried offsets from both actual nozzle offset to the center of the effector and it doesn't seem to make the difference. I've tried software endstop offsets, disabling or leaving Z_RAISE_AFTER_PROBING enabled, re-calibrating towers to center, slowing down the speed variables...and still a bit mystified why that section consistently comes out too tight and snuffs the filament out cause it can't make it out the nozzle. I'm going to do several iterations of bed level and a print then pastebin it :)

Natealus, 2015-07-31:
@thinkyhead Here's a debug where I had to reset it midstream because it was probing midair and went back and collided with my Z tower and skipped some steps but the rest of the debug is there. One thing I've noticed in the tests is the debug in the auto leveling doesn't seem to list the z-probe hit on the endstop switch. I'm not sure if that's significant or if it just didn't get put out in the output.

http://pastebin.com/P1XFdr7U

thinkyhead, 2015-07-31:
Ok, let's take it one small bit at a time.

First, it looks like the results of `G28` seem ok. Your home position has XY in the center, and Z at 368.05. That should place your nozzle at 368.05mm above the bed.

Next, in `G29` the nozzle is supposed to move down so that it ends up 17mm above the bed. This output shows that to be the assumed case:
```
<-> current_position: (0.00, 0.00, 368.05)
<-Z Raise to z_before 17.00
<-> do_blocking_move_to_z 17.00
<-do_blocking_move_to: (0.00, 0.00, 17.00)
<-prepare_move_raw > destination: (0.00, 0.00, 17.00)
```
Next the nozzle moves to X=48, Y=-39 keeping the same Z:
```
<-> do_blocking_move_to_xy 48.00, -39.00
<-do_blocking_move_to: (48.00, -39.00, 17.00)
<-prepare_move_raw > destination: (48.00, -39.00, 17.00)
```
Now it deploys the probe:
<-> ProbeDeploy
<-deploy_z_probe > current_position: (48.00, -39.00, 17.00)
```
Next is should do the first probe and stop when the probe is triggered:
```
<-run_z_probe (DELTA) 1
<-prepare_move_raw > destination: (48.00, -39.00, -10.00)
<-run_z_probe (DELTA) 2 > current_position: (48.00, -39.00, -10.00)
```
This seems to be where the trouble starts. The probe is a servo probe (I presume) and so it should enable the probe switch and cause movement to stop when the trigger is hit. From the output above it looks like the probe never reaches the bed and never gets triggered. But this output could also indicate that the probe switch is not enabled, or the probe code is not treating it like an endstop probe. In either of those cases the probe would try to keep moving below the bed.

Next, the probe stows… and so on…
```
<-> ProbeStow (stow_z_probe will do Z Raise)
<-stow_z_probe > current_position: (48.00, -39.00, -10.00)
```

thinkyhead, 2015-07-31:
Ok, so I've added more logging to [debug_G29](https://github.com/thinkyhead/Marlin/tree/debug_G29) which will output when endstops are being enabled and disabled. Give this a try and post the new logs so we can see whether endstops are being enabled at all during the probing process. I haven't dived into the code yet to see where it deals with a Z Probe that exists separate from a Z Max Endstop, but you might try enabling the `#define Z_PROBE_ENDSTOP` option to see how it affects things.

Natealus, 2015-07-31:
Yeah heres another sample of normal numbers I didn't really notice anything out of the ordinary on this set. Longest stream of successful ones in a while and I couldn't paste the whole run so if you want more I have another 3 probings I couldn't paste due to site size limit.

http://pastebin.com/YnrUHKnM

Natealus, 2015-07-31:
So one thing of note...I would think that if my probe didn't have a wheel on the bottom it would cause more of an issue, but it stows the probe before moving up Z after the probe point. Another mechanical probe might tear tape in that circumstance.

Should I have ENDSTOPS_ONLY_FOR_HOMING disabled?

Natealus, 2015-07-31:
@thinkyhead I do have Z_PROBE_ENDSTOP enabled by the by.

thinkyhead, 2015-08-12:
@Natealus Once again I've updated (and moved) https://github.com/thinkyhead/MarlinDev/tree/debug_G29 so it has all the latest code. But we have even more changes coming in this area, so you might want to wait and keep an eye on https://github.com/MarlinFirmware/MarlinDev/pulls for the pending leveling-related items to get merged. Delta probing is definitely one of the things we want to address asap.

thinkyhead, 2015-08-12:
Relating to the Z probing speed, in `run_z_probe` for `DELTA` it is doing this:
```cpp
feedrate = homing_feedrate[Z_AXIS] / 4;
```

Natealus, 2015-08-12:
@thinkyhead Yeah so for a time I tried adjusting the that homing feedrate and it did indeed slow it down. But not sure you're technically allowed to have uneven feedrates on delta homing but it worked. I set back to normal for the recent testing to make sure that wasn't the issue.

I'd prefer it if we could adjust it separately. Some people's system might be accurate and repeatable at faster speeds while others might need a slow probe to be sure.

And I'm totally down to debug but if further changes are just minutes/hours away maybe we should wait for those ones and see?

giliammc, 2015-08-31:
what kind of switch are you using

thinkyhead, 2015-09-01:
@Natealus I agree, the feed-rate being based on homing divided by four is not so great on a delta, because deltas can home so quickly. I'll put that on my list – to allow a slower probing speed, and it would also be useful to allow a second slow-probe as with cartesians. It will require a little bit of new code, possibly new configuration options, so I may wait till after the 1.1.0 release before I implement that.

Natealus, 2015-09-17:
Sorry I've been gone for a while guys. I had to step off development because I had some production prints for clients to get going plus several upgrades to my printer. Replaced axis motors with 400 step/rev upgrades. Upgraded extruders and printed them in Alloy 910. Sturdy stuff! 

I've replaced the mechanical micro switch for Z-Probe with a capacitive proxy sensor. Repeatability seems much better but I still experience the issue of the lower right section of the print being too tight. Its almost like the delta bed leveling code thinks the origin is over by the Y tower rather than dead center.

Natealus, 2015-09-17:
As it is, I can get a near perfect plane with manual calibration of the X Y Z endstop adjustment screws and minor nozzle Z height adjustments. When I let it print off of the auto bed leveling grid...it can skew the plane so much it almost tears the kapton tape if I'm not careful. And the side opposite of the tight side is too loose and the print lines don't merge. I've taken offsets back and forth to see if its a point of reference issue with my Kraken nozzle offsets, but it doesn't seem to fix it.

zlikovski, 2015-09-17:
Hi, a complete 3d printing newbie here. I'm just about to finish my first delta (yup, that much of a newbie), so maybe my comments are off. But are you sure you aren't getting to the physical limit of your build? 

For example, my printer will have GT2 timing belt with pulley that has 12.22mm diameter (which is pretty much usual thing as far as i could see). So, taking into account 1.8° step of the motor (200step/rev), each motor step translates into 0.19mm linear travel. Even if you move to 400step/rev motor, that just halves that number to 90 microns. So without reduction gears its hard to get lower, but then you sacrifice speed.

Second thing is thermal fluctuations. My vertical structure (65cm long) is made from aluminium (thermal expansion coefficient 22.2e-6 1/K) which translates into roughly 14 micons/K of linear expansion. So, if the temperature fluctuates just by a few degrees (and you are heating a plate just below and warm air rises) you get easily 50-100 microns error just from stretching of the bars.
Adding errors that will come from other mechanical imperfections in the system, I highly doubt in the 10 micron probing resolution you are trying to achieve. Sure, you can measure it that precisely, but I believe you are being mislead by the numbers on the screen. Your real resolution won't be near that number.

Again, I'm really new at this so i can be easily very wrong.

brainscan, 2015-09-17:
You haven't taken micro steps into account, 0.19 mm per step would be awful resolution.

giliammc, 2015-09-17:
@zlikovski first of all you don't seem like a noob with the kind of questions you are asking. i have a printrbot  simple 2014 so 100 microns deviancy seems like nothing to me. if you have three rails and they all stretch somewhat evenly would this not simply increase the maximum height slightly. this could be a problem if your printer's end stops where at the top of those rails then your print might be 100 microns above the bed. but if you are using a ABL probe it should compensate for that

zlikovski, 2015-09-17:
@brainscan yep, I haven't. But the issue is not whether you can move the motor for smaller angle, the issue is whether you can do so reliably and repeatably. And from what I've gathered around, the answer is "weeeellll, kindof, maybe". It's kind-of OK when you start reducing steps (1/2, 1/4..)  but goes bad very, very fast (http://www.micromo.com/microstepping-myths-and-realities). What one needs in this case is get a factor of 20 in resolution (or 10 if you get 400steps/rev motor) and that seems a lot to me. And this would be disregarding all other sources of error.

@giliammc your analysis would be correct if the situation would be completely thermally static. However, you will first heat up the bed. Then you will do bed leveling and start printing before side bars would get into their static thermal distribution. And then printing comes into play when you have a hotend whirling around just above the hot bed. And just 1°C of average change over the bar is enough to mess it up by 14microns. Can one really be sure that the temperature won't vary even that little?

I really am a complete beginner when it comes to 3D printing, and I'm trying to learn as much as I can about it. And I wouldn't even comment on this thread because I don't know yet enough about all the intricacies of how change of one parameter affect the print, was it not for the goal of getting a 10um resolution (that's what got my eyebrows raised up).  
Occasionally I push atoms/molecules around for a living, down on the nanometer scale. And I have (decent) experience with constructing mechanical devices that need micrometer (or 10um) precision (mechanically), and it's not something i can see my delta printer getting to, not even close (even though it's beautiful, and shiny, and will eventually print something out).

KiteLab, 2015-09-17:
Because the belts are stretching about the same amount as the towers, the hight of the towers measured in steps is about constant.
If we assume the motors and the bed are fix - then the tops of the towers move up with temperature. We are mostly interested in highs close to the bed - small hights. Here even the absolute error is smallest.

thinkyhead, 2016-02-07:
More discussion needed? How does the latest RC stack up in this area?

closed 2016-04-23 by jbrazio