MarlinFirmware/Marlin #2489: prepare_move_raw for non-Delta?

Wurstnase, 2015-07-22:
Is that right? I can't find prepare_move_raw for non-Deltas.

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L1334

thinkyhead, 2015-07-23:
Correct. The `prepare_move_raw` function exists only for the benefit of Delta. It does a "raw" linear move (which may be curved in cartesian space) instead of an interpolated move, which converts long moves into several short (raw) moves.

Wurstnase, 2015-07-23:
This doesn't makes any problem because there is no
```#ifndef DELTA
prepare_move_raw() {};
#endif```
?

This function is called often in non Delta specific parts.

AnHardt, 2015-07-23:
Ahh. The difference between G0 and G1 on "real" CNC machines.
These scary looking concave moves. Do we want them at all?

thinkyhead, 2015-07-24:
> This function is called often in non Delta specific parts.

It is called only in blocks of code bounded by `#ifdef DELTA` or `#ifdef Z_PROBE_ALLEN_KEY`. The latter can only be enabled on a delta.

thinkyhead, 2015-07-24:
> These scary looking concave moves. Do we want them at all?

Not really, and there's no compelling reason to include them that I can imagine, except to optimize movement. We should be able to replace all calls to `prepare_move_raw` with `prepare_move`, or replace the two middle lines in `prepare_move_raw` with `prepare_move_delta(destination)` (if we want to skip the call to `clamp_to_software_endstops`.)

Wurstnase, 2015-07-24:
> It is called only in blocks of code bounded by #ifdef DELTA or #ifdef Z_PROBE_ALLEN_KEY. The latter can only be enabled on a delta.

Ah, now I see it. I saw some cartesian/corexy with an allen key for the leveling and tought this is also for them.

closed 2016-04-23 by jbrazio