MarlinFirmware/Marlin #2490: Cannot disable volumetric once set and committed to EEPROM

tadawson, 2015-07-22:
To disable volumetric extrusion, the docs say set M200 D0 . . . But in Marlin_main.cpp, the conditional 

    // slicers either generate in extruder values as cubic mm or as as filament 
feeds
    // for all extruders
    volumetric_enabled = (diameter != 0.0);
    if (volumetric_enabled) {
      filament_size[target_extruder] = diameter;

Basically discards this setting, since it is equal to zero.  Thus, filament_size is not set, and when M500 is executed, the old filament_size is still passed, and the EEPROM maintains the old value.  Thus, running M200 D0 will temporarily reset to off, but either a printer reset, or an M501 brings back the old setting, and reenables.  An M502 also clears, but again, following with M500, the old values persist in the prom, and volumentric E is still set.  

To verify, I temporarily edited the code in configuration_store.cpp to force filament_size to zero, and verified that it *DID* persistently disable volumetric E. 

This was verified to still be present in the current git, as of the time of writing this issue.

thinkyhead, 2015-07-22:
> M501 brings back the old setting, and reenables 

From my analysis it looks like `volumetric_enabled` will be disabled for a moment by a reset (that is the initial default value), but then `volumetric_enabled` is loaded from the EEPROM along with the saved `filament_size` values. If `volumetric_enabled` was enabled when `M500` was done, it will be enabled when `M501` is done, regardless of any `filament_size` values. The only time a filament size setting affects `volumetric_enabled` is in the `M200` handler.

The question is, should `M200 D0` only set `volumetric_enabled` to `false` or should it also set `filament_size[target_extruder]`  to `DEFAULT_NOMINAL_FILAMENT_DIA` at the same time?

If you want that behavior, you can remove the condition `if (volumetric_enabled) ...` and let the code inside that block run regardless. That will effectively set `filament_size[target_extruder] = DEFAULT_NOMINAL_FILAMENT_DIA` when you issue `M220 D0`.

tadawson, 2015-07-22:
Allow me to gently propose that your analysis is incorrect, since that is clearly *NOT* what is happening. 

Initially, I the line in Configuration.h for DEFAULT_NOMINAL_FILAMENT_DIA was uncommented, and set to 1.75. 

Upon booting, the printer would be in a state of volumentric_e enabled, and would not feed correctly. A simple M502/M500 would get things back to working correctly, and since volumetric e was now disabled, the M500 should have written volumetric_enabled to the EEPROM as 0, correct? 

Reset the printer.  Volumetric e is now once again enabled, at the setting of 1.75. 

I zeroed the value in Configuration.h, and reuploaded.  No change.  M502/M500 after reload, no change.  I hard wired volumetric_enable to 0 in configuration_store.cpp to ensure that it was truly disabled.  No change - always sets back to the last value. Force setting M200 D0, M500, reset, comes back at last value set. 

The issue I see is that the flag volumetric_enabled is *NOT* tested when the values for filament diameter are read from the EEPROM - only in the routine which echos the config.  And considering that in Marlin_mail.cpp, again, there is no test, just a setting of volumetric_enabled based on a non-zero value of filament size, I think we have the root of the problem. When you execute M200 D0 to disable, it takes directly, but since it never sets the diameter to zero (it exits), the prior value is still in the variable.  Thus when you do an M500, the old diameter value is written, since there is no test for volumetric_enable in the write code for the prom - thus we get volumetric_enabled=0, and non zero diameter data.  On read, since again, there is no test of volumetric_enabled prior to reading the diameters, the M200 Dxx commands are passed to Marlin_main.cpp with non zero values, hit the non-zero test, and promply reenable volumetric e.  The *ONLY* was I could get out of this loop was to hard wire the write value in configuration_store.cpp for filament_size to 0, to ensure that wrote, and then after an M500, a reset again came back with volumetric e disabled.  Any non zero filament size set, followed my an M200 D0 to disable, restores to the value set last prior to the M200 D0 . . . 

So, I think there are a couple of ways to resolve this:

in SonfigurationStore.cpp, add the test so that when volumetric_enabled is zero, do *NOT* subsequently read the three filament_size values - they are invalidated by the disable (line 346). 

In ConfigurationStore.cpp, when writing filament_size, force it to zero if volumetric_enabled is zero (line 112)

In Marlin_main.cpp, introduce a test to actually *SEE* the status of volumetric_enabled as read from the prom, and take no action. 

Myself, I'm a fan of the first option, since it would leave the values intact, and just no-op on them until things are reenabled.  Considering that the only way to reenable is to set a value, I'm not certain that there is any value in maintaining this data in the prom. ? ? ? ? 

I also note that when in this 'deadlocked' condition, I can no longer toggle the volumetric e status on the LCD controller - it's on, and it's damn well staying that way.  Once I do an M200 D0, then I can toggle the value as expected . . . so somethings are apparently respecting the volumetric_disabled flag better than others . . . 

I suggest that you give this an actual test . . . I'm not making this up . . . this is what it does . . . 

- Tim

thinkyhead, 2015-07-24:
It looks like you must be using 1.0.2. My head is only in the latest Development code, but I will have a quick look at the old code and see if there's something obvious going on that I can help you patch.

tadawson, 2015-07-24:
I think I may have looked at 1.0.2 when I pulled my code fragments, but the printer doing this is on 1.0.3-dev for certain, and I did verify that I can see no code changes in git that invalidate anything in  my report.

I already know what I can do to patch - any of the three things I mentioned would be trivial to code. I am more interested in getting a potential bug out of the codebase, or I would have just patched and marched on . . .  I just don't feel that thrilled at having to maintain my own fork, and reconcile changes all the time.

thinkyhead, 2015-07-24:
> …doing this is on 1.0.3-dev for certain

Ok. You were referring to some old files there. `ConfigurationStore.cpp` is now `configuration_store.cpp` for example.

So, the behavior should be as I described. On initial boot, `volumetric_enabled` is `false`. The `filament_size` entries are all set to `DEFAULT_NOMINAL_FILAMENT_DIA`. The values of all `volumetric_multiplier` entries is 1.0. If you run your printer without EEPROM this is what you should get.

Next, when you load from EEPROM with M501, the value of `volumetric_enabled` is read:

    EEPROM_READ_VAR(i, volumetric_enabled);

…followed by the `filament_size` values…
```cpp
    for (int q = 0; q < 4; q++) {
      EEPROM_READ_VAR(i, dummy);
      if (q < EXTRUDERS) filament_size[q] = dummy;
    }
```
…and lastly the `volumetric_multiplier` values are set. If `volumetric_enabled` is `false` these will be set to 1.0…

    calculate_volumetric_multipliers();

tadawson, 2015-07-24:
I can tell you for certain, that the only way I can get extrusion back to normal is to set filament size in the EEPROM to zero . . . otherwise, I get radically short extrusions . . . And the M503 base output echos exactly the llast value that was entered by M200 Dn, so in that regard, it's accurate, since nothing in Marlin will zero those values. Even if I set to my actual filament size(M200 D1.75), I still get wrong volume. Frankly, were there a conditional to disable this feature, that would have been my solution . . .

tadawson, 2015-07-24:
OK, testing . . . I have added trace echo messages in Marlin_main.cpp as well as configuration_store.cpp to give status and values of volumetric_enabled as well as diameter when the setting function in Marlin_main is hit, as well as either when read or write of the eeprom works with volumetric_enabled or filament_size . . .

Started with clean prom, M503 gives "Filament settings disabled" as expected.

M200 D1 entered. Hit trace in Marlin_main that volumetric_enabled is set, and diameter is 1.0

M500 entered. Got trace from write that volumetric_enabled is set, diameter written as 1.0, as expected.

M503 entered. Echos M200 D1.00 as expected.

M200 D0 entered. Hit trace of volumetric_disabled in Marlin_main, no diameter change made - variables still at 1.0

M500 entered. Trace shows volumetric_disabled in EEPROM, wrote value 1.0

M503 entered, get back "Filament settings disabled"

Printer reset

M503 entered. Get back Filament settings: M200 D1.00

Entered M200 D0, trace proper as above.

Entered M500, trace proper as above

Entered M503, shows Filament settings disabled

Entered M501, get back read trace of volumetric_enabled as true in trace in read routing in configuration_store.cpp

Entered M503, get back M200 D1.00

OK, now for potential fixes:

I changed Marlin_main.cpp to:

if (code_seen('D')) {
float diameter = code_value();
// setting any extruder filament size disables volumetric on the assumption that
// slicers either generate in extruder values as cubic mm or as as filament feeds
// for all extruders
volumetric_enabled = (diameter != 0.0);
if (volumetric_enabled) {
SERIAL_ECHO_START; //Dawson - debug
SERIAL_ECHOPGM("Volumetric Enabled in Marlin_main.cpp");
SERIAL_ECHOPAIR("Diameter set to ",diameter);
SERIAL_EOL;
filament_size[tmp_extruder] = diameter;
// make sure all extruders have some sane value for the filament size
for (int i=0; i<EXTRUDERS; i++)
if (! filament_size[i]) filament_size[i] = DEFAULT_NOMINAL_FILAMENT_DIA;
}
else {
SERIAL_ECHO_START;
SERIAL_ECHOPGM("Volumetric disabled in Marlin_main.cpp");
SERIAL_EOL;
filament_size[tmp_extruder] = 0;
// make sure all extruders have some sane value for the filament size
for (int i=0; i<EXTRUDERS; i++)
if (! filament_size[i]) filament_size[i] = DEFAULT_NOMINAL_FILAMENT_DIA;
}
}
else {
//reserved for setting filament diameter via UFID or filament measuring device
return;
}
calculate_volumetric_multipliers();
}

Pushing the zero value to the filament size on disable resolves the problem. The zero values are seen to write to prom, and M503/M501 report back "Filament settings disabled"

I backed that out, and set configuration_store.cpp to always return zero for diameter if it sees volumetric_enabled as false:

for (int q = 0; q < 4; q++) {
EEPROM_READ_VAR(i, dummy);
SERIAL_ECHO_START; //Dawson
SERIAL_ECHOPGM("Hit volumetric read");
SERIAL_EOL;
if(volumetric_enabled) {
SERIAL_ECHO_START;
SERIAL_ECHOPGM("Volumetric enabled is true in read");
SERIAL_EOL;
}

// Dawson - conditional to return 0 when disabled . . .
if(!volumetric_enabled) {
dummy = 0;
SERIAL_ECHO_START;
SERIAL_ECHOPGM("Volumetric Disabled in read");
SERIAL_EOL;
}
if (q < EXTRUDERS) filament_size[q] = dummy;
}

This, interestingly, does NOT work - M501 right after M500 comes back with a status of volumetric_read is enabled, right after watching it set to false in the write. It can't be a bad prom, since when the value written is zero, it does return disabled . . .

So, I added another trace point, to output the value of i, the EEPROM offset at the time of an operation, and the result was interesting, to say the least . . .

  SERIAL_ECHO_START;
  SERIAL_ECHOPGM("volumetric_enable offset is: ");
  SERIAL_ECHO(i);
  SERIAL_EOL;

EEPROM_WRITE_VAR(i, volumetric_enabled);

  SERIAL_ECHO_START;
  SERIAL_ECHOPGM("volumetric_enable offset is: ");
  SERIAL_ECHO(i);
  SERIAL_EOL;
EEPROM_READ_VAR(i, volumetric_enabled);

On write, I see a single write to location offset 377 . . .

On read, I see a single read from location 389 . . .

Well, this seems to explain a lot . . . Putting the same trace on the diameter read/write, I see a similar offset . . . 378, 382, 386, 390 for write, and 390, 394, 398, 402 for read. Since the code write all four values, and I have 1 extruder, that explains how I get the value back - 390 is common to both, but on read, volumetric_enabled is reading a latter byte of the third value, so when values are zero, it unsets.

<* Insert much sleepless debugging here . . . *>

I found it! Simple, yet incredibly annoying, and yes, this has been fixed in later builds . . .

    for (int q=4; q--;) EEPROM_READ_VAR(i, dummy); // Ki, Kd, Kc, Km

Was the culprit! The write routing (as well as later code) corrected this to 3 . . . which gets rid of the 4 byte offset, and now it WORKS!

Thanks for your time (and sorry for wasting it . . . a one character change in a section of configuration_store.cpp that wasn't even related to volumetric extrusion . . . grrrr . . .

But I know the Marlin source a heck of a lot better now!

Feel free to close this as already fixed/NPF . . .

thinkyhead, 2015-07-24:
> wasn't even related to volumetric extrusion

Yes that can happen if the EEPROM read or write is is off anytime earlier. The only way to catch such bugs is to read through the storage code and look for mismatches. I think we're getting better at avoiding that problem lately, commenting the code more, etc.

closed 2015-07-24 by thinkyhead