MarlinFirmware/Marlin #2499: Gcode : Performing one more task after each gcode

PisalPrasad123, 2015-07-23:
Hi , 
I am planning to do following tasks
[1] Use the STL file. Print one line of Gcode , then trigger the cooling fan
[2] Print the next line of gcode and then again trigger the cooling fan

printing and fan trigger will be executed one after another. 

Please let me know the suggestions for the same.

AnHardt, 2015-07-23:
@PisalPrasad123 
For what should it be good for?

PisalPrasad123, 2015-07-23:
may be power management

AnHardt, 2015-07-23:
Why not feeding in the wanted g-code sequence?

PisalPrasad123, 2015-07-23:
do you mean...Gcode Nth line and N+1th line. There won't be any break in between.
From line one to line N
It will be printing continuously.

AnHardt, 2015-07-23:
What means trigger?
Set to certain PWM value? Set to variable PWM value? Switching on?

PisalPrasad123, 2015-07-23:
trigger : start or change the speed as per our requirement or stop it for a while

we can keep the speed of fan variable. 
when we try to print anything in layer by layer. 

At some point , we need to extrude less material and sometimes , it needs to extrude more material.
Increasing fan speed will take more power, and decreasing fan speed will take less power.

AnHardt, 2015-07-23:
If you want to set the fanspeed by g-code - just do it - the fan will keep its speed until you change it. Setting it after every g-code is overkill.

If you want an automatic (firmware calculated) adjustment of the fanspeed  this will be quite complex.
It would depend on width, length, height of the track we lay out. Also we than want to know if we are close to the heatbed. Would not be bad to know the time how long a certain point is open, before next layer will cover it. As the fan is not cooling a certain spot, but an area, adjusting its speed for a single spot or track makes no sense. A job for the slicer i think.

closed 2015-07-24 by thinkyhead