MarlinFirmware/Marlin #2500: E3D Cyclops

stanrob, 2015-07-23:
Hi,
I just bought a E3D Cyclops nozzle and I need tout help to set Marlin setting to use it.
It is a single nozzle with two extruder motor (mixing extruder) 
I didn't find anything to do tgis by myself si if someone know how can I do it will be awsome 
Thank's

AnHardt, 2015-07-23:
Same as #1613.

closed 2015-07-23 by AnHardt

stanrob, 2015-07-24:
You closed this issue but I don't have my answer. I don't find the soution in #1613 topic

AnHardt, 2015-07-24:
@stanrob 
Closing this one and pointing to an other means.
We already have an issue with the same topic. Continuing this issue here will split the available information to different places.

The original poster of #1613 at least gave a partial solution.
MagoKinbra advertised his branch.

Please make further comments in #1613 .