MarlinFirmware/Marlin #2501: How to add a boot command in Marlin

linjinc, 2015-07-23:
Where should change

if (led_off > 0)
{
enqueuecommands_P((PSTR("M42 P11 S255")));
}
if (led_off < 1)
{
enqueuecommands_P((PSTR("M42 P11 S0")));
}

Wackerbarth, 2015-07-23:
What is the source of "led_off"?

Just what are you trying to accomplish?

AnHardt, 2015-07-23:
At the end of setup() in Marlin_main.cpp could be a good place. But setting the pin directly seems to be better - enqueuecommands_P has a monstrous overhead.

AnHardt, 2015-07-23:
The handling of 
POWER_SUPPLY, PS_DEFAULT_OFF, PS_ON_PIN, powersupply, setup_powerhold(), 
could be the full featured version of what you are looking for.

linjinc, 2015-07-24:
I want to control the LED switch

Led_off I have been added to the EEPROM, you can use, but the control switch I will only use M42

LED_PIN 11

-------------------------------
EEPROM_WRITE_VAR(i,led_off);
EEPROM_READ_VAR(i,led_off);
led_off = LEDOFF;
float led_off;
extern float led_off;
#define LEDOFF 1
---------------------------------
static void lcd_control_motion_menu()
{
    START_MENU();
    MENU_ITEM(back, MSG_CONTROL, lcd_control_menu);
#ifdef ENABLE_AUTO_BED_LEVELING
    MENU_ITEM_EDIT(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, 0.5, 50);
#endif
    MENU_ITEM_EDIT(float3, MSG_LEDOFF, &led_off, 0, 1);
---------------------------------

thinkyhead, 2015-07-24:
In that case you can use:

    MENU_ITEM(gcode, MSG_LEDOFF, PSTR("M42 P11 S0"));
    MENU_ITEM(gcode, MSG_LEDON, PSTR("M42 P11 S255"));

…or…

    MENU_ITEM(function, MSG_LEDOFF, led_off);
    MENU_ITEM(function, MSG_LEDON, led_on);
    ...
    bool led_is_on = false;
    void led_off() {
      enqueuecommands_P(PSTR("M42 P11 S0"));
      led_is_on = false;
    }
    void led_on() {
      enqueuecommands_P(PSTR("M42 P11 S255"));
      led_is_on = true;
    }

linjinc, 2015-07-24:
thanks

但是我应该修改那个文件

-----------------------------------------
bool led_is_on = false;
void led_toggle() {
  led_is_on = !led_is_on;
  if (led_is_on)
    enqueuecommands_P(PSTR("M42 P11 S255"));
  else
    enqueuecommands_P(PSTR("M42 P11 S0"));
}
...
if (led_is_on)
  MENU_ITEM(function, MSG_LEDOFF, led_toggle);
else
  MENU_ITEM(function, MSG_LEDON, led_toggle);

closed 2015-07-25 by linjinc