MarlinFirmware/Marlin #2502: Possible issues in #ifdef blocks for CoreXZ

grytole, 2015-07-23:
File: stepper.cpp (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L377)

Block at lines 377-379 looks like a bug. It tries to add '}' after Y axis processing for CoreXY (correct) and for CoreXZ (incorrect).

Only CoreXY check should be left in this block.
Same check for CoreXZ should be added for Z axis (missed now).

pdbogen, 2015-07-23:
This looks like it was indeed a copy-paste error. If you have a COREXY device, can you make the appropriate changes, do some smoke testing, and then submit a pull request? (I could change and submit PR, but I don't have a way to verify the code actually works for COREXY other than by inspection.)

grytole, 2015-07-23:
Probably related:  in planner.cpp at lines 605-609 there is no Y axis activation like in other configurations.

Looks that Y axis will be permanently disabled for CoreXZ build.
It just needs a copy of line 612.

grytole, 2015-07-23:
Just tried G28 on CoreXZ setup - with d746c4f692d4eac98233977c791d891f583b27fd homing works as expected, but Y axis is dead without change in planner.cpp.

thinkyhead, 2015-07-24:
@grytole You've got that right. I've got a patch for it now at … #2509.

closed 2016-04-23 by jbrazio