MarlinFirmware/Marlin #2511: Endstops unresponsive

RippinRoo, 2015-07-24:
Hi there,

I've recently aquired an old reprap mendel printer and am currently trying to get it up and running (it was in a pretty awful state, but all wiring and circuitry etc seems in perfect working order). Although I was told the firmware I was given was the most up to date (printer was assembled 3 years ago I believe) it's having some serious issues with registering the endstops I think. Basically when I tell the system to autohome it will just smash into the endpoint switch and keep grinding to try and push past it (for x y and z it seems). This is my first time dealing with actual printer firmware, and I have solved a certain amount of issues with it but I can't seem to find the reason for this. I can include the endstop code below if that helps? I'm pretty sure it's the Marlin v1 firmware it's based on.

Thanks for any help received.

eboston, 2015-07-24:
Have you tested the actual endstops to verify they work? Also check the
wiring to the board doesn't have an internal break. If they used to work
with the firmware, but don't now, sounds like it is a good chance something
other than the firmware has failed. Endstop, wires, controller board.

AnHardt, 2015-07-24:
@RippinRoo 
Before you begin to post code, lets begin with the basics.
You can test your endstops with the M119 command. I that is giving unlogic results begin to play with ?_?_ENDSTOP_INVERTING and ENDSTOPPULLUP_? (in Configuration.h). 
A link to your Configuration.h will be helpfull to determin your version and obvious configuration errors.

RippinRoo, 2015-07-24:
Thanks for the quick replies. Here is a link to my configuration file as it is

https://dl.dropboxusercontent.com/u/26550015/Configuration.h

I've tested all the wiring etc with a multimeter, everything seems to be fine. It was printing perfectly until supposedly the printer had the wrong firmware installed (actual ultimaker firmware in an attempted and unnecessary upgrade) and it crashed it. So I'm bringing it back to life hopefully.

I ran the M119 code there, it shows some info on the settings etc but no errors and nothing looks out of the ordinary, although it's mostly unfamiliar to me.

Roxy-3DPrintBoard, 2015-07-24:
You can manually move your nozzle towards the end stops and away from the end stops.  You should see the status change on the M119 command.   You should check all 3 axis to see what is being reported.

pdbogen, 2015-07-25:
@RippinRoo Can you clarify what version (i.e., what git branch / tag) you're attempting to build and install? It might be worth throwing away whatever source code you were given and starting from scratch with tag 1.0.2-1, the latest blessed release.

thinkyhead, 2015-07-26:
@RippinRoo If you don't have any max endstops, be sure to uncomment the setting `#define DISABLE_MAX_ENDSTOPS` in `Configuration.h` as a first step.

RippinRoo, 2015-07-27:
@thinkyhead thanks so much, your idea of changing the #define DISABLE_MAX_ENDSTOPS worked a charm. I had some issues with the x-axis autohoming and moving in the wrong direction but I was able to get it all sorted now by changing a few other settings. The firmware I was given was very close to what was needed thankfully. The printer is just after completing it's first run and it looks great, so thanks so much for the assist.

RippinRoo, 2015-07-27:
Also thanks to the suggestions from the other people here, I didn't have a chance to try them before using @thinkyhead solution, but thanks for taking the time to help.

closed 2015-07-27 by RippinRoo