MarlinFirmware/Marlin #2517: Is 1.1 considered "stable"?

Diaoul, 2015-07-25:
Or should I use Development branch?

AnHardt, 2015-07-25:
1.1 is stable in the sens of - it will not be changed anymore.
1.2 has a very little chance to get further bug fixes, but will get no new features.
1.3 dev is changing all the time.

If we are talking about "has less bugs" i recommend 1.3 dev (most of the time)
If we are talking about "is best for everyday use" i can't  give a clear answer today. It depends on your machine, the features you want to use and your workflow.

closed 2015-07-25 by Diaoul

Wackerbarth, 2015-07-25:
@Diaoul -  I think that @AnHardt has a "typo" in his response.
1.0.1 longer has maintenance.
1.0.2 may get some bug fixes, but will have no additional features
1.0.3 is a "development" branch and currently incomplete.  But it is likely to never be released under that designation.
The 1.1 tag in the repository is NOT a release (or release candidate) DO NOT USE IT.
The work on that branch is available on "Development" (or, equivalently, "dev". It is nearing a point of being ready for testing. However, there are still issues about bed leveling.