MarlinFirmware/Marlin #2518: Endstop abort not working on 1.0.2-1

ronhogwarts, 2015-07-25:
I have enabled the #define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED in the configuration_adv.h file and manually on the printer under "control> motion> endstop abort: on" and then started a print and manually triggered the sensors with a magnet, but nothing happens. Homing works fine and a M119 shows all endstops trigger correctly, yet triggering them during a print does nothing. Any ideas?

pdbogen, 2015-07-25:
Hm, in https://github.com/MarlinFirmware/Marlin/blob/1.0.2-1/Marlin/stepper.cpp#L194; the abort on endstop hit code is only enabled if you also have SDSUPPORT enabled. Note also that the feature is described as "Enable the option to stop SD printing when hitting and endstops, needs to be enabled from the LCD menu when this option is enabled," which may not be what you expected.

Do you have SD card printing enabled? Were you printing from SD when you ran your test?

ronhogwarts, 2015-07-25:
I was printed from the SD card at the time, and hence it was obviously enabled. I also changed https://github.com/MarlinFirmware/Marlin/blob/1.0.2-1/Marlin/stepper.cpp#L73 to true, so it is enabled by deafult, so I don't have to re-enable it each time the printer reboots.

pdbogen, 2015-07-25:
OK, great. Just checking the basics.

Did you see an LCD and/or SERIAL message when you triggered the endstop? (this will indicate if the problem is in the "stopping" part or in the "endstop" part.)

ronhogwarts, 2015-07-25:
I didn't see any other messages at the time.

pdbogen, 2015-07-25:
So it sounds like the problem isn't in the abort code, it's in the code that sets flags when endstops are triggered. Just as a further sanity check while I dig into where the problem might be, can you confirm you _weren't_ testing this as part of a "home" operation?

pdbogen, 2015-07-25:
Another thing to check: Did you comment out ENDSTOPS_ONLY_FOR_HOMING here: https://github.com/MarlinFirmware/Marlin/blob/1.0.2-1/Marlin/Configuration_adv.h#L89 ?

ronhogwarts, 2015-07-25:
Ah I didn't comment that out, I'll check tomorrow if this fixes it. Should it not be commented out by default? Surely most people want the printer to stop if it hits the endstops, to avoid mechanical damage? On a similar note, why does the option for aborting when it hits the ends stops have to be enabled manually on the printer under "control> motion> endstop abort: on" on each startup?

pdbogen, 2015-07-25:
I can't speak to the whys or shoulds, sorry. I *did* submit a PR to at least document that the ABORT_... function requires either enabling endstop checking with M121 or disabling ENDSTOPS_ONLY...; so at least this problem should be more visible in the future.

If I had to guess, I'd suppose the concern is around the extra work required to poll 3/6 endstops very frequently, while (presumably) managing heaters, steppers, and trying to plan future moves as well. There's a lot going on for the little Arduinos Marlin targets. :/

ronhogwarts, 2015-07-26:
Well commenting out the below, as you suggested, does indeed work, for anyone else who wonders. Thanks for the help.
 https://github.com/MarlinFirmware/Marlin/blob/1.0.2-1/Marlin/Configuration_adv.h#L89

closed 2015-07-26 by ronhogwarts

thinkyhead, 2015-07-27:
To get `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED` to work during an SD print, you need add the following GCode to your prefix. It must be added _after_ any `G28` or `G29` because these commands will disable your endstops if `ENDSTOPS_ONLY_FOR_HOMING` is enabled in your firmware.
```gcode
M540 S1 ; enable abort on endstop hit
M120 ; enable endstops during printing
```