MarlinFirmware/Marlin #2519: Feature Request - G2 G3 True Arc Movement

Jzatopa, 2015-07-25:
Current Marlin Behavior
"By default G2 will draw arcs with 1mm segments, so arcs below that threshold are not recommended."

I would like it if marlin could draw in .1mm or smaller segments or move as close to a true arc as possible.  I know this is limited by the processing power of the hardware marlin runs on but I think that this could be optimized.

thinkyhead, 2015-07-26:
What kind of results do you get if you change the segment size? By default it is 1mm, but you can change this value in `Configuration_adv.h` to 0.1 if you prefer:

    #define MM_PER_ARC_SEGMENT 1

Jzatopa, 2015-07-27:
I didn't realize this was in Configuration_adv.h there doesn't seem to be a lot of information available concerning G2 and G3.  I am going to close this.

closed 2015-07-27 by Jzatopa

thinkyhead, 2015-07-27:
> there doesn't seem to be a lot of information available concerning G2 and G3

They're pretty new to the codebase. I will add notes about the setting and the way they work to the Marlin wiki pages on [G2](http://www.marlinfirmware.org/index.php/G2) and [G3](http://www.marlinfirmware.org/index.php/G3).

rosicn, 2018-03-08:
This codes does not work on Marli 1.1.8

thinkyhead, 2018-03-08:
@rosicn — Make sure you have `ARC_SUPPORT` enabled.
If `G2` and `G3` still don't work, I guess that will be a permanent feature of 1.1.8.
The codes do work in the `bugfix-1.1.x` and `bugfix-2.0.x` branches last we checked.
The `bugfix-1.1.x` branch will soon be released as 1.1.9.

rosicn, 2018-03-08:
@thinkyhead thank you for fast response. Yes that was solution, enabling ARC_SUPPORT in configuration_adv.h
I spent a lot of time searching on the internet without success. After checking this file I found solution.
I made laser mod on 3d printer and ARC_SUPPORT is necessary for Inksape gcode output.