MarlinFirmware/Marlin #2524: Shapingbits 3fxtrud illegaly witholding marlin source code

mistsoul, 2015-07-26:
I have purchased a 3fxtrud 3d printer from a company called shapingbits a few months ago (prior to the kickstarter campaign). The printer uses marlin and i need to add additional extruders and change thermistors amongst other upgrades i am making.
 
I have requested the modified marlin source code which corresponds to my machine from the manufacturer who claims that the machine is open source, however as he is not happy that i am making upgrades to the machine he has declined providing me with the source code which i need in order to move forward with my work.

I understood that he is required to provide me the modified code based on the gnu/gpl rules which state:

http://www.gnu.org/licenses/gpl-3.0.en.html

''Our General Public Licenses are designed to make sure that you have the freedom to distribute copies of free software (and charge for them if you wish), that you receive source code or can get it if you want it, that you can change the software or use pieces of it in new free programs, and that you know you can do these things."

It is black and white in the excerpt above, by not providing the source code shapingbits is in infraction of the parameter describing that i am entitled to the source code if i want it.

The owner won't share what types of key components are used such as controller board and motors, so for me acquiring the source code as a base to begin my changes and to be able to revert back to the original source code setting if i need to in the future, is a must, however shapingbits believes that they are above the rules, when other leading 3d printer companies like ultimaker, solidoodle, vellemen, tazz to name a few fully comply with the gnu/gpl rules.

I wanted to make sure other users of this machine are aware of the restrictions that shapingbits has imposed on its customers by restricting the availability of the source codes which the customer is fully entitled to. Some will never need this but for those planning to make changes and upgrades, you will have to start from scratch and that is utterly unfair and illegal given that you paid for the machine in full, which includes all components as well as the modified source code if you wish to have it.

fmalpartida, 2015-07-26:
If the machine is not open source, i.e. All its sources are available to modify, they are not entitled to use Marlin. 
If they use Marlin and don't share their modifications or use it and don't acknowledge its use, they are not entitled to use Marlin and in clear breach of its licensing terms and conditions.

fmalpartida, 2015-07-26:
I wonder how many others out there are also infringing the license.

Nprod, 2015-07-26:
Can you post a picture of the electronics? Judging by the shape of the box it looks like a standard Mega2560/RAMPS1.4 type deal with a ReprapDiscount smart controller LCD. None of these kickstarter companies design their own electronics. It may take a few tweaks and configs, but you should be able to install stock-Marlin on there. Worst case scenario is you'd have to spend ~10$ on a couple of new thermistors if you can't find out the type of the current one.

And yes, according to the GPL they are obligated to publish their changes. So they're just being overprotective open-source leeches. However, i doubt they changed the firmware much, apart from configuring it.

clefranc, 2015-07-26:
@mistsoul Interesting reading: www.rosenlaw.com/pdf-files/Rosen_Ch12.pdf

@fmalpartida And what is the legal action against such acts? After some readings, it seems the "owner" (or even collaborators) of such open source software can have leverage on misuse of their ownership (or their work). Am I right?

Is there precedents where the use of Marlin was denied to a licensor?

This kind of story pisses me off because I was too victim of GPL abuse.

fmalpartida, 2015-07-26:
I was also a victim of an infringement.
Without going into legal stuff and seeking a lawyer the most pragmatical approach is to have a Marlin blacklist and publish the infringements in social networks. 
I have a good list of printers that use Marlin and are not open source. 
Bad publicity is one way about it.

fmalpartida, 2015-07-26:
Including variations of i3 with new mechanical addons for which they haven't released the sources.

Roxy-3DPrintBoard, 2015-07-26:
You can notify any established player they are using for their sales and distribution.  If it is Kick Starter, or eBay or PayPal...   You send a detailed message to their legal department requesting they no long support or facilitate this company's sales because they are in violation of the licence agreement.  It would almost be the same thing as telling eBay somebody is selling copyrighted videos or songs.  The person doing it clearly does not have the right to do it. 

My guess is they will get into compliance so that they can use the established player's services.

clefranc, 2015-07-26:
@fmalpartida Excellent idea! The information about Marlin being open source (who, why, how, etc.) should be put on the main page of http://www.marlinfirmware.org/, and of course with a link to a nice page of the deviants.

The link to the wiki should also be put on the LCD splash screen and output to console (M115). I think there also should be an About submenu in the main menu with info about version and build, including the wiki link.

@thinkyhead Can it be added to your to do list?

fmalpartida, 2015-07-26:
I think that could work well. For those that are confirmed the could be showcased in Marlins main repo. 

License infringement section and we should add a big "thank you" for supporting the project by infringing is terms of use and license.

mistsoul, 2015-07-26:
Thank you all, it is good to hear that i am not alone in this, my next step was to reach out to Albuquerque's attorney office who deals with consumer complaints, they have a process to file complaints and i plan to do so next week if i still don't receive the source code. I like your ideas also though, to get this out in the public. 

My aim is not to damage their reputation as i hope that they will realize that what they are doing is illegal and if they don't, karma eventually catches up to you one way or the other... What i wanted to do was to alert other users or potential users that are more hands on, of what they are getting into if they purchase a 3fxtrud from shapingbits, so i think the suggestions above would serve this purpose well.

fmalpartida, 2015-07-26:
This oils be a good topic for the next HO. How to enforce Marlin's license. 
Weak so have to recognize those who want to keep their spice closed, so this could be a nice option to start getting a right to use license.

mistsoul, 2015-07-26:
For the kickstarter suggestion provided earlier, i would not be able to file a complaint, the requirement is for one of the marlin developper to file the complaint of infringement at https://www.kickstarter.com/help/forms/dmca/new

eboston, 2015-07-26:
I think contacting Kickstarter won't do any good as the campaign has 
ended and the money would have already been sent to them.  Not much more 
Kickstarter can do at this point.

mistsoul, 2015-07-26:
True, those waiting to receive the printers could be alerted, but at least some may stumble on this thread, and the information may be useful to them.

eboston, 2015-07-26:
Have you thought about posting a question to them in the comments page 
of the Kickstarter asking about this?  That would bring a lot of 
attention to it.

mistsoul, 2015-07-26:
I cannot post as my purchase is not kickstarter related so i don't have access to that forum.

thinkyhead, 2015-07-27:
> Can it be added to your to do list?

@clefranc I will add some notes about the open source nature to the front page, and a bold paragraph that is explicit about the obligations of anyone who uses Marlin as firmware for their printers.

@mistsoul I have contacted ShapingBits (through their YouTube channel) to inform them of their violation of the GPL and that they are at risk of having their license to distribute Marlin revoked. If they fail to comply and inform us, I will put them onto a "blacklist" page on the Marlin wiki.

foosel, 2015-07-27:
If they don't rectify the situation promptly, I'm fairly sure there is a way to build up some pressure through social networks. The g+ community on 3d printing alone has over 180k followers now. Keep us in the loop please. Violations of open source licenses are unacceptable and infuriating.

fmalpartida, 2015-07-27:
@foosel I fully agree with you. Specially since a good range of the OS licenses are so easy to comply with. 
Furthermore, it is a way to respect the desire of the creator or group of creator/supporters. In many cases for free. 

This will surely make an interesting topic for the next meeting.

mistsoul, 2015-07-27:
Thank you all, i will try to get the source code again and will keep you all updated!

mistsoul, 2015-07-27:
@thinkyhead if they fail to comply, would you also be able to file a complaint with kickstarter via the following form. This should inform the backers who are waiting to receive a machine identical to mine.

https://www.kickstarter.com/help/forms/dmca/new

thinkyhead, 2015-07-27:
@mistsoul Surely! Just keep us posted here.

mistsoul, 2015-07-28:
success! i was able to get the source code. They have it for download on their website but somehow misunderstood my request and thought that I wanted them to modify the code for my new setup. It is not clear whether this was just added or if it has been there all along but I can move forward with my work now. Thanks you all for the support and I will let you know if I need help moving forward with my modifications.

fmalpartida, 2015-07-28:
I pull back some suggestions after reading through, again, the gpl license. 
I think I have been fully misled by the contents of the licensing section and as me many other are surely going to be too. 
I suggest one of two things:
- remove altogether and just leave the GPL license without any additional wording. 
- ask kindly to donate to support the project for closed sourced commercial projects if that was the original intention of the message. But here I see many corner cases which I have already shared in the different HO and I don't like it.

thinkyhead, 2015-07-29:
@mistsoul I have a reply from Bogdan at ShapingBits. It's rather long, but the gist of it is that the configuration and link to Marlin are posted on the site, but you have to have a login. I created an account for myself, and indeed there are links in the main menu:

[image]

I rather annoyed Bogdan with my inquiry! So my only question is, does this comply with the GPL as we understand it? In order to gain access all I had to do was give ShapingBits an email address, name, street address, and phone number. This doesn't put GPL code behind a "paywall" so I don't have any major problem with it. It's just a minor inconvenience.

mistsoul, 2015-07-29:
Thanks for following up and sorry for the trouble, i was able to download it as well finaly. Initially when i asked i was told several times that it could not be provided and that i just had to download the code from the marlin website and figure everything out on my own as a good learning experience, in a rather condescending tone. 

However he had perhaps misunderstood my request, he said that he thought that i was asking him to modify the code for my new setup. It is not stated anywhere in his user manual that the code is downloadable from his site and that you have to log in first, so i advised that this be clarified for other users. 

I personaly don't have an issue with it being available only with access to his site since most people who would be looking for it would have one of his machines and therefore access. We can consider this case closed.

closed 2015-07-30 by thinkyhead