MarlinFirmware/Marlin #2530: Make Serial1 Available via MarlinSerial or drop MarlinSerial

pdbogen, 2015-07-27:
MarlinSerial etc. seems intended to support ATtiny-based devices lacking a hardware UART. I have no idea how it's supposed to accomplish this, since it's just a copy-paste of an old HardwareSerial.h and HardwareSerial.cpp; in particular, it only stores characters onto its buffer as a result of M_USARTx_RX_vect (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/MarlinSerial.cpp#L53), which won't happen without a UART.

What it *does* do is *not* implement any of the other UARTs that might be present on a microcontroller, like the Serial1 on my Sanguino, or the Serial2 and Serial3 on an Arduino Mega.

thinkyhead, 2015-07-27:
What is required to support the other serial interfaces? Do you want Marlin to be able to use them for serial printing?

Wackerbarth, 2015-07-27:
There is a very serious problem supporting Sanguino, etc. for ANY serial operation because the main ISR for the steppers thinks that it knows what is happening on the serial port(s) and gets involved.
When the core serial handlers don't know about the stepper ISR, things quit working.

That's why, even while I build on the latest IDE 1.6.6, I'm still building my AT90USB-based versions on a core derived from an Arduino 1.0 core.

Roxy-3DPrintBoard, 2015-07-27:
> But my primary effort is currently working on an advanced bed leveling that will work for deltas and "warped" Cartesian bots.

That sounds good!   How long until there will be code to test?   I have some small warps I can put to good use with this code.

pdbogen, 2015-07-27:
For my part, I want to use Serial1 so that the controller board can utilize a slave board for additional extruders/heaters/thermometers. I extending MarlinSerial would mostly require extending the preprocessor macros to understand different sets of registers for the additional serial ports; I should be able to do this work.

I guess the involvement you're referring to is the checkRx; I gather this is because the stepper interrupt lasts so long that we would otherwise miss serial events.

(Hm. Reading, there might be a bug in checkRx; since according to the datasheet, reading UDRx tells the shift register to shift in a new value, and we read UDRx before we know if the serial buffer has space, we may be unnecessarily discarding characters; but I'm not sure how big the hardware shift register is (one byte? more?) so it may not make a difference; if being unable to accept the byte means we're spilling it out of the wire instead of out of the register<->buffer connection; or maybe we're hitting this fast enough that we check multiple times per byte even at a 250,000 baud rate)

thinkyhead, 2015-07-27:
> can utilize a slave board
:+1: So that's interesting. A `MARLIN_SLAVE_MODE` setting would allow Marlin to be built with only the parts you want to be able to control, plus an alternative serial connection for GCode (I assume) and slave messages. I like it.

Wackerbarth, 2015-07-27:
I'm a little concerned about synchronization of the various actuators. Wouldn't there be a delay between the local movements and the remote ones?

pdbogen, 2015-07-28:
The RepRapPro design (for which I'm implementing modern firmware support) uses a dedicated pin on either end to step the remote stepper; so the serial link is used to enable/disable, set direction, get temperature readings; but not to directly control steppers. It triggers the pin change interrupt for the pin on the slave (https://github.com/pdbogen/Marlin/blob/newmarlin/Slave/Interrupt.h) so it's very fast. (Enabling/disabling is slower than normal, of course, but this happens rarely; and with a 250000baud connection, sending across a 10-byte packet doesn't take all that long.)

(The basic design is fine; the printer worked with RRP's firmware, but their implementation of serial communication was buggy and it was built on a rather old Marlin, so I'm working on fixing both of those issues.)

(My design doesn't send GCode across the serial line, though there's no reason you _couldn't_; rather, it sends multi-byte fixed-length packets containing a "command" field and a payload, which means the slave doesn't have to do any parsing; just a proper CRC16 to ensure the packet was received, and then it's just pumped bit-by-bit into an identical struct and accessed directly.) 

So yeah, in my case it's two Melzis (sanguinos with built-in stepper drivers) that can manage three extruders plus a bed heater.

thinkyhead, 2015-07-28:
> the slave doesn't have to do any parsing

Even better!

closed 2016-04-23 by jbrazio