MarlinFirmware/Marlin #2532: LCD_PROGRESS_BAR ignores errors

Loader009, 2015-07-27:
The function "LCD_PROGRESS_BAR" let the LCD display a progress bar (how obvious).
If there is an error (e. g. THERMAL RUNAWAY) the progress bar stays there.
The error is not displayed (had only once during print with SD).

PS: Can stepper driver cause the "THERMAL RUNAWAY" error?
If yes, we need something to distinguish what exactly is causing that error.

thinkyhead, 2015-07-27:
I did the original implementation of `LCD_PROGRESS_BAR` and it's supposed to show any message that is set right away, then show the progress bar again only after a delay. But perhaps the status screen isn't getting updated enough once the `kill()` function is called. On my machine I haven't seen this issue, but I use the default settings…
```cpp
    // Amount of time (ms) to show the bar
    #define PROGRESS_BAR_BAR_TIME 2000
    // Amount of time (ms) to show the status message
    #define PROGRESS_BAR_MSG_TIME 3000
    // Amount of time (ms) to retain the status message (0=forever)
    #define PROGRESS_MSG_EXPIRE   0
    // Enable this to show messages for MSG_TIME then hide them
    //#define PROGRESS_MSG_ONCE
```

> Can stepper driver cause the "THERMAL RUNAWAY" error?

No, it only applies to the hot-end and heated bed. Thermal runaway will be thrown if the heater takes too long to reach temperature (e.g., if you _only_ power your board over USB without the PSU and issue `M104 S80` you will get this error). It will also throw this error if the temperature is seen to climb when the heater is off.

thinkyhead, 2015-07-27:
@AnHardt I'm seeing something weird now. I changed `WATCH_TEMP_PERIOD` to 50 so that `THERMAL_PROTECTION_PERIOD` will be triggered first. Then I issued `M104 S80` with only USB power, so the heaters will stay cold. The hot-end temperature reads __21/80°__. After 6 minutes in this state I still have no error. But I did notice that it took me several clicks before the controller exited the Status Screen. Once it finally did switch to the menu screen, clicks resumed working properly. But then after another couple minutes, it takes several clicks again.

Clearly we still have some bad logic around thermal protection.

AnHardt, 2015-07-27:
@thinkyhead 
Jep. WATCH_TEMP_PERIOD-test fails when > 32. 
WATCH_TEMP_PERIOD is later multiplied with 1000. Multiplying with 1000UL solves the problem.
PR is on its way.

Sorry could not reproduce any problems with clicking the button.

thinkyhead, 2015-07-28:
> could not reproduce any problems with clicking the button

It's a weird one. Of course now your patch fixes that too.

Loader009, 2015-07-28:
I use also the default settings for the progress bar.
It is possible that the progress bar appears again after a message has been shown so that the message won't be visible. The progress bar didn't disappear again after a THERMAL_RUNAWAY (I haven't noticed the "stop" of the printing. After I came back it has already stopped and no error message but the progress bar was shown.)

To the THERMAL_RUNAWAY thing:
Afaik the protection only activates after the temperature has been reached. So the initial heat-up shouldn't make any THERMAL_RUNAWAY.
If you want to reproduce the LCD_PROGRESS_BAR-bug(?) you need to reach the temperature for getting a THERMAL_RUNAWAY error.

PS: Noticed a bug after THERMAL_RUNAWAY (the stepper motors are moving very slowly/nearly nothing). I will search for that and make sure I don't report a duplicate bug.

thinkyhead, 2015-07-29:
> only activates after the temperature has been reached

True, and easy to forget. It's the `WATCH_TEMP_PERIOD` stuff that catches the failure to heat up.

thinkyhead, 2015-07-29:
> you need to reach the temperature for getting a THERMAL_RUNAWAY error

I will try getting the temperature up, then try (1) unplugging the thermistor and (2) turning off the heater power.

closed 2016-04-23 by jbrazio