MarlinFirmware/Marlin #2533: Tools for debugging Marlin?

PSimonovich, 2015-07-27:
I was trying to determine why ABL was not working as expected for my Markerfarm i3v printer by using the VisualMicro Debugger with Visual Studio 13 or Atmel Studio 6.2 software (Windows 7 64-bit). I am able to compile the Marlin release build successfully and download to the Atmega2560 board. However, when I try to compile a debug build, the compile fails as follows:

Compiling debug version of 'Marlin' for 'Arduino Mega w/ ATmega2560 (Mega 2560)'
Build folder: file:///C:/Users/Pete/AppData/Local/V.Micro/Arduino/Builds/Marlin/mega_atmega2560
VM_DBG.h:153:2: error: 'HardwareSerial' does not name a type
VM_DBG.h:154:13: error: 'HardwareSerial' has not been declared
VM_DBG.h:155:13: error: 'HardwareSerial' has not been declared
VM_DBG.h:In member function 'void VisualMicroDebug::setVariable(T&, int, int)'
VM_DBG.h:323:2: error: 'transport' was not declared in this scope
Error compiling

I'm guessing that its related to the MarlinSerial implementation of HardwareSerial module.
Has anyone resolved this issue or can suggest a work around?
I just received my Atmel-ICE programmer/debugger hardware and will try that next.

Wackerbarth, 2015-07-27:
Peter,

Yes, it is most certainly related to the implementation of HardwareSerial.
As I mentioned on another issue, the implementation of the stepper ISR has been imtimately merged into an Adruino 1.0 (or earlier) version of HardwareSerial. This prevents the use of “modern” Arduino cores at this time.
To get around the issue on my AT90USB-based boards, I have created my own version of “core” using the old serial driver and some of the parts of newer Arduino cores.

I don’t know how the VisualMicro Debugger ties into the Serial port. So, without further information, I may not be able to provide much help.

thinkyhead, 2015-07-27:
@PSimonovich I'd be very interested to hear whether serial-based debugging works. I've had to print out all the relevant variables to serial output in order to debug homing and bed leveling. If you would like to get detailed debug output for `G28` and `G29` for the current Development branch, you can use this branch in my fork:

https://github.com/thinkyhead/Marlin/tree/debug_G29

It has the latest code as of today.

PSimonovich, 2015-07-28:
Sure. I'll let you know when I have made further progress using the VisualMicro Debugger with either HardwareSerial or SoftwareSerial debug ports. I am also trying to attach my Atmel-ICE in-circuit Debugger to the Atmega2560 board which would also provide execution traces and profiling information.

PSimonovich, 2015-07-29:
I merged the debug_G29 code with my i3v (300x300mm) configuration, fixed the compilation errors and uploaded it to the printer. I didn't run the G29 command yet since the G28 Z-axis homing doesn't work - the servo probe is not deployed.
Marlin_main.cpp corrections:
line 1834:    set_axis_is_at_home(axis); // Set axis position to home plus offsets
line 1920:    if (servo_endstops[axis] >= 0) {
lines 3033-3037: 
        current_position[Z_AXIS] = -zprobe_zoffset + (z_tmp - real_z);
        #if ENABLED(Z_PROBE_ALLEN_KEY) || ENABLED(Z_PROBE_SLED) || ENABLED(Z_ENDSTOP_SERVO_NR)  // ?? ENABLED(SERVO_ENDSTOPS)
          current_position[Z_AXIS] = current_position[Z_AXIS] + Z_RAISE_AFTER_PROBING;
        #endif

For starters, here is the G28 debug output:
>>> G28
SENDING:G28
gcode_G28 >>>
enable_endstops(true)
>>> home_axis(0
> TRIGGER ENDSTOP > current_position: (0.00, 0.00, 0.00)
axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
> AFTER axis_is_at_home > current_position: (0.00, 0.00, 0.00)
axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
<<< home_axis(0
> homeX: (0.00, 0.00, 0.00)
>>> home_axis(1
> TRIGGER ENDSTOP > current_position: (0.00, 0.00, 0.00)
axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
> AFTER axis_is_at_home > current_position: (0.00, 0.00, 0.00)
axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
<<< home_axis(1
> homeY: (0.00, 0.00, 0.00)
> Z_SAFE_HOMING >>>
Raise Z (before homing) by 4.00
> home_all_axis > current_position: (0.00, 0.00, 0.00)
> home_all_axis > destination: (181.00, 158.00, 4.00)
>>> home_axis(2
SENDING:M84
> TRIGGER ENDSTOP > current_position: (181.00, 158.00, 0.00)
axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (181.00, 158.00, 5.20)
> AFTER axis_is_at_home > current_position: (181.00, 158.00, 5.20)
axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (181.00, 158.00, 5.20)
> SERVO_ENDSTOPS > Stow with write()
<<< home_axis(2
<<< Z_SAFE_HOMING
> (home_all_axis || homeZ) > final: (181.00, 158.00, 5.20)
ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
<<< gcode_G28

Observations:
------------
The probe never deployed prior to the G28 Z homing. I manually pushed the probe endstop button
to prevent the hotend from crashing into the bed.

The probe servo also shows significant jitter during Z-axis up/down movement.

The probe did work correctly in a past development build (need to backtrack to find out when it
became inoperative).

I'll send the G29 results after I find out why the probe did not deploy.

thinkyhead, 2015-07-29:
Patched the compile error. Reading your logs…

thinkyhead, 2015-07-29:
The patch to 3033-3037 should actually just replace `ENABLED` with `defined` for `SERVO_ENDSTOPS`:

      current_position[Z_AXIS] = -zprobe_zoffset + (z_tmp - real_z)
        #if defined(SERVO_ENDSTOPS) || ENABLED(Z_PROBE_ALLEN_KEY) || ENABLED(Z_PROBE_SLED)
          + Z_RAISE_AFTER_PROBING
        #endif
        ;

Was your addition of `|| ENABLED(Z_ENDSTOP_SERVO_NR)` meant to replace my botched use of `ENABLED(SERVO_ENDSTOPS)`?

thinkyhead, 2015-07-29:
Here's the block in `homeaxis` where it should have deployed your probe:
```cpp
#if SERVO_LEVELING && !defined(Z_PROBE_SLED)

  // Deploy a probe if there is one, and homing towards the bed
  if (axis == Z_AXIS) {
    if (axis_home_dir < 0) deploy_z_probe();
  }

#endif
```
The `deploy_z_probe` function isn't even being invoked. Reasons why this may not happen in this block include:
- Failed to enable `ENABLE_AUTO_BED_LEVELING`
- Failed to enable `DEACTIVATE_SERVOS_AFTER_MOVE`
- Accidentally set `Z_PROBE_SLED`
- Accidentally set `Z_HOME_DIR` to +1

Continuing to look into other

thinkyhead, 2015-07-29:
@AnHardt To protect and servo! What do you think of this interesting requirement within `homeaxis`? If you don't have `DEACTIVATE_SERVOS_AFTER_MOVE` set then it won't try to deploy the Z probe. It seems like it should only require that there be a Z probe, and that it is the endstop. I'm actually a little unclear on which `Z_PROBE`, `ENDSTOP`, and `SERVO` settings combine to say "This machine only has a Z probe as its min endstop."

PSimonovich, 2015-07-29:
Yes.

PSimonovich, 2015-07-29:
Yes. The DEACTIVATE_SERVOS_AFTER_MOVE flag was unintentionally off. G28 and G29 operations are now operational. However, the hotend is approximately 5mm above the bed when the print begins!
The printer configuration and G28/G29 debug output log follows:
Print started at: 07:34:12
gcode_G28 >>>
enable_endstops(true)
>>> home_axis(0
> TRIGGER ENDSTOP > current_position: (0.00, 158.00, 13.19)
axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 158.00, 13.19)
> AFTER axis_is_at_home > current_position: (0.00, 158.00, 13.19)
axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 158.00, 13.19)
<<< home_axis(0
> homeX: (0.00, 158.00, 13.19)
>>> home_axis(1
> TRIGGER ENDSTOP > current_position: (0.00, 0.00, 13.19)
axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 13.19)
> AFTER axis_is_at_home > current_position: (0.00, 0.00, 13.19)
axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 13.19)
<<< home_axis(1
> homeY: (0.00, 0.00, 13.19)
> Z_SAFE_HOMING >>>
Raise Z (before homing) by 4.00
> home_all_axis > current_position: (0.00, 0.00, 0.00)
> home_all_axis > destination: (181.00, 158.00, 4.00)
>>> home_axis(2
deploy_z_probe > current_position: (181.00, 158.00, 0.00)
> TRIGGER ENDSTOP > current_position: (181.00, 158.00, 0.00)
axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (181.00, 158.00, 5.20)
> AFTER axis_is_at_home > current_position: (181.00, 158.00, 5.20)
axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (181.00, 158.00, 5.20)
> SERVO_LEVELING > stow_z_probe
stow_z_probe > current_position: (181.00, 158.00, 5.20)
Raise Z (after) by 8.00
> SERVO_ENDSTOPS > do_blocking_move_to_z 21.20
do_blocking_move_to: (181.00, 158.00, 21.20)
<<< home_axis(2
<<< Z_SAFE_HOMING
> (home_all_axis || homeZ) > final: (181.00, 158.00, 21.20)
ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
<<< gcode_G28

Note - Successful deployment of Z probe, without jitter.

gcode_G29 >>>
G29 Auto Bed Leveling
enable_endstops(true)
z_before = (before) 8.00
probe_pt >>>
> ProbeAction:1
> current_position: (181.00, 158.00, 21.20)
Z Raise to z_before 8.00
> do_blocking_move_to_z 8.00
do_blocking_move_to: (181.00, 158.00, 8.00)
> do_blocking_move_to_xy 46.0028.00
do_blocking_move_to: (46.00, 28.00, 8.00)
> ProbeDeploy
deploy_z_probe > current_position: (46.00, 28.00, 8.00)
run_z_probe > current_position: (46.00, 28.00, 5.12)
<<< probe_pt
z_before = (between) 5.00
probe_pt >>>
> ProbeAction:0
> current_position: (46.00, 28.00, 5.12)
Z Raise to z_before 10.12
> do_blocking_move_to_z 10.12
do_blocking_move_to: (46.00, 28.00, 10.12)
> do_blocking_move_to_xy 171.0028.00
do_blocking_move_to: (171.00, 28.00, 10.12)
run_z_probe > current_position: (171.00, 28.00, 5.25)
<<< probe_pt
z_before = (between) 5.00
probe_pt >>>
> ProbeAction:0
> current_position: (171.00, 28.00, 5.25)
Z Raise to z_before 10.25
> do_blocking_move_to_z 10.25
do_blocking_move_to: (171.00, 28.00, 10.25)
> do_blocking_move_to_xy 171.00158.00
do_blocking_move_to: (171.00, 158.00, 10.25)
run_z_probe > current_position: (171.00, 158.00, 5.26)
<<< probe_pt
z_before = (between) 5.00
probe_pt >>>
> ProbeAction:2
> current_position: (171.00, 158.00, 5.26)
Z Raise to z_before 10.26
> do_blocking_move_to_z 10.26
do_blocking_move_to: (171.00, 158.00, 10.26)
> do_blocking_move_to_xy 46.00158.00
do_blocking_move_to: (46.00, 158.00, 10.26)
run_z_probe > current_position: (46.00, 158.00, 5.22)
> ProbeStow (stow_z_probe will do Z Raise)
stow_z_probe > current_position: (46.00, 158.00, 5.22)
Raise Z (after) by 8.00
> SERVO_ENDSTOPS > do_blocking_move_to_z 21.22
do_blocking_move_to: (46.00, 158.00, 21.22)
<<< probe_pt
> probing complete > current_position: (46.00, 158.00, 21.22)
ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Eqn coefficients: a: 0.00066608 b: 0.00044693 d: 5.12152242
planeNormal x: -0.000666 y: -0.000447 z: 1.000000
set_bed_level_equation_lsq > current_position: (46.01, 158.01, 21.12)

Bed Level Correction Matrix:
+1.000000 +0.000000 +0.000666
-0.000000 +1.000000 +0.000447
-0.000666 -0.000447 +1.000000
> BEFORE apply_rotation_xyz > z_tmp  = 21.12
> BEFORE apply_rotation_xyz > real_z = 21.22
> AFTER apply_rotation_xyz > z_tmp  = 21.20
> corrected Z in G29: (46.01, 158.01, 13.18)
<<< gcode_G29

Setting Print and Travel Acceleration: 1000.00
Print paused at: 07:36:00

The hotend was approximately 5mm ABOVE the bed during the print!

Printer Configuration:
---------------------
Markerfarm i3v (12') with E3D v6 hotend
300x300 mm bed
AUTO_BED_LEVELING_GRID
Z_SAFE_HOMING
DEACTIVATE_SERVOS_AFTER_MOVE
LEFT_PROBE_BED_POSITION   15    
RIGHT_PROBE_BED_POSITION 140
FRONT_PROBE_BED_POSITION  20    
BACK_PROBE_BED_POSITION  15
X_PROBE_OFFSET_FROM_EXTRUDER -31  
Y_PROBE_OFFSET_FROM_EXTRUDER -8     
Z_PROBE_OFFSET_FROM_EXTRUDER -5.2  
Z_RAISE_BEFORE_HOMING 4     
XY_TRAVEL_SPEED 8000     
Z_RAISE_BEFORE_PROBING 8     
Z_RAISE_BETWEEN_PROBINGS 5    
Z_RAISE_AFTER_PROBING 8

thinkyhead, 2015-07-31:
> the hotend is approximately 5mm above the bed when the print begins

It looks like you might have done `M851 S5.2`. Can you re-check after doing `M851 S0`?

PSimonovich, 2015-08-01:
The hotend is now correctly positioned on the bed after removing the extra raise statement. 
I then merged the other changes found in your development build and observed that the Z-probe no longer responds to extend/retract commands. The debug output looked OK. 
I pinpointed the problem to the #if HAS_SERVOS statement (line #125) in the pins_RAMPS_13.h file. If I restore the original #ifdef NUM_SERVOS statement, the Z probe servo works again. I'm not sure why the new macro expansion doesn't work, unless its because its defined in the conditionals.h file instead of the configuration.h file.

AnHardt, 2015-08-01:
Got it. The new macro expansion works. Not so the conditions.
Renaming from SERVO? to SERVO?_PIN was incomplete. In `Cnditionals.h` 4 occurrences remained.
PR is on its way.

AnHardt, 2015-08-01:
Rubbish. The `PIN_EXISTS()` macro extends the `_PIN`

thinkyhead, 2015-08-03:
@PSimonovich

I'm gratified to hear that the fix worked!

The most up-to-date code drops the servo conditions in the pins files so all the servo pins are always defined. I'll be updating my debug_G29 branch with these changes soon, but it sounds like you should be doing pretty well now with the latest MarlinDev code.

duckieHuang, 2016-03-07:
@PSimonovich I am having exactly the same problem compiling marlin using Visual Micro as you stated at the beginning of this issue. I saw that you've solved this problem later. 

So would you please tell me how do you fix this compilation problem? Did you successfully debug Marlin using VM or instead you did it using Atmel-ICE programmer?

jbrazio, 2016-03-07:
Seems to me he used M111 and the `DEBUG` flag of Marlin which will be really verbose on serial.

To proper debug an AVR i.e. break points and such I do believe you'll need an AVR Dragon with the proper fuses set.

PSimonovich, 2016-03-07:
The MarlinSerial redefinition of the SoftwareSerial module causes conflicts with VisualMicro's use of SoftwareSerial/HardwareSerial. I was not able to find a suitable solution and found it easier to add additional debug print statements to Marlin and capture the results in the Pronterface communications panel. 

BTW, I may be reviewing the ABL code again since I'm still not satisfied with the auto correction performance on my Makerfarm i3v 12" printer. I need to enable Mesh bed leveling and compare the results.

jbrazio, 2016-03-07:
It's possible to have more issues than that, we don't know what will happen when the avr-glibc changes nor the compiler. The last time I had to handle something like that I found Arduino was relying in a WinAVR version from the last century, do not know now the status with 1.6 tough.

thinkyhead, 2016-03-08:
What do you think of this thing…? https://github.com/Ultimaker/Ultimaker2Marlin/tree/master/MarlinSimulator

jbrazio, 2016-03-08:
What is that, a virtual Marlin platform ?

thinkyhead, 2016-03-09:
Apparently it is. @daid pointed it out to me. It should run in many places, as it's based on SDL.

daid, 2016-03-09:
The bases is an C level "AVR" simulator, that captures reads/writes to AVR registers. Which allows you to work with just about any Arduino library.

Note that I made it to develop the Ultimaker2 firmware. The status is that it works for me.

jbrazio, 2016-03-09:
@daid do you have a jump start guide ? That seems helpful instead of waiting for AVR to burn the code between tests.

daid, 2016-03-09:
Sorry, no guides here. (I suck at documentation) Just grab your favorite IDE and try to compile it. I've provided my code::blocks project that I use. You need to have the SDL1.2 library installed as well (easy on Linux, as it is in the package managers. On windows you can download it pre-compiled from libsdl.org)

thinkyhead, 2016-03-16:
I'll see if I can get the simulator running in OS X as well. It should be straightforward. Once I get it working I'll post a fork. But not today… Must focus on new release… Must focus on new release…

jbrazio, 2016-03-17:
Don't want to be the one getting you off tack but I'm looking forward to learn how to work with this simulator.

closed 2016-05-04 by jbrazio