MarlinFirmware/Marlin #2536: Auto level seems fine running G29, but after that, head proceeds to crash into bed, despite 0mm offset.

nxumdon, 2015-07-27:
Running the latest development on a rigidbot with dual extruders, my configuration.h is here 

https://drive.google.com/file/d/0B9FLCTwYo5h3WUNUVVcydmg0U1E/view?usp=sharing﻿

for review...the issue is that running a G29 seems to do level correctly but when printing with a;

G28 X0 Y0
G28 Z0
G29
Print.....
It crashes into the bed seeking the mechanical endstop and ignoring the probe having seen the metal and light up...any thoughts?

Thanks

nxumdon, 2015-07-28:
Update...found that having #define Z_RAISE_AFTER_PROBING set to 10mm caused the head to crash after the final probe point...setting it to '0' corrected the issue...seems to be inverted, no?

Roxy-3DPrintBoard, 2015-07-28:
> Update...found that having #define Z_RAISE_AFTER_PROBING set to 10mm caused the head to crash after the final probe point...setting it to '0' corrected the issue...seems to be inverted, no?

That definitely sounds wrong.   Which branch are you using?

nxumdon, 2015-07-28:
I downloaded the development 1.2 last night...

closed 2015-07-28 by nxumdon

thinkyhead, 2015-07-29:
The position correction at the end of bed leveling is supposed to include `Z_RAISE_AFTER_PROBING` so that it knows the head is actually 10mm higher. Up until very recently (basically today) this factor was left out, so at the end of leveling your machine didn't know it was 10mm higher and tried to move too far down. Sorry about that! Give the latest code a try and see if this issue persists. Be ready to hit the power button if it looks like it might crash!