MarlinFirmware/Marlin #2538: THERMAL_RUNAWAY "stops" stepper motors

Loader009, 2015-07-28:
After a THERMAL_RUNAWAY happened the stepper motors are slown down to nearly zero speed. They nearly don't move even if I manually move them (10mm).

To make a THERMAL_RUNAWAY: Reach the desired temperature and then cut the power so the temperature drops. The THERMAL_RUNAWAY happens after some time (look into your config).

After that you can try to move manually the stepper motors (not by hand; by PC or something else). The motors will move but nearly nothing. The auto home feature doesn't work.

thinkyhead, 2015-07-29:
Hmm. A thermal runaway is actually supposed to kill the machine until you reset it.

Loader009, 2015-07-29:
Isn't it safer to let the user move the extruder out of the prints? (Or move automatically the extruder a few mm up - Z axis.)

It is possible that the plastic cools down and the extruder might get stuck in it.

thinkyhead, 2015-07-29:
@Loader009 The condition is so rare we haven't received any reports or complaints on that front (that I have seen) but I agree it's a good idea to (try to) move the print-head on a failure in addition to turning off the heaters, since the machine will need a reset anyway.

AnHardt, 2015-07-30:
@Loader009 
On what branch - version are you.
I tried hard, but can't reproduce this.
In `deveolopment` the error ends in a kill() - you have to reset and can't do anything else.
In `release` the error ends in a Stop(). It's ought to stop the printer until it sees a M999 or a reset.

We are thinking about to move the nozzle away in #2548

Loader009, 2015-07-31:
I use the release branch, version 1.0.2.
I will reproduce the problem and film that with my phone.

thinkyhead, 2015-07-31:
@Loader009 We're getting very close to a 1.1.0 release. That might just have to be documented as a "known issue" for 1.0.2 and a "bug fixed" in the 1.1.0 changelog.

Loader009, 2015-08-04:
Ok, I will wait for the 1.1.0 release.
I will be on vacations until end of the month and then I have not much time. So any test from my side will wait about 1 1/2 months.

thinkyhead, 2015-08-07:
@Loader009 Sounds good. Keep in touch!

closed 2015-09-25 by AnHardt