MarlinFirmware/Marlin #2541: Licensing issue on ReadMe

Jnesselr, 2015-07-28:
I've noticed some issues with license infringement and some discussion on ticket #2524 which is a bit frightening to me. I'd like to clear up a few things, and bring up an issue about the ReadMe as well.

First of all, a machine does not have to have any part of it open sourced to be compliant with the Marlin license. In actuality, they don't even technically have to release the firmware, unless asked. Now, because of the way Marlin configuration works, you could argue that they are modifying and distributing the software. This does not mean their entire design has to be open source. This only applies to the firmware being used.

If the board being used is GPL, and they've made no modifications to it at all (say they bought a bunch of RAMPS boards to use), then they don't have to release the source, but they do have to say that they're using that board and they have to be willing to give you the source and license. Technically, they're supposed to distribute the license with the software/hardware as they're distributing it. That is a problem that we have currently, and something I'd like to help rectify.

As for the ReadMe, the License section says that Marlin is not to be used on any machine that is closed source. As I've already pointed out, their use of open source software does not require the entire machine to be open sourced, simply that this software and any modifications must be released. If no modifications were made, then they only have to specify that they're using Marlin, and give the license used. This statement is also not enforceable, and I think does more harm than good. If you're a closed source company and you see this statement, then you're more likely not to release Marlin's source code because you know the Marlin group isn't going to be happy with you for doing it.

tl;dr: The project creators released this under GPL, and so GPL it is. The sentence in the readme about not using it in closed source software has to change to being a preference or go completely.

Also, https://tldrlegal.com/license/gnu-general-public-license-v3-%28gpl-3%29

foosel, 2015-07-28:
+1

Additionally, that sentence as it is phrased currently could maybe even make Marlin GPL incompatible (if it is legally binding, IANAL), in which case it would violate the GPLv3 license of the works it is a derivative of (Sprinter and grbl) which I'm certain nobody wants to risk.

eboston, 2015-07-28:
While not directly stated in the actual terms and conditions, it is recommended that each source file carries an appropriate copyright notice.  Samples are shown after section 17 [here](http://www.gnu.org/licenses/gpl-3.0.en.html).

However, section 0 does state that any interactive user interface must display "Appropriate Legal Notices."  It's been awhile since I fired up my display, but I don't think what is displayed meets the requirement.

fmalpartida, 2015-07-28:
I am one of the contributors that has been misled by the licensing section of the readme. Never payed too much attention to it, other than being very aware I was contributing to a GPL project, till the ticket was opened. 

I don't know what the original intention was or is but perhaps it has to go. I was against having Marlin go down the route of "please consider donating to support the project..." as it poses a lot of question. 
What I am supportive of is having a blacklist of OS license infringements, in this case, not releasing the sources when requested, with other licenses it may be different. Simply to satisfy the author's or contributor's wishes for sharing their work.

Jnesselr, 2015-07-28:
What's a blacklist going to do? If you mean point out whenever someone is
using your firmware without releasing the source, then I'd urge you to
discuss that with them not publicly shame them. I'm noticing that many
developers here are thinking that these companies are pure evil, bit the
reality is that they may just not know they're infringing on any license.

I mean honestly, the people who should know all about the licenses (us) are
currently arguing about them. What makes you think some company is going to
get it right?

fmalpartida, 2015-07-29:
You do have a very good point.
I guess I am a bit sensitive to the topic after having a range of designs being stepped over their license. What's worse, having it rubbed on. 

I always thought of dialog is the first approach and politely request it the second. But in too many occasions I have experience elusive responses. 

Right now I have gone down the pragmatical approach, the design is there for any one to use. Take it, use it, share it, sell it, modify it, it's there as is, as long as you don't hold me liable for anything I'm good. 

Thanks all for the great discussion.

Jnesselr, 2015-07-29:
Sure, and it's definitely something we as a group need to do better about
enforcing. Luckily, with GPL, you cannot be held liable for your software,
or at least not for what others do.

I also understand that this is a sensitive topic for all of us, and I
appreciate that we can have a level headed discussion about it.

I would think a list would be useful, especially after they blatantly
refuse to issue the source. If they simply don't respond, well then I'm not
sure what I'd do in that case. My main concern is publicly shaming a
company that's not trying to be bad, but just made a mistake.

fmalpartida, 2015-07-29:
I think we are all very much emotional when it comes to sharing part of your time/work and seeing that in too many occasions your wishes are not being respected. In most cases is as simple as, if you make a contribution please share it, it may be useful for someone else. 

In these later years I've been sharing mainly HW designs and having to put up with "it is your obligation to ...", "you have to give me support to interface to...", "it doesn't work if..." even when clearly stating that it comes as is. And also too often seeing how credit has been taken for things that are not theirs. 

I used CC with a clear non liability clause an lately moved to CERN OSL. 

I don't think we can fix the "world" or some "worlds" but do think we can make people more empathetic towards what people freely share.  It is slowly getting there but...

thinkyhead, 2015-07-29:
> This only applies to the firmware being used

Yes, that is true, and the only thing I would try to enforce. You are correct that the hardware doesn't have to be open source for anyone to utilize and distribute Marlin, and we can't restrict commercial use either under GPL. All we can do is ask kindly that when users ask for the source code to the firmware on their printer, they get it. And we can warn them that if they fail to comply it endangers their standing as good open source citizens.

thinkyhead, 2015-07-29:
I propose language for the README more like this:

>Marlin is published under the [GPL license](/Documentation/COPYING.md) because we believe in open development. While we can't prevent the use of this code in products (3D printers, CNC, etc.) that are closed source or crippled by a patent, we would prefer that you choose another firmware or make your own.

Jnesselr, 2015-07-29:
I think that's good. Definitely makes sure marlin can't get into any
trouble. You also might want to add something about what to do if they want
to use marlin in their projects. Their obligations as it were.

thinkyhead, 2015-07-29:
Extending it to include that obligation…

> Marlin is published under the [GPL license](/Documentation/COPYING.md) because we believe in open development. The GPL comes with both rights and obligations. Whether you use Marlin firmware as the driver for your open or closed-source product, you must keep Marlin open, and you must provide your compatible Marlin source code to end users upon request. The most straightforward way to comply with the Marlin license is to make a fork of Marlin on Github, perform your modifications, and direct users to your modified fork. While we can't prevent the use of this code in products (3D printers, CNC, etc.) that are closed source or crippled by a patent, we would prefer that you choose another firmware or make your own.

Jnesselr, 2015-07-29:
Sounds perfect, no objections here.

emartinez167, 2015-07-29:
Sounds good to me!

daid, 2015-07-29:
@thinkyhead Putting marlin in a printer and not giving the sourcecode to the people who own the printer is a violation of GPL. A few open source projects have taken this to court and won. So the actual readme text is WRONG right now. As putting Marlin in your printer and not releasing the source violates all our copyright, our license, and the right of the end-user.

I think one of the major cases was Busybox vs a major TV manufacturer. Google it.

Also, read GPL. It's not THAT hard to read.

eboston, 2015-07-29:
http://www.computerworld.com/article/2537947/open-source-tools/open-source-legal-group-strikes-again-on-busybox--suing-verizon.html

http://softwarefreedom.org/news/2008/mar/17/busybox-verizon/

thawkins, 2015-07-29:
@daid 

I think that at the very least a company should distribute the configuration.h file for thier instance qnd the git tag for tbe version they forked. 

Technicaly the source is alreasy available to everybody via this repo. But being able to reproduce and enhance the software may depend on knowledge about that platform that end users do not have.

daid, 2015-07-29:
@thawkins that still violates GPL. Changing the configuration.h means a modified source code (as it is part of the source code). And per GPL, if you modify it, you need to supply it, in full. If you use command line switches (on the makefile for example) you also need to supply that. GPL is written quite well in that aspect.

What you DON'T need to do, is open source the design of your electronics or printer. That's a different cookie. (I think that's what @thinkyhead is trying to say. But you can read it completely differently)

But if you don't supply the source, you are putting the company at risk. After all, a big lawsuit is a nice way to kill a competing company. And with the 3D printing playground getting crowded, it's bound to happen at some point.

Wackerbarth, 2015-07-29:
In some respects, Tim is correct. I would like to encourage EVERY manufacturer that is utilizing Marlin to have their Configuration file(s) as a part of our repository and for us to be able to invoke their build with nothing more than a command line (or menu selection) option.
In other words, in a form where the entire source is defined by some git sha that is a part of our repository.

That way, they can meet all of the GPL distribution requirements by simply providing a link to our repository along with the appropriate commit tag and selection option. Doing so would satisfy almost every user.

To protect themselves from the possible disappearance of this repository, they need only maintain a git copy of it which is under their control. Since the GPL only requires that the source be made available, not that it be preemptively distributed, they would retain the ability to send a copy if requested by an actual owner of their product. And, if I recall correctly, they even could charge a nominal fee if they were required to perform the extra effort.

Jnesselr, 2015-07-29:
Nope, no fee. It's a requirement, not something extra that they should do.
And at that point, they should just keep a fork anyway.

Wackerbarth, 2015-07-29:
On Jul 29, 2015, at 9:54 AM, Justin Nesselrotte <notifications@github.com> wrote:
> Nope, no fee. It's a requirement, not something extra that they should do.
> And at that point, they should just keep a fork anyway.

I explicitly said that they should keep a copy of the repository.

And, as to fees, I quote from the GPL v3. Please note 6 b (1).

6. Conveying Non-Source Forms.
You may convey a covered work in object code form under the terms of sections 4 and 5, provided that you also convey the machine-readable Corresponding Source under the terms of this License, in one of these ways:

b) Convey the object code in, or embodied in, a physical product (including a physical distribution medium), accompanied by a written offer, valid for at least three years and valid for as long as you offer spare parts or customer support for that product model, to give anyone who possesses the object code either (1) a copy of the Corresponding Source for all the software in the product that is covered by this License, on a durable physical medium customarily used for software interchange, for a price no more than your reasonable cost of physically performing this conveying of source, or (2) access to copy the Corresponding Source from a network server at no charge.

Jnesselr, 2015-07-29:
Exactly, so if you gave it on a flash drive, you could charge for the cost
of the drive, but since it's on github, you can't.

Sorry about the fork part, I got confused with the sha hashes and the form.

thinkyhead, 2015-07-31:
We'll keep refining the language as-needed. I'm glad we at least have some extra notes added now.

closed 2015-07-31 by thinkyhead

thawkins, 2015-07-31:
What about a logo program, Design a "Genuine Marlin Inside" logo that
manufacturers can put on their bots, minimum compliance is a config file
and a tag. Opens the door for a contributions scheme to fund resources for
the project. As a Marketing cost its pretty cheap, and it encourages good
behavior.

thinkyhead, 2015-07-31:
@thawkins :+1: Seems like a good thing to get up on marlinfirmware.org – the only question is who will administer it.

Jnesselr, 2015-07-31:
> @thawkins <https://github.com/thawkins> [image: :+1:] Seems like a good

I'd be a little wary of "marlin inside" for two reasons.

1) problem with the machine? Your problem now.
2) why are they going to bother with a sticker? It's something they'll
probably have to buy from you unless you're subsidizing the cost and it's
useless to their end users.

thawkins, 2015-07-31:
They would be responsable for providing the sticker/logo. There would be
pressure for them to comply, if the site url is part of of the logo, such
that it is something like "marlinfirmware.com inside", then any manufacture
that does not comply with the requirements we could blacklist on the site.

This should not be a scheme where we have to do anything  other than
approve an application, or blacklist. For an approved application, they get
the right to print the logo, and they get an entry on the site of
manufacturers that have complied, with a link to thier latest config file
and tag. Some will use the logo without complying, it will happen, we will
just pubish them on a list of machines that should not be trusted.

> @thawkins <https://github.com/thawkins> [image: :+1:] Seems like a good
> thing to get up on marlinfirmware.org – the only question is who will
> administer it.
>
> <
https://github.com/MarlinFirmware/Marlin/issues/2541#issuecomment-126554471>

—
<https://github.com/MarlinFirmware/Marlin/issues/2541#issuecomment-126554654>.<img
src="
https://ci3.googleusercontent.com/proxy/yRXZVnCCPlRwdTehyfAFPewO9Z9GXD1GQTwtQTAl91wPy_Vko-19zVfXadeanUYgp2Pd1VNU0pIwUV85mF2pTEsUvQdzceDq7PuJVPiUTCn8GgICHs0iOjRzp4kSxNKqZBLDg9R0B2DvVgB5OpHk0tJtdjxKwA=s0-d-e1-ft#https://github.com/notifications/beacon/AAA4yTpeHwI87BvcHl0zik-j-Ze2HV7lks5oit-EgaJpZM4Fhiu1.gif
">

Jnesselr, 2015-07-31:
Seriously? I've seen this before with Marlin and I hate to bring it up
because there's not a lot of other options fire 8-bit AVRs. Honestly, for
some reason Marlin devs whether current or even in the past make it sound
like they're the only firmware in existence.

I don't mean to be rude by saying that, but seriously? Blacklists? Because
they don't put some stupid sticker on there that has no meaning and isn't
useful at all? Who's going to care? Oh you just blacklisted some company
ABC for not putting your sticker on there. I mean ABC has a forked repo and
maintains their machine profile but a sticker? Really?

And company DEF is blacklisted because they're running a Kickstarter and
didn't know you had to put stickers on something just for this silly little
firmware. I mean that's literally not how any other aspect of the printer
works. And guess what? Their backers don't care. What's the sticker going
to do? Raise awareness for the only real firmware for 8-bit AVRs?

If you want the company to give credit where credit is due, don't make them
put some sticker on it, it does no good. If you want a sticker, come up
with a cool logo like @foosel has for OctoPrint, but don't try to make it
part of being "compliant". Last I checked, this repo is GPL and the GPL has
nothing about stickers or blacklists.

This rant has come to a conclusion. Apologies if I offended someone.

thawkins, 2015-07-31:
We are not saying at all that companies need to register to use marlin, i
am saying that if they want to put an official marlin logo on thier product
or marketing, to increase consumer confidence in thier offering, then the
requirements are that they send us a configfile and a tag, for that effort
(minimal), they get the rights to print the logo downloaded from the
marlinfirmware.com site on thier marketing, stick it on tv, put it on thier
website, do what they want with it as far as promoting thier product is
concerned. We can also add them to a directory of manufacturers on the
site, with a link to thier store. Tech companies love those little
standards logos, they are badges of honour, and make thier products look
cooler. Thats the carrot.

If we also reach out to some of the FE/slicer makers like daid, fossel, etc
and see if we can have the "works with marlin" logo on thier products that
would be cool.

If the use the logo without doing the tiny amount of work thst we ask of
them, we will put them on another list of manufacturers on the site who
have not met the requirements above. Thats the stick.

If they dont want to play that is fine, but they are still have to be gpl
complient, but this is a simple way for them to be complient by creating a
mechaism that allows thier users to rebuild thier firmware, which after all
is the purpose of the gpl.

Rather than using a stick of we will take action against you if you break
the gpl, which quite honestly most companies dont care about because they
know nobody will take any action. We create a carrot of getting them some
help with thier marketing,  and if only 10 companies take advantage of it
because it effectivly is a zero cost option, then we have won.

I think you are reading far too much into this, i suspect you would havevno
problem with us hosting a list of companies on the wiki who have violated
gpl, so why is this any different.

You catch more flies with honey than with vinegar.

thinkyhead, 2015-07-31:
> Marlin devs … make it sound like they're the only firmware in existence

I thought we were discussing ideas to encourage GPL compliance and build some kismet with vendors that use Marlin. It's certainly not my view that Marlin is in any way exceptional. If I had a Smoothieboard I would use Smoothie, and probably be developing on that instead.

> Blacklists? Because they don't put some stupid sticker on there

No one has suggested that. There was some discussion about how to handle those who might actively violate the GPL. I think we can all agree, such vendors would only harm themselves, not Marlin. But in fact we don't have any bad citizens out there at the moment in spite of our recent false alarm. So, these are just things to think about.

> You catch more flies with honey than with vinegar.

We have a nice Marlin logo on the boot screen on graphical controllers. We don't put anything up on character-based LCDs but we could. The serial output states that the firmware is Marlin. So at least users can become aware that Marlin is the firmware. Perhaps it would help not to necessarily provide a sticker program, but at least to provide a standard blurb that vendors can use in their materials, something that explains what Marlin is and points users in our direction when they need updates or additional help.

thawkins, 2015-07-31:
Agreed on all points....

emartinez167, 2015-07-31:
It all makes a lot of sense to me.

Regards,

Ernesto

thawkins, 2015-07-31:
Perhaps we could update the graphic splash screen to put the website url on
the bottom.