MarlinFirmware/Marlin #2544: PROPOSAL: Replace Makefile with platform.io

thawkins, 2015-07-29:
Platform.io is an advanced bluild system generator for AVR (and SAM aand ARM and PIC etc)  embedded projects. I creates make, cmake and other build files and can create project files on multiple platforms, multiple targets for multiple IDE's including VS, cLion, Evclipse, Netbeans 

The following is a transcript of  building Marlin on a machine with no marlin source, no arduino, no avrdude. the Configuration is set for RAMPS_EFB single extruder and RFEPRAP_DISCOUNT_FULL_ GRAPHIC_CONBTROLLER. 

The platformio file that needs to be maintained is tiny (platformio.ini) and the system has a built in package manager that makes downloading and installing the U8Glib library needed for the graphic Controller a breeze. 

[thawkins@timdev04 ~]$ mkdir platformiotest
[thawkins@timdev04 ~]$ cd platformiotest
[thawkins@timdev04 platformiotest]$ git clone git@github.com:MarlinFirmware/Marlin.git
Cloning into 'Marlin'...
remote: Counting objects: 19948, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 19948 (delta 0), reused 0 (delta 0), pack-reused 19944
Receiving objects: 100% (19948/19948), 24.44 MiB | 322.00 KiB/s, done.
Resolving deltas: 100% (12298/12298), done.
Checking connectivity... done.
[thawkins@timdev04 platformiotest]$ sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
[sudo] password for thawkins: 

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/lib/python2.7/site-packages
[SUCCESS]

==> Installing PlatformIO and dependencies ...
Requirement already up-to-date: setuptools in /usr/lib/python2.7/site-packages
Collecting platformio
Requirement already up-to-date: requests>=2.4.0 in /usr/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: bottle in /usr/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial in /usr/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: click>=3.0 in /usr/lib/python2.7/site-packages (from platformio)
Installing collected packages: platformio
Successfully installed platformio-2.2.1
Collecting http://dl.platformio.org/scons.zip
  Downloading http://dl.platformio.org/scons.zip (863kB)
Installing collected packages: scons
  Running setup.py install for scons
Successfully installed scons
[SUCCESS]

 ==> Installation process has been successfully FINISHED! <==

Usage: platformio [OPTIONS] COMMAND [ARGS]...

Options:
  --version    Show the version and exit.
  -f, --force  Force to accept any confirmation prompts
  --help       Show this message and exit.

Commands:
  boards       Pre-configured Embedded Boards
  ci           Continuous Integration
  init         Initialize new PlatformIO based project
  lib          Library Manager
  platforms    Platforms and Packages Manager
  run          Process project environments
  serialports  List or Monitor Serial ports
  settings     Manage PlatformIO settings
  update       Update installed Platforms, Packages and Libraries
  upgrade      Upgrade PlatformIO to the latest version

[thawkins@timdev04 platformiotest]$ cd Marlin

[thawkins@timdev04 Marlin]$ cd Marlin
[thawkins@timdev04 Marlin]$ cp ~/Public/MarlinWork/Marlin/platformio.ini .
[thawkins@timdev04 Marlin]$ 
[thawkins@timdev04 Marlin]$ more platformio.ini
;
; Project Configuration File
;
; A detailed documentation with the EXAMPLES is located here:
[... 55 lines cut ...]
avr-g++ -o .pioenvs/megaatmega2560/src/LiquidCrystalRus.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10605 -DPLATFORMIO=020201 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/Wire -I.pioenvs/megaatmega2560/Wire/utility -I.pioenvs/megaatmega2560/U8glib_ID7 -I.pioenvs/megaatmega2560/U8glib_ID7/utility -I.pioenvs/megaatmega2560/LiquidCrystal -I.pioenvs/megaatmega2560/SPI LiquidCrystalRus.cpp
avr-g++ -o .pioenvs/megaatmega2560/src/MarlinSerial.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10605 -DPLATFORMIO=020201 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/Wire -I.pioenvs/megaatmega2560/Wire/utility -I.pioenvs/megaatmega2560/U8glib_ID7 -I.pioenvs/megaatmega2560/U8glib_ID7/utility -I.pioenvs/megaatmega2560/LiquidCrystal -I.pioenvs/megaatmega2560/SPI MarlinSerial.cpp
avr-g++ -o .pioenvs/megaatmega2560/src/Marlin_main.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10605 -DPLATFORMIO=020201 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/Wire -I.pioenvs/megaatmega2560/u8g_polygon.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_rect.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_rot.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_scale.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_state.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_u16toa.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_u8toa.o .pioenvs/megaatmega2560/U8glib_ID7/utility/u8g_virtual_screen.o
avr-ranlib .pioenvs/megaatmega2560/libU8glib_ID7.a
avr-g++ -o .pioenvs/megaatmega2560/LiquidCrystal/LiquidCrystal.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10605 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/Wire -I.pioenvs/megaatmega2560/Wire/utility -I.pioenvs/megaatmega2560/U8glib_ID7 -I.pioenvs/megaatmega2560/U8glib_ID7/utility -I.pioenvs/megaatmega2560/LiquidCrystal -I.pioenvs/megaatmega2560/SPI .pioenvs/megaatmega2560/LiquidCrystal/LiquidCrystal.cpp
avr-ar rcs .pioenvs/megaatmega2560/libLiquidCrystal.a .pioenvs/megaatmega2560/LiquidCrystal/LiquidCrystal.o
avr-ranlib .pioenvs/megaatmega2560/libLiquidCrystal.a
avr-g++ -o .pioenvs/megaatmega2560/SPI/SPI.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10605 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/Wire -I.pioenvs/megaatmega2560/Wire/utility -I.pioenvs/megaatmega2560/U8glib_ID7 -I.pioenvs/megaatmega2560/U8glib_ID7/utility -I.pioenvs/megaatmega2560/LiquidCrystal -I.pioenvs/megaatmega2560/SPI .pioenvs/megaatmega2560/SPI/SPI.cpp
avr-ar rcs .pioenvs/megaatmega2560/libSPI.a .pioenvs/megaatmega2560/SPI/SPI.o
avr-ranlib .pioenvs/megaatmega2560/libSPI.a
avr-g++ -o .pioenvs/megaatmega2560/firmware.elf -Os -mmcu=atmega2560 -Wl,--gc-sections,--relax .pioenvs/megaatmega2560/src/BlinkM.o .pioenvs/megaatmega2560/src/ConfigurationStore.o .pioenvs/megaatmega2560/src/LiquidCrystalRus.o .pioenvs/megaatmega2560/src/MarlinSerial.o .pioenvs/megaatmega2560/src/Marlin_main.o .pioenvs/megaatmega2560/src/Sd2Card.o .pioenvs/megaatmega2560/src/SdBaseFile.o .pioenvs/megaatmega2560/src/SdFatUtil.o .pioenvs/megaatmega2560/src/SdFile.o .pioenvs/megaatmega2560/src/SdVolume.o .pioenvs/megaatmega2560/src/Servo.o .pioenvs/megaatmega2560/src/cardreader.o .pioenvs/megaatmega2560/src/digipot_mcp4451.o .pioenvs/megaatmega2560/src/motion_control.o .pioenvs/megaatmega2560/src/planner.o .pioenvs/megaatmega2560/src/qr_solve.o .pioenvs/megaatmega2560/src/stepper.o .pioenvs/megaatmega2560/src/temperature.o .pioenvs/megaatmega2560/src/tmp_ino_to.o .pioenvs/megaatmega2560/src/ultralcd.o .pioenvs/megaatmega2560/src/vector_3.o .pioenvs/megaatmega2560/src/watchdog.o -L.pioenvs/megaatmega2560 -Wl,--start-group -lm .pioenvs/megaatmega2560/libFrameworkArduinoVariant.a .pioenvs/megaatmega2560/libFrameworkArduino.a .pioenvs/megaatmega2560/libWire.a .pioenvs/megaatmega2560/libU8glib_ID7.a .pioenvs/megaatmega2560/libLiquidCrystal.a .pioenvs/megaatmega2560/libSPI.a -Wl,--end-group
avr-objcopy -O ihex -R .eeprom .pioenvs/megaatmega2560/firmware.elf .pioenvs/megaatmega2560/firmware.hex
"avr-size" --mcu=atmega2560 -C -d .pioenvs/megaatmega2560/firmware.elf
AVR Memory Usage
----------------
Device: atmega2560

Program:  123138 bytes (47.0% Full)
(.text + .data + .bootloader)

Data:       4804 bytes (58.6% Full)
(.data + .bss + .noinit)

======================================================================= [SUCCESS] Took 281.53 seconds =======================================================================
[thawkins@timdev04 Marlin]$

eboston, 2015-07-29:
Is there a version for windows?

thawkins, 2015-07-29:
Yes its supposed to work on all platforms, however you need a functioning
python interpreter.

Wackerbarth, 2015-07-29:
@thawkins -- You make a very good argument for the deletion of the Makefile.
As for the supposed advantages of the scons build, the Arduino IDE will also do all of that once I convince people that we need to get rid of the 1.0.6 support.

As a Makefile replacement, I have few reservations. However, there are a few points that bother me.
I assume that the python environment would not be an issue. (However, I do find it inconvenient that I cannot use python3.)

1) It is not obvious which version of the avr compiler tools is being used.
2) Similarly, what is the source of the avr "core" on which the build is based. (Can we easily control/replace it with one of our choosing?  There are issues with the integration of Marlin with the current implementation of the serial ports)
3) As with the compiler, I wonder about avrdude.  And what about support for boot loaders?
4) For many in our user base, the absence of a GUI is a drawback. Unfortunately, I think that many would-be users are unprepared to do much more that click a button, or two, in a GUI on a Windows-based PC.

thawkins, 2015-07-29:
The makefile is adquate at the moment, but i was looking for something a
bit better, the toolchains from platformio track the latest avr-gcc
compilers and tools, im not 100% sure of how to determine the framework
source versions.  I would support removal of the Makefile so long as we
maintain the platformio.ini file, and we provide howto docs to setup with
platformio, and test it on windows and mac which i will try to do this
week, I have access to all 3 platforms.

Both the makefile and platformio gives those of use who dont want to use
the arduino system a mechanism that allows us to use professional ide's.
Platformio takes it up a notch by supporting remote debugging on ecliose
and netbeans, if you have a usb jtag interface.

Staying locked to the arduino toolset means the system build tools will
never evolve, as i have already shown, platformio allows true libriaries
with the advantage of being able to reduce program size through more
efficient linking, this means more features can be fitted into 128k systems
like printrboards etc (which platformio supports really well through its
teensy platform support).  Being able to organize and modualize the code
using subfolders would be cool.

This same toolset supports the SAM and ARM processors, ARM6, 7 etc. If we
do decide to put a hal in place its an ideal way of doing it.

The future of Marlin may not be arduino tools, they quite litteraly a
millstone around the projects neck.

Wackerbarth, 2015-07-29:
> On Jul 29, 2015, at 10:45 AM, Tim Hawkins <notifications@github.com> wrote:
> 
> The makefile is adequate at the moment, but i was looking for something a
> bit better

I am also …

But, the “millstone” is not the Arduino IDE, per se.
Most of what you suggest below will also apply to the Arduino IDE 1.6.6+

My primary concern is that we be able to support the “click and shoot” user that knows nothing about real software development. (KISS)

There are also issues with the present Marlin codebase that make it incompatible with the latest versions of various components of the development tools. I know how to address that within the Arduino platform templates.  (Like platformio, cmake, ant, etc. the Arduino IDE builds a “makefile”(or equivalent) and invokes the toolset.

> , the toolchains from platformio track the latest avr-gcc
> compilers and tools, im not 100% sure of how to determine the framework
> source versions.

That’s an issue for me. unless I know exactly what is going into the build, can archive a copy of it, and control the version selected, I cannot use it.

> I would support removal of the Makefile so long as we maintain the platformio.ini file

See the Platformio branch in my repository. Perhaps that will indicate a workable organization to accomplish it.
There will then remain only two issues.
(1) Can we avoid maintaining multiple build-configuration schemes?
(2) If we drop using the Arduino IDE, how will we handle CI tests and how will we provide a simple “click and shoot” environment that is readily available to an unsophisticated user.

> , and we provide howto docs to setup with
> platformio, and test it on windows and mac which i will try to do this
> week, I have access to all 3 platforms.

Mac is not an issue. All of my screens are on MacOS X. My Linux boxes run headless.

> Both the makefile and platformio gives those of use who don't want to use
> the Arduino system a mechanism that allows us to use professional IDE’s.

I agree. However, those who are sophisticated enough to be able to use those tools can, almost trivially, reinvent the supporting configuration file.

> Platformio takes it up a notch by supporting remote debugging on ecliose
> and netbeans, if you have a usb jtag interface.
> 
> Staying locked to the arduino toolset means the system build tools will
> never evolve, as i have already shown, platformio allows

Obviously you have not been following the development of the Arduino IDE.
The current version already does much of that. And they are developing the infrastructure to handle the rest.

(Don’t get me wrong. We do need to get away from Arduino IDE 1.0.x and reorganize the code. But neither of those aspects are directly related to the choice of IDE support)

> true libriaries
> with the advantage of being able to reduce program size through more
> efficient linking, this means more features can be fitted into 128k systems
> like printrboards etc (which [...]

Both teensy and arduino/wiring have “issues” that need to be “worked around”.

> This same toolset supports the SAM and ARM processors, ARM6, 7 etc.

The same is true of the current Arduino IDE.

> If we do decide to put a hal in place its an ideal way of doing it.

Agreed.

> The future of Marlin may not be arduino tools, they quite litteraly a
> millstone around the projects neck.

thawkins, 2015-07-29:
I can have a go at the teensy stuff too, i have two printrboards stuck in
my junk pile, i gave them up for ramps boards when ramps became
ridiculously cheap, they are pain in the ass to update (requires dicking
around with jumpers and reset buttons, which loses about 90% of joe
public), are only 128k flash, and are very picky about the setup of the
tools, in the printrbot community there is a preloaded hacked version of
arduino 0.23, with the teensy mods already installed, which is the tool of
choice for rebuilding printrboard Marlin. The printrbot version of Marlin
also has hacked lcd menues, to keep it under the 128k limit. Thats why the
platformio linking mechanism is of great interest, as it packs more
functionality into the 128k flash.

They also use ceramic resonators instead of crystals for clocking, which
for me in sub tropical philippines meant that the clock source for the usb
interface was drifting all over the place with temp changes, and usb comms
got flakey. One of them just plain stopped working when the temp went over
38c becuase the clock litteraly stopped. It was weird, the bot would slow
down as it apporoached that temp like a clockwork toy at the end of a
winding, then once the temp came back down, it would slowly wind up again.
But by then the hotend had melted a big hole in the print and quite often
the hotend had gone cold, and was glued to the print.

thawkins, 2015-07-29:
I should have added that platformio supports builds on multiple CI
aystems,  including travis.

Wackerbarth, 2015-07-29:
On Jul 29, 2015, at 11:58 AM, Tim Hawkins <notifications@github.com> wrote:

> I can have a go at the teensy stuff too, i have two printrboards stuck in
> my junk pile, i gave them up for ramps boards when ramps became
> ridiculously cheap, they are pain in the ass to update [...]

My concern is in finding, and configuring, the various “tool” pieces.
For example, it appears that some AT90USB1280 based boards are distributed with an AVR109 bootloader.
I guess that it works for some, but I couldn’t get it to work with avrdude on my Mac. (I flashed a HID/halfkay bootloader instead. In that configuration everything works OK) But I really want to get away from the Teensy pin numbering. These boards have no TPs associated with that labeling scheme and supporting multiple ARTIFICIAL pin numberings (they don’t have Arduino pins either) is a PITA.

> They also use ceramic resonators instead of crystals for clocking, which
> for me in sub tropical philippines meant that the clock source for the usb
> interface was drifting all over the place with [...]

I haven’t hit THAT issue. 38c is “everyday” here (if it doesn’t reach 40) at this time of the year. But then I don’t run the printer outdoors.

closed 2016-04-23 by jbrazio