MarlinFirmware/Marlin #2546: Nextion TFT HMI

MagoKimbra, 2015-07-29:
I'm developing for use Nextion hmi TFT LCD ...
For now I'm still at the beginning, I can make the home for each axis with the touch and read and set temperatures. LCD connects to a serial Arduino is very easy to use. The only difference with other LCD is that you have to plan ahead with his tools and the graphics functions, marlin you should only worry about updating the fields and detect touches to perform functions.
It could also be of interest to you?
Maybe there is someone who helps me with the graphics that I am bad ....

https://youtu.be/YQNf5X-c8tA

freezingzz, 2015-07-31:
@MagoKimbra  What graphics help you need ? If just need some one to draw some image for you, I think I can do it. But if you need me to make the HMI or TFT file for you, I need to know what your controller send and what info you need when press some button.

IceflowCW, 2015-08-05:
Thats very interesting indeed!! :D ill keep watching for more updates on this. Im also "playing" with the nextion editor and a couple of nextion lcds, but i think im not that good with the code part... the editor its a bit messy... :(

clefranc, 2015-08-06:
@MagoKimbra What's the size of the TFT screen you're using (part number)?

MagoKimbra, 2015-08-06:
I am out for this week. When i return to my home i response you all your question..

Kappa27, 2015-08-06:
I'm also working with a Nextion HMI and the MarlinFW, it looks quite good :)
I hope that Nextion will improve the HMI editor cause now looks really like a messy "demo version".

closed 2022-04-10 by thinkyhead