MarlinFirmware/Marlin #2567: echo:Unknown command: ""

clefranc, 2015-07-31:
Using dev (1aa2012902), get an "Unknow command" with Repetier-Host v1.5.4 and OctoPrint v1.2.4. Did not get the error with Pronterface. Repetier showing more error (line number).

Think I'll ask here first, if it's a host thing, will raise an issue in proper place.

Log from Repetier-Host
```
10:39:50.875 : Printer reset detected - initalizing
10:39:50.875 : start
10:39:50.876 : echo: External Reset
10:39:50.877 : Marlin 1.0.3 dev
10:39:50.880 : echo: Last Updated: 2015-06-00 12:00 | Author: (none, default config)
10:39:50.881 : Compiled: Jul 31 2015
10:39:50.883 : echo: Free Memory: 5118  PlannerBufferBytes: 1232
10:39:50.885 : echo:Hardcoded Default Settings Loaded
10:39:50.885 : echo:Steps per unit:
10:39:50.888 : echo:  M92 X100.00 Y100.00 Z100.00 E186.00
10:39:50.889 : echo:Maximum feedrates (mm/s):
10:39:50.891 : echo:  M203 X500.00 Y500.00 Z500.00 E150.00
10:39:50.893 : echo:Maximum Acceleration (mm/s2):
10:39:50.895 : echo:  M201 X9000 Y9000 Z9000 E10000
10:39:50.897 : echo:Accelerations: P=printing, R=retract and T=travel
10:39:50.899 : echo:  M204 P3000.00 R3000.00 T3000.00
10:39:50.906 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
10:39:50.909 : echo:  M205 S0.00 T0.00 B20000 X15.00 Z15.00 E5.00
10:39:50.910 : echo:Home offset (mm):
10:39:50.911 : echo:  M206 X0.00 Y0.00 Z0.00
10:39:50.912 : echo:Endstop adjustment (mm):
10:39:50.913 : echo:  M666 X0.00 Y0.00 Z0.00
10:39:50.917 : echo:Delta settings: L=delta_diagonal_rod, R=delta_radius, S=delta_segments_per_second
10:39:50.919 : echo:  M665 L439.40 R259.00 S200.00
10:39:50.919 : echo:PID settings:
10:39:50.921 : echo:  M301 P22.20 I1.08 D114.00 C1.00
10:39:50.924 : echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
10:39:50.925 : echo:  M207 S5.00 W13.00 F6000.00 Z0.00
10:39:50.927 : echo:Recover: S=Extra length (mm) F:Speed (mm/m)
10:39:50.929 : echo:  M208 S0.00 W0.00 F6000.00
10:39:50.933 : echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
10:39:50.933 : echo:  M209 S0
10:39:50.935 : echo:Filament settings: Disabled
10:39:50.935 : echo:  M200 D1.75
10:39:50.936 : echo:  M200 T1 D1.75
10:39:50.937 : echo:  M200 D0
10:39:52.287 : echo:Unknown command: ""
10:39:52.291 : Error:Line Number is not Last Line Number+1, Last Line: 0
10:39:52.291 : Resend: 1
10:39:52.307 : FIRMWARE_NAME:Marlin 1.0.3 dev SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:BI V2.5 EXTRUDER_COUNT:2 UUID:00000000-0000-0000-0000-000000000000
10:39:52.315 : echo:Active Extruder: 0
10:39:52.315 : echo:Active Extruder: 0
10:39:54.717 : FIRMWARE_NAME:Marlin 1.0.3 dev SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:BI V2.5 EXTRUDER_COUNT:2 UUID:00000000-0000-0000-0000-000000000000
10:39:54.717 : echo:Active Extruder: 0
10:39:54.723 : echo:Active Extruder: 0
```

Log from OctoPrint:
```
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: start
Send: M110
Recv: echo: External Reset
Recv: Marlin 1.0.3 dev
Recv: echo: Last Updated: 2015-06-00 12:00 | Author: (none, default config)
Recv: Compiled: Jul 31 2015
Recv: echo: Free Memory: 5118  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z100.00 E186.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z500.00 E150.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X9000 Y9000 Z9000 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X15.00 Z15.00 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Endstop adjustment (mm):
Recv: echo:  M666 X0.00 Y0.00 Z0.00
Recv: echo:Delta settings: L=delta_diagonal_rod, R=delta_radius, S=delta_segments_per_second
Recv: echo:  M665 L439.40 R259.00 S200.00
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00 C1.00
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:  M207 S5.00 W13.00 F6000.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:  M208 S0.00 W0.00 F6000.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:  M209 S0
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 T1 D1.75
Recv: echo:  M200 D0
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: M21
Recv: ok
Send: M105
Recv: echo:Unknown command: ""
Recv: ok
Recv: ok T:25.0 /0.0 B:24.4 /0.0 T0:25.0 /0.0 T1:25.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:25.0 /0.0 B:24.3 /0.0 T0:25.0 /0.0 T1:25.0 /0.0 @:0 B@:0
Changing monitoring state from 'Operational' to 'Closed'
Connection closed, closing down monitor
```

Roxy-3DPrintBoard, 2015-07-31:
I've seen this error also.

thinkyhead, 2015-07-31:
Two conditions will throw an "Unknown command" error:
- If the command `M209 Sn` is received with a value other than 0 or 1.
- If a badly-formed command is received, such as `Q` or `A123`.
It's possible that sending one or more spaces on a line by itself could result in `Unknown command: ""`. The workaround for that would be to skip the error if a command has nothing else in it. It means adding one more case to the `switch(command_code)` block in `process_next_command`:
```cpp
  switch(command_code) {
  case 'G': ... break;
  case 'M': ... break;
  case 'T': ... break;
  case 0: break; // ignore an empty command
  }
```

ntoff, 2015-07-31:
I got "unknown command" errors in repetier host as well, I just figured it was something they were sending out that looked for a feature in their own firmware that wasn't present in marlin or something. Never seemed to affect my prints.

zakkos, 2015-08-04:
Hello all, nearly the same problem here except it happens when I print from SD. 

I noticed some strange behaviour from my machine when I loaded the then latest dev version (b17b286c32e6d7697aa8eada0ae0b363bdfbaf4d from 27/07). 
It would occasionally, while printing, depart from its job and send an axis (x or y but never z, for now) to 0 then resume from where it interrupted (of course I don't know if it's EXACTLY the same spot but so it looks, probably it misses a command but i can't confirm). 
Sometimes it would just hang in the same spot for some time and then resume, other times it just stops with motors engaged, heaters and bed at their supposed temp.
 
So today i hooked up my machine to pronterface during a test print and seen the dreaded "echo:Unknown command:"  and the link to the strange behaviour is clear. 

 
The first two lines of this pronterface log shows where the printer sent an axis to 0 and back, while the rest happened at a fraction of second and it's when the machine hangs.

```

echo:Unknown command: "8 Y208.839 E10.43657"
echo:Unknown command: "360.498 Y228.918 E25.45064"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
```

Hope this helps tracking this issue down. I'm going to try and add @thinkyhead workaround but I don't see how this could help with the truncated commands.

thinkyhead, 2015-08-05:
@zakkos That really looks like some kind of communication problem to me, not a code problem. The parser simply can't end up in the middle of a command string unless that's the string the card reader (or USB connection) produced. If the printer did some spurious move to 0 and back, that is a clear sign of a connection problem – I've seen it. You should make sure the connections to your LCD controller and/or SD card reader are nice and tight. You will notice that you can actually cause the error to occur (and weird moves to happen) by wiggling the connection wires. If there is a bad or incomplete contact, the printer will do exactly these weird things.

zakkos, 2015-08-05:
Indeed. Yesterday after a few more monitored prints i've got an unknown command echo full of uncommon characters (like ÿ, graphic characters etc) that are a clear sign of communication problems, for the records I tracked it down to the sd/arduino connection since everything's ok printing from a host. 
I think I will still add the null skipping case in the commands switch, just to allow the machine to stop in a decent way instead of coming to a grinding halt!

Thank you anyway, sorry to have polluted the thread ;)

thinkyhead, 2015-08-07:
@zakkos Thank you for adding to the thread, actually! I would love to be able to throw a better error like "communication fault" but we have to be careful. In my personal opinion, the `Unknown command: ""` pretty much equals a communication glitch. But, we'll have to see if it can be produced in any "normal" way by sending commands through the console (for example, could something like `N123 *0` throw this error?).

closed 2015-09-25 by AnHardt