MarlinFirmware/Marlin #2576: Status (LCD) updates , "heating done" .. forever

AndKe, 2015-08-01:
I see "heating done" (or equivalent message) in the beginning of the print.
Then the print is done, heaters are off.  - even motors are off, still - same message.
- Please add more, or remove status updates once they are irrelevant.

thinkyhead, 2015-08-03:
Which version of Marlin are you using?

As a workaround, after any `M109` command in your G-Code you can add an `M117` command.

AndKe, 2015-08-03:
Hi, sorry, not sure about the version, but it's a master from around February, apparently I just got at .zip and do not have git data about it.
Would you recommend release 1.0.2-1 , or should I really go for development version ? 

Working around is not very important to me, just wanted to let you know so you could polish this great open source by fixing it in the future.

AndKe, 2015-08-03:
Edit: in my Configuration.h I 've found "Version 1.0.2"  - but a compare to freshly downloaded stable (1.0.2-1 ?) contains VERY many differences  - I am a little confused here.

AndKe, 2015-08-03:
Apparently, judging by the separate "pins" files  - I have a dev version.
Also, please note that clicking the big "download zip" button on the right on https://github.com/MarlinFirmware/Marlin   , does provide development version, not stable as would be expected.

thinkyhead, 2015-08-04:
> …does provide development version…

When I click that button it gives me the head of the Release branch. If you want the head of the Development branch, you would click the button at https://github.com/MarlinFirmware/Marlin/tree/Development instead.

thinkyhead, 2015-08-04:
@AndKe This is still an extant issue in the latest code. I added `M117 Printing...` to my slicers' start-GCode, so I guess I've been hiding it.

It's an unusual case. I don't think we want to abandon the LCD message, but at the same time, we don't put up such a message when picking a temperature from the LCD menu, for example. I've thought about a middle ground, where the message only hangs around for a minute or so, but then if a user walks away from the machine they might want to see the most recent status message.

AndKe, 2015-08-04:
@thinkyhead   No, I do not suggest removing the message, just add more updates.
Maybe make it show heating, printing, then " done / cooling / idle", or just blank the last message when not used.
Also, the line could be updated before calling kill() , "killed: temperature error"  , "killed: idle timeout"  etc..

AnHardt, 2015-08-04:
void kill(const char *);
has a working display message in `Development`.

If you want more display messages, just send them with a `M117 Text`.
We are using more and more valuable PMEM for strings.

I think is slic3r where you can opt in to display the remaining print time.

thinkyhead, 2015-08-07:
@AnHardt Well, one problem is `M109` might happen in the GCode _after_ any initial "Printing..." message, so we'd need to show the "Printing..." message as the first E move is detected, and that's not something we want to have to add to the loop. Also, host printing is passive, so the printer doesn't even know it's "printing." Darn all these silly pitfalls….

closed 2015-09-25 by AnHardt

blueeagle69, 2018-05-30:
Hi.
Is there any update regarding the "heating done" stuck on the LCD screen when using any manual heating Gcode M109 etc in Start Gcode? I using 1.1.8 and it still does it :( You can indeed hide it by using M117.
But then, on my Anet A8 at least, you are stuck with that part of the LCD screen blank throughout the length of the print. I usually USB print using MatterControl as my slicer, and I have Progress Reporting set to M73 which worked fine. Perhaps I should change it back to M117 if I use M109 in my start Gcode, followed by an M117 to hide the stuck message?

thinkyhead, 2018-05-30:
Yes, it now clears the line and replaces it with the name of the current printing SD file or other pertinent information.

blueeagle69, 2018-05-31:
Thanks.
Could you tell me what branch/version it's fixed on if you know?
Unless I'm mistaken, I'm running 1.1.8 on my Anet A8, and the heating done message sticks on the LCD if I use M109 etc in my start GCode.
Perhaps I need to jump to the 2.x branch.
Thanks again.

thinkyhead, 2018-05-31:
`bugfix-1.1.x` and `bugfix-2.0.x`