MarlinFirmware/Marlin #2580: Recent servo changes a day ago disrupted servo subsystem

Natealus, 2015-08-02:
Due to the overridable changes I reconfigured on a new configuration.h. But by the time I got that done after work it seemed a couple changes had happened as well but it seems to have broken the servo subsystem. I doesn't work on index 0 or 2. I tried G29 E, M401, M402, and M280 P0 S90 and such but none of them seemed to affect the z-probe servo.

Natealus, 2015-08-02:
I uploaded an archived version of the firmware from 7-10-2015 and it's confirmed the wiring and servo still work correctly.

AnHardt, 2015-08-02:
https://github.com/MarlinFirmware/Marlin/issues/2533#issuecomment-126846397
Indeed it seems, the macro is not expanded.
Workaround is to replace:
```
#if HAS_SERVOS
````
 in you pins_*.h-file with:
```
#if defined(NUM_SERVOS) && NUM_SERVOS > 0
```

Wackerbarth, 2015-08-02:
Moved to MarlinDev issue #3

closed 2015-08-02 by Wackerbarth