MarlinFirmware/Marlin #2581: Z_DUAL_ENDSTOPS Compile error

linjinc, 2015-08-03:
-----------------------------------------
#if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_MAX_PIN 36       
    const bool Z2_MAX_ENDSTOP_INVERTING = false;
    #define DISABLE_XMAX_ENDSTOP         
  #endif
----------------------------------------
Change to,My machine is Z reset MIN,Motherboard is ULTIMAKER.h
-----------------------------------------
#if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_MIN_PIN 26       
    const bool Z2_MIN_ENDSTOP_INVERTING = true;
    #define DISABLE_YMIN_ENDSTOP         
  #endif
----------------------------------------

Compile error

stepper.cpp: In function 'void update_endstops()':
stepper.cpp:436: error: expected ';' before 'byte'
             byte z_test = TEST_ENDSTOP(Z_MAX) << 0 + TEST_ENDSTOP(Z2_MAX) << 1; // bit 0 for Z, bit 1 for Z2
             ^
stepper.cpp:438: error: 'z_test' was not declared in this scope
             if (z_test && current_block->steps[Z_AXIS] > 0) {  // t_test = Z_MAX || Z2_MAX
                 ^
expected ';' before 'byte'

AnHardt, 2015-08-03:
At the moment the
`#if ENABLED()` macro is not working in the pins_*.h files.

As a workaround you can replace the line by:
`#ifdef Z_DUAL_ENDSTOPS`

We are working on a more general solution.

Wackerbarth, 2015-08-03:
I don't think that the ENABLED() macro is supposed to work in the pins_* files.

That file defines the BOARD characteristics, not whether, or not, it is actually being USED.

thinkyhead, 2015-08-03:
I agree. Pins should just define pins. But as I recall, in some cases `SDSUPPORT` causes some pins to be shuffled around to avoid conflicts. We should clean these up, for sure.

thinkyhead, 2015-08-03:
Darnit, this is actually a [MarlinDev](https://github.com/MarlinFirmware/MarlinDev) issue.

Wackerbarth, 2015-08-03:
The SDSUPPORT situation has to do with duplicate pin definitions.  I'll need to see why the SD routines are using their own definition of some pins.

AnHardt, 2015-08-10:
@linjinc 
Sorry. Sometimes we are at a totally wrong track for a while.
Resolved by https://github.com/MarlinFirmware/MarlinDev/pull/67

closed 2015-08-11 by thinkyhead