MarlinFirmware/Marlin #2582: Message text thermal runaway  Heater_ID wrong

RogerRepRap, 2015-08-03:
10:42:47.878 : Error:Thermal Runaway, system stopped! Heater_ID:
9

Runaway message has a problem. The message always reports heater id as "9" so there is no way to tell which heater had problem. This makes troubleshooting a very intermittent problem difficult. Not a showstopper. 

Keep up the good work.

Roger

thinkyhead, 2015-08-03:
What version of Marlin are you using? The error message differs in the latest development code (no newline after "Heater_ID:").

thinkyhead, 2015-08-03:
Apparently in Marlin 1.0.2 the number "9" indicates the bed. We may patch this.

AnHardt, 2015-08-03:
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/temperature.cpp#L566

Wackerbarth, 2015-08-09:
Does this need to be ported to the development tree?

AnHardt, 2015-08-10:
No, it's completely reworked there.

closed 2015-08-10 by Wackerbarth