MarlinFirmware/Marlin #2583: Hangout -- Saturday August 15 UTC 2000

Wackerbarth, 2015-08-03:
Name  | Summer Time Zone
------------ | -------------
@thinkyhead    |  UTC-7
@Wackerbarth  |   UTC-5
@Jnesselr    |    UTC-5
@clefranc     |   UTC-4
@alexborro    |   UTC-3
@fmalpartida   |  UTC+1
@AnHardt     |    UTC+1
@Wurstnase    |   UTC+1
@CONSULitAS   |   UTC+1
@paclema     |    UTC+1
@boelle    |      UTC+1
@emartinez167   |  UTC+8
@ntoff    |       UTC+10

[The "Official" UTC Clock](http://www.timeanddate.com/time/aboututc.html)
[Link to the Hangout](https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia)

Please provide input about what we need to discuss.
[Link to Google Document listing discussion topics (Week 33)](https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing)

fmalpartida, 2015-08-05:
I'll try to make it. On holidays...

fmalpartida, 2015-08-05:
A Spanish college pointed me towards: http://marlinkimbra.it
Cool initiative.

fmalpartida, 2015-08-05:
Git repo: https://github.com/MagoKimbra/MarlinKimbra
Nice initiative by the Italian community.

AnHardt, 2015-08-05:
And we are lucky that @MagoKimbra is contributing here from time to time. Thank you.

Wackerbarth, 2015-08-05:
"Nice initiative"  -- Well it would be nice if they hadn't forked our repository and detached it from our root.
Not only have they destroyed all attribution to our work, but they make it very difficult for us to fold in anything that they do.

Wurstnase, 2015-08-05:
I'm with @Wackerbarth.
It's nearly impossible to take some code from his repository.

fmalpartida, 2015-08-05:
I thought that you were going to get upset for the way they "forked". It's a shame because there is very interesting complemental stuff going on there.

paulusjacobus, 2015-08-05:
@fmalpartida is it possible to give a synopsis of features they have added? I have looked around and poked in the code and it seems they have kept it up to date. Especially I was interested if they fixed the ABL feature independently.

fmalpartida, 2015-08-15:
Hi folks, i am currently on vacation with very limited inept access. I don't think I will be able to make it to tomorrow's HO. 
I wanted to report that I am good to release on all my Cartesian printers with auto bed calibration and AT90USB based boards. 
Next week I'll be back on track and ready to continue.

Jnesselr, 2015-08-15:
I may be really bad with time zones, especially since I'm now UTC-6, but wasn't the meeting supposed to start 7 minutes ago? Or do I have the wrong hangouts link?

darkjavi, 2015-08-15:
Is is going to happen? I don't have much time till dinner but I thought in stepping by to introduce myself. I checked the link but is says is over

Jnesselr, 2015-08-15:
Hangouts might be having some issues. Try to jump in again. I'm in the chat
room currently.

darkjavi, 2015-08-15:
I'm in now :)

fmalpartida, 2015-08-15:
Limited inet bandwidth! Can't make it tonight. Sorry folks.

thinkyhead, 2015-08-15:
Sorry to have missed this one, but we're about to go on holiday (and have to move house when we return) – so it's a little nuts around here! Mainly I hoped to discuss the last sticking-points before RC1, but we can cover those in an issue discussion, and perhaps in the next hangout.

closed 2015-08-27 by Wackerbarth