MarlinFirmware/Marlin #2585: 4 Extruders

mistsoul, 2015-08-04:
Hello all,

I have some questions as I am upgrading my machine.

My machine has a RAMP 1.4 board, i don't think I can run 4 extruders with it, or are there any tricks to easily upgrade it to do so?

If not, I was thinking about getting an Azteeg X3 Pro fully loaded so once I get there i may need some help to configure it and ensure that the amps are correct on each motors, if there are tricks to do so not knowing the current capacity of my current motors, please let me know. 

Also are there any parts of my Ramp board that I can use with the Azteeg board, which would allow me to not necessarily get it fully loaded, like the stepper drivers?

The duet + duex can do 4 extruders also but i don't think it is sold in the US?

At this point I am working on re-configuring Marlin and I am having some challenges with setting the extruders. I asked the manufacturer what motors he was using but he did not share the information. 

Do you guys have any trick to figure out exactly what motors are used in the machine? From the dimensions (42x42x40mm) I can guess that the ones used are Nema 17 and not more than 45 N-cm, but no idea on the current and ohms, also other axis motors seem to be the same as the extruders. I thought it would be good to know the exact model if possible in case I want to buy the same exact motors so I can leave the steps per mm the same in the firmware.

The machine is dual extruder and if I can’t get the same extruder motors, I will need to get two more motors for the 2 additional extruders that I am planning. Those would not be mounted on the head so I was looking at maybe 59 N-cm but not sure if the additional torque is not needed. 

Also for the new extruders would it make a whole lot of difference if I get them at .9 degrees instead of 1.8, i would think it would for the axis but not sure if it is that useful for the extruders. 

Now if I get different motors than the ones i currently have (E0=E1 existing and E2=E3 new), what do I do with the steps per mm, since it seems like i cannot not set a different steps/mm? I would like to set them as #define DEFAULT_AXIS_STEPS_PER_UNIT {X,Y,Z,E0, E1, E2, E3} but I was told that Marlin won't know what to do with it? Has anyone set different steps/mm for their multi extruders using different motors? if not, how do you cope with having the same steps/mm for all extruder motors.

Sorry I have a lot of questions, thanks for your help!

@MagoKimbra I did look at your standard code, but do let me know if you have any tips on the multi-extruding and different steps/mm

thinkyhead, 2015-08-07:
Did I answer this for you already? I remember doing so, but now I can't find it. I'll re-explain the various options if needed here.

mistsoul, 2015-08-07:
Yes you did thank you so much, this was more for others to be able to pitch in. Have not heard from anybody else but i am moving forward with the changes. Can you point me to repositorys or threads discussing servos and how they are configured in marlin especially for extruder tooling. My search returns do not seem to return anything or i may not be using the engine correctly.

thinkyhead, 2015-08-08:
@mistsoul The only place Marlin does anything with servos is moving them between 2 positions (deployed / stowed) for a bed probe (either as a Z endstop or as a separate Z probe). There's currently no hook to handle servos as part of a Tool Change (e.g., `T2`) command, but that's been on my mind a lot, thinking about how to make an Extruder (or Tool) class that encompasses all the parts of an extruder, including heater, thermistor, motor, and additionally a motor direction and a servo position (for servo-switchable extruders that share a single motor). Right now this is all just in the conceptual phase.

mistsoul, 2015-08-08:
That sounds very cool, i have seen that a few folks have attempted what you are describing and have used marlin to do so successfully. I need to find out more about how servos can be automated via marlin code change especially but i am having no luck with my searches on here maybe because i am using the engine incorrectly, yet i know that there are forum discussions on here about it.

AnHardt, 2015-08-08:
At the moment there are two ways servos are used.
You can use one of the servos to move a servo-probe,
all other servos can be moved by M280.
That's all what is implemented in Marlin.

For many thinkable systems it's already possible to do a (automatic) extruder- or filament-change by sending the right G-code ___sequence___ to Marlin.

mistsoul, 2015-08-08:
@anhardt thanks for the insight, so the M280 command can be used to moved the the servo 3 positions? 90 deg left, rest (or middle position) and 90 deg right? Are there ways to move it in increment of a few deg any direction needed rather than the full 90 deg? I am lookinflg to achieve at least 3 positions, move left, move back to middle, move right, or visa versa.

AnHardt, 2015-08-08:
@mistsoul 
Please have a look on the arguments of M280.
```
// M280 - set servo position absolute. P: servo index, S: angle or microseconds
```
You can set it to every angle you want (and your servo can handle).

mistsoul, 2015-08-08:
Yes i was able to confirm it by checking the argument earlier, thank you for all the help!

mistsoul, 2015-08-09:
@anhardt when this is programmed into Marlin, for example for a single motor double extruder, how does the slicing engine when generating gcode know to use only E0 for color changes after servo switch (M280). I think this was asked in another area of the forum but there were no answers.

AnHardt, 2015-08-09:
@mistsoul 
As said before - there is noting in Marlin to support a servo_switched_ single_nozzle_single_stepper_dual_filament_extruder.

But maybe you can find a sequence of g-codes what can do the job for you.
Retract filament A - switch servo - switch stepper direction - change temperature - change feedrate - ... - unretract filament B (or similar).

Your personal g-code post-processor (for example a sed-script) then hast to replace every T1 to the sequence, every T0 to an other sequence and every E1 to a E0. (or similar)

That's a concept only - not a ready solution!

I never tried to reverse a stepper direction on the fly. Maybe a negative steps/mm value will do the trick.

emartinez167, 2015-08-09:
I tried this in the past with an HPX2-MAX (single motor, two nozzles) with no success. I believe some other people were in the same boat, but this never progressed so I moved to dual motors.

Regards,

Ernesto

mistsoul, 2015-08-09:
Thanks for the input guys.

thinkyhead, 2015-08-11:
For extruder setups like the HPX2-MAX to be fully supported in Marlin we'll need to wait until we have a better abstraction of Extruder as a state machine. Especially since the single motor reverses direction depending on the current extruder number. You could hack it possibly, setting the pins for the two extruder stepper motors the same – except reversing the direction, but I would much rather see this as a real supported feature in the long run. But, it will need to wait until after this release.

Wackerbarth, 2015-08-11:
@thinkyhead -- I endorse the idea of replacing the current handling of issues such as these with state machine driven logic. Obviously in some future release.  Adding additional "hacks" has led us to the current mess and we need to avoid compounding the problem with more of them.

closed 2016-04-23 by jbrazio