MarlinFirmware/Marlin #2587: Support for reverse x-axis direction of H-bot is needed.

DWONH, 2015-08-05:
Unlike corexy, direction of X-axis in H-bot can be reversed depends on position of belt clamp.
And it can not be configured by any of existing variables.(rel 1.0.2)
Like depicted in photo below, setup#1 is general configuration and is supported in current release version. But for setup#2, current code moves head reverse direction which is not desired.

Motion equation should be changed for setup#2, as seen in picture.

[image]

It would be nice if we have option for this in configuration.h.

Wackerbarth, 2015-08-05:
Yes, this is a kinematics feature that we need to add.
@boelle -- Can you move this to MarlinDev?   (And tell me how to use the tool so I can move individual items such as this myself)

This request will be closed here and reopened in the MarlinDev repository.

whosawhatsis, 2015-08-05:
Can't you reverse it in hardware by swapping the two motor cables (so that in this diagram, A would be on the right and B would be on the left) and reversing both motor cable connectors to reverse the motor directions?

grob6000, 2015-08-06:
That was my thought. The equations bear it out: notice that equations for dA(#2) = -dB(#1) and dB(#2) = -dA(#1).

thinkyhead, 2015-08-07:
@whosawhatsis Hardware solution FTW! But I suppose we can still add an option here for people who prefer to have their cable plugs all face the same way and have the same color-order.

Wackerbarth, 2015-08-07:
@thinkyhead -- If you go to corresponding item  (#55) in MarlinDev, you will find that we were able to fix this totally in the pin definitions and configuration. Basically, it is a HAL layer issue and requires no additional code.

closed 2015-10-14 by AnHardt