MarlinFirmware/Marlin #2595: I have 2 questions

ruggb, 2015-08-17:
1. Has any form of "dry run" been implemented as yet?
2. Every time I connect to my printer I get this error RE checksum after FIRMWARE NAME: - can u tell me why? this is the whole log output
11:55:48.124 : Printer reset detected - initalizing
11:55:48.124 : start
11:55:48.124 : echo:Marlin 1.1.0 dev
11:55:48.126 : echo: Last Updated: 2015-08-00 12:10 | Author: Bill Ruggirello
11:55:48.127 : Compiled: Aug 11 2015
11:55:48.131 : echo: Free Memory: 3188  PlannerBufferBytes: 1232
11:55:48.133 : echo:V20 stored settings retrieved (378 bytes)
11:55:48.134 : echo:Steps per unit:
11:55:48.136 : echo:  M92 X200.00 Y200.00 Z800.00 E299.90
11:55:48.137 : echo:Maximum feedrates (mm/s):
11:55:48.139 : echo:  M203 X500.00 Y500.00 Z8.00 E25.00
11:55:48.140 : echo:Maximum Acceleration (mm/s2):
11:55:48.143 : echo:  M201 X9000 Y9000 Z100 E10000
11:55:48.144 : echo:Accelerations: P=printing, R=retract and T=travel
11:55:48.146 : echo:  M204 P3000.00 R3000.00 T3000.00
11:55:48.154 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
11:55:48.157 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
11:55:48.157 : echo:Home offset (mm):
11:55:48.158 : echo:  M206 X0.00 Y0.00 Z0.00
11:55:48.159 : echo:Material heatup parameters:
11:55:48.161 : echo:  M145 M0 H210 B70 F255
11:55:48.162 : echo:  M145 M1 H260 B100 F255
11:55:48.164 : echo:PID settings:
11:55:48.165 : echo:  M301 P21.96 I1.86 D64.76 C1.00
11:55:48.166 : echo:Filament settings: Disabled
11:55:48.168 : echo:  M200 D3.00
11:55:48.168 : echo:  M200 T1 D3.00
11:55:48.169 : echo:  M200 D0
11:55:51.802 : echo:SD init fail
11:55:51.893 : FIRMWARE_NAME:Marlin 1.1.0 Development From Archive SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:CC-BOT 2E&QH v2e EXTRUDER_COUNT:2 UUID:277494b7-b7c8-4eb1-a653-c2e695c24911
11:55:51.897 : Error:No Line Number with checksum, Last Line: 3
11:55:51.903 : Error:Line Number is not Last Line Number+1, Last Line: 3
11:55:51.903 : Resend: 4
11:55:51.909 : echo:Unknown command: ""
11:55:51.909 : Error:Line Number is not Last Line Number+1, Last Line: 3
11:55:51.909 : Resend: 4
11:55:51.915 : echo:Active Extruder: 0
11:55:51.916 : Begin file list
11:55:51.916 : End file list
11:55:51.925 : Error:Line Number is not Last Line Number+1, Last Line: 11
11:55:51.925 : Resend: 12
11:55:51.931 : echo:Active Extruder: 0

Wackerbarth, 2015-08-17:
Obviously, this applies to a development build rather than a release build.
1) Yes
2) The firmware received line 3 from your host program and then is out-of-sync. It looks as if you have lost some characters in the communications stream and your host program is not recovering.

Since you fail to give any hints about what kind of hardware, host computer, or host interface program you are using, I do not have enough information to even attempt a better answer.

thinkyhead, 2015-08-17:
I concur with Richard. It looks like a communication issue causing bad GCode to be sent to Marlin. Try tightening up connections to the LCD controller, using better/shorter USB cable, etc., and see if it clears up.

ruggb, 2015-08-18:
thx
I am using arduino mega 2560 and ramps 1.4 I also have a full graphics controller with SD card - but I am not using that yet. It is driving from either of my laptops and gets the same error on either.
There r more errors also. U mention LCD controller - mine is connected to the ramps bd with 3m cables. Is there an issue with cable length on this device? Or did u mean something else by LCD controller?

ruggb, 2015-08-18:
If I take the mega and load a dev build and it produces the error. 
I then load a release load and it does not produce ANY errors.
This is a different but identical board than in the system (which started the whole thing).
This bd has no ramps bd attached.
This bd had an old dev build which produced the same error.
I loaded a rel build, no error.
Then load newest dev build, error
then load a rel build, no error
then load dev build, error.
there seems to be a pattern here, wouldn't u say?
I can't blame the cable on this behavior.

Wackerbarth, 2015-08-19:
Obviously, SOMETHING is different. However, since you have not disclosed HOW you are doing the builds, or what HOST software you are using, I don't have enough information to help you figure out just what is causing the different behavior.  I'm not disputing that you are observing incorrect behavior. However, it does appear to be a communications issue. That issue might be caused by (1) The new firmware, (2) the compiler toolset that you are using, or (3) an incompatibility between the protocol in the dev firmware and the host software that you are using, or (4) some "setting" that is improperly configured.hh55q3

ruggb, 2015-08-19:
I do not know what u mean by "HOW" I am doing the builds.
I configure configuration.h and configuration_adv.h and load via Arduino 1.6.5
I have the latest release build that I previously used.
I have the 8/0/15 dev build and do the same.
I load via my desktop usb port.
I believe the configs r the same but because the files r different I cannot be certain.
+++ it would be AMAZINGLY helpful if there was one file with parameters that worked across all builds so that one would not have to keep rebuilding the configs. ++++

the firmware is the same on the boards - IE it doesn't change from load to load except for the marlin.
the compiler toolset is identical.
If there is an incompatibility between the protocol and the dev f/w, and NOT so with the rel f/w, then I would consider that an dev issue, wouldn't you?
If there is a "setting" that is improperly configured - I would think it would cause other issues but maybe not and as I stated above, if there is a dif I have no clue, but I have checked many times and except for Quickhome set in dev there is no dif.
OK, i will turn it off and try again......

ruggb, 2015-08-19:
with QH disabled it does the same thing.
I have moved all my changes to the head of the file so that it would be simpler.
Alas, since the files keep changing it doesn't help alot.
If u want the rel config changes, I can post those also.

this is dev configuration.h
=========
//** my changes = all of these lines are commented out in the body
#ifndef MOTHERBOARD
#define MOTHERBOARD BOARD_RAMPS_13_EEB
#endif

#define STRING_CONFIG_H_AUTHOR "Bill Ruggirello" // Who made the changes.
//#define STRING_SPLASH_LINE1 BUILD_VERSION // will be shown during bootup in line 1=CHANGED
#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
#define STRING_SPLASH_LINE2 STRING_DISTRIBUTION_DATE // will be shown during bootup in line 2
#define SHOW_BOOTSCREEN

#define CUSTOM_MACHINE_NAME "CC-BOT 2E&QH v2e" // was CUSTOM_MENDEL_NAME

#define MACHINE_UUID "277494b7-b7c8-4eb1-a653-c2e695c24911"

#define EXTRUDERS 2

#define EXTRUDER_OFFSET_X {-30.0, 00.00} // (in mm) for each extruder, offset of the hotend on the X axis
// #define EXTRUDER_OFFSET_Y {0.0, 0.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1

#define EXTRUDER_WATTS (12.0*12.0/3.5) //  P=I^2/R
#define BED_WATTS (12.0*12.0/1.1)      // P=I^2/R

// PID settings
#define PIDTEMP
#define BANG_MAX 255 // limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#ifdef PIDTEMP
  //#define PID_DEBUG // Sends debug data to the serial port.
  //#define PID_OPENLOOP 1 // Puts PID in open loop. M104/M140 sets the output power from 0 to PID_MAX
  //#define SLOW_PWM_HEATERS // PWM with very low frequency (roughly 0.125Hz=8s) and minimum state time of approximately 1s useful for heaters driven by a relay
  #define PID_FUNCTIONAL_RANGE 10 // If the temperature difference between the target temperature and the actual temperature
                                  // is more then PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.
  #define PID_INTEGRAL_DRIVE_MAX PID_MAX  //limit for the integral term
  #define K1 0.95 //smoothing factor within the PID
  #define PID_dT ((OVERSAMPLENR * 10.0)/(F_CPU / 64.0 / 256.0)) //sampling period of the temperature routine

// CC-BOT E0 PLA @ 210
#define  DEFAULT_Kp 21.96
#define  DEFAULT_Ki 1.86
#define  DEFAULT_Kd 64.76

// CC-BOT E0 ABS @ 230
//#define  DEFAULT_Kp 17.46
//#define  DEFAULT_Ki 1.25
//#define  DEFAULT_Kd 64.11

#endif // PIDTEMP

/*
not in this version file - now in adv.h
#define THERMAL_RUNAWAY_PROTECTION_PERIOD 40 //in seconds
#define THERMAL_RUNAWAY_PROTECTION_HYSTERESIS 4 // in degree Celsius

//#define THERMAL_RUNAWAY_PROTECTION_BED_PERIOD 20 //in seconds
//#define THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS 2 // in degree Celsius
*/
[... 100 lines cut ...]
//** this is just here for reference - no change yet
//#define BABYSTEPPING
#ifdef BABYSTEPPING
  #define BABYSTEP_XY  //not only z, but also XY in the menu. more clutter, more functions
  #define BABYSTEP_INVERT_Z false  //true for inverse movements in Z
  #define BABYSTEP_Z_MULTIPLICATOR 2 //faster z movements

  #ifdef COREXY
    #error BABYSTEPPING not implemented for COREXY yet.
  #endif
  
    #ifdef DELTA
    #ifdef BABYSTEP_XY
      #error BABYSTEPPING only implemented for Z axis on deltabots.
    #endif
  #endif
#endif
*/
//#define BABYSTEPPING
#ifdef BABYSTEPPING
  #define BABYSTEP_XY  //not only z, but also XY in the menu. more clutter, more functions
  #define BABYSTEP_INVERT_Z false  //true for inverse movements in Z
  #define BABYSTEP_Z_MULTIPLICATOR 2 //faster z movements
#endif
=========

ruggb, 2015-08-19:
this might be interesting
the first list is what is different when I press the reset on the arduino
so the questions are:
Why is it different?
Why does the diff produce the error and an unknown command listed == which does not appear using the rel version?

reset pressed
.................same to this point
10:30:12.546 : N3 M111 S6 *68
10:30:12.547 : N4 T0 *30
10:30:12.547 : N5 M20 *52
10:30:12.547 : N6 M80 *61
10:30:13.422 : N7 M105 *0
10:30:15.696 : echo:SD init fail
10:30:15.788 : FIRMWARE_NAME:Marlin 1.1.0 Development From Archive SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:CC-BOT 2E&QH v2e EXTRUDER_COUNT:2 UUID:277494b7-b7c8-4eb1-a653-c2e695c24911
10:30:15.788 : N8 M220 S100 *73
10:30:15.788 : N9 M221 S100 *73
10:30:15.790 : echo:Active Extruder: 0
10:30:15.790 : N10 M111 S6 *118
10:30:15.790 : N11 T0 *42
10:30:15.791 : Begin file list
10:30:15.791 : End file list

Click connect in Repetier-host
same to this point...........

10:31:39.282 : N3 M105 *4
10:31:39.645 : N4 M111 S6 *67
10:31:39.646 : N5 T0 *31
10:31:39.646 : N6 M20 *55
10:31:39.646 : N7 M80 *60
10:31:42.348 : N1 M110 *2
10:31:42.348 : N2 M115 *4
10:31:42.348 : N3 M111 S6 *68
10:31:42.367 : echo:SD init fail
10:31:42.367 : N4 T0 *30
10:31:42.458 : FIRMWARE_NAME:Marlin 1.1.0 Development From Archive SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:CC-BOT 2E&QH v2e EXTRUDER_COUNT:2 UUID:277494b7-b7c8-4eb1-a653-c2e695c24911
10:31:42.460 : N5 M20 *52
10:31:42.461 : Error:No Line Number with checksum, Last Line: 3
10:31:42.467 : echo:Unknown command: ""
10:31:42.467 : N6 M80 *61
10:31:42.467 : N7 M105 *0
10:31:42.468 : echo:Active Extruder: 0
10:31:42.470 : Begin file list
10:31:42.470 : End file list

ruggb, 2015-08-20:
OK, I guess I must have clicked the wrong button.
back again

nitewing76, 2015-08-21:
Please download the newest Dev version and try again. You can also download Notepad++ and the Compare plugin to make updating your config.h easier...by loading only your current and new config.h into Notepadd++, then click on Plugins-->Compare-->Compare. 

When referencing any Dev version, it will help the developers know the date and time you download the Dev version. Also, when you change to a new Dev version, create a new folder to help you track the changes. For me, I have a separate folder for Release 1.0.2, @thinkyhead's G29debug, and a separate folder for each time I download the Dev version and have no errors (I am new, so I currently am using the folder named "Marlin-Dev_7").  Doing this will also help you discover which "wrong button" you clicked...it happens to all of us.

ruggb, 2015-08-21:
the problem with the compare is that the files keep changing so that things don't compare very well.
I have separate folders for EACH download.
the "wrong" button that I clicked was in the forum software.
I thought 8/00/15 was pretty definitive. R u telling me there is more than one version with the same date code?
sounds like a documentation issue - thought I couldn't quite understand day 00

nitewing76, 2015-08-22:
True, they do not compare perfectly....but they do compare close enough for basic troubleshooting. 

Yes...possibly a documentation error, but one easily overcome by knowing what date I downloaded the firmware ZIP file.

ruggb, 2015-08-23:
the doc thing I referred to was re the distribution date of the zip file.
I am now using my dnld date for the folder name.
I then chg the date in Default_Version.h to match it, but I am ignoring time.
I just thought that is what 
#define STRING_DISTRIBUTION_DATE "2015-08-00 12:00"
was for. But what do I know?

ruggb, 2015-08-23:
I have tried to search for possible causes for that error and find nothing.
I don't understand why it is showing undefined command "" around that error re line 3
But I reloaded the release version again and THIS time got the same error.
So it is a ghost and u can't work with ghosts, so I will close this.

closed 2015-08-23 by ruggb

thinkyhead, 2015-08-26:
The __Unknown command: ""__ message almost always indicates a communication problem between the computer and the Arduino, or a connection problem between the board and LCD controller.