MarlinFirmware/Marlin #2598: servo deactivation delay servo not deploying

giliammc, 2015-08-19:
my servo is twitching so i enabled servo deactivation delay and now the servo will not deploy when i run g28 or g29 but if i use m280 it works fine. 
PS thanks in advance

nitewing76, 2015-08-21:
Have you ever opened any files besides configuration.h, configuration_adv.h, or pins.h?  If you went exploring in other files to learn about them, you might have a stray character somewhere (probably in marlin_main.cpp). The easiest thing to do is to re-download that file and overwrite the current one....or rename your config files with _old at the end and overwrite everything.

giliammc, 2015-08-21:
i only edited pins.h and configuration.h i can upload my copy of marlin if you would like to see it

nitewing76, 2015-08-21:
Please check your configuration.h and see if lines 745 (near the bottom) look like this:

#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

Lines 751-752:

#define SERVO_ENDSTOPS {-1, -1, 0} // Servo index for X, Y, Z. Disable with -1
#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 101,170} // X,Y,Z Axis Extend and Retract angles

giliammc, 2015-08-21:
```
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
```
and

  ```
  #define SERVO_ENDSTOPS {-1, -1, 1} // Servo index for X, Y, Z. Disable with -1
#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 0,90} // X,Y,Z Axis Extend and Retract angles
```

giliammc, 2015-08-22:
here is a link to my copy of marlin 
https://drive.google.com/folderview?id=0B4v4FAQe3MjufkRidzdXWk91M3hqc1YxbkExWGVHN0Q3NGlYOXBOTHdJSi1VSnVCRHN4aEE&usp=sharing

nitewing76, 2015-08-22:
I think I see your problem....try changing your motherboard to say "#define MOTHERBOARD BOARD_PRINTRBOARD "...numbers are no longer being used.  Also, you may need to increases your servo deactivation delay to 200ms, then decrease it as needed. 

Also...your steps/mm (DEFAULT_AXIS_STEPS_PER_UNIT) and DEFAULT_MAX_FEEDRATE  seem to be incorrect. No 2 motors are ever the same and should be to at least 4 decimal places or greater. I will attach a link for the Excel spreadsheet I made to do the calculations, you can look at Tom's Hardware or another channel on YouTube for information on the process and get calibration cubes from Thingiverse (e.g. 20mm cube, 0.5mm thin wall test, etc)

https://www.dropbox.com/s/yg1wotj6a12gfcv/Motor%20calibration.xlsx?dl=0

giliammc, 2015-08-23:
for the steps per unit my x and y axis are the same mechanism for moving. those values seem to work and give accurate prints. i will try changing the board. when you say i need to change servo deactivation delay to 200ms do you mean enter 200 after define servo deactivation delay?

Roxy-3DPrintBoard, 2015-08-23:
> No 2 motors are ever the same and should be to at least 4 decimal places or greater.for the steps per unit my x and y axis are the same mechanism for moving. those values seem to work and give accurate prints. 

Careful here!  Most likely all three (four) of his motors ARE the same.  And these are going to be integer numbers.  The Z axis numbers will be scaled because of the threads/inch number of the Z axis screw.

> i will try changing the board. when you say i need to change servo deactivation delay to 200ms do you mean enter 200 after define servo deactivation delay?

I have a PrintRboard and I have my deactivation delay at a full 1000 ms.   There is no harm to going extra long.

giliammc, 2015-08-23:
Setting it to 1000ms didn't work so I tried setting it to 2000ms and it still won't deploy.

giliammc, 2015-08-23:
My x and y axis have the same motors and mechanisms for moving.

nitewing76, 2015-08-24:
Yes, the X/Y probe offset for G29 are integers, but the steps per mm are not required to be. Given the low tolerances of hobbyist level motors, the chances of two motors being the exact same to 4 decimal places are low. But if 100mm prints as 100m and verified with a good set of calipers, then consider yourself blessed. lol

giliammc, 2015-08-24:
@nitewing76 are you not useing stepper,s

nitewing76, 2015-08-26:
Yes, I have stepper motors.

thinkyhead, 2015-08-27:
I'm pretty sure this was fixed in the latest MarlinDev code. We had a long discussion about it and posted new code. @giliammc Are you able to use the latest Marlin?

giliammc, 2015-08-28:
@thinkyhead i fixed it by upgrading to marlin dev. PS thanks for all the help

closed 2015-08-28 by giliammc